XCSoar/XCSoar #1314: Android BLE, Engine Sensor Box stopped working (no Data)

fastdown101, 2023-11-19:
XCSoar versions having and not having the problem
-------------------------------------------------
Latest master

System information
------------------
Android 12, Redmi Node 9S.
I can see, that the ble device is sending data, and that in general die android device is capable receiving (tested with Serial Bluethooth App). But XCSoar connects to the Sensor Box and than claims "no Data". 

Steps to reproduce the behavior
-------------------------------
I cant debug it right now, because the docker image for build ANDROID is broken, or am I missing something?
I tested several ble devices, all with the same problem, "no Data"

Expected behavior
-----------------
Simply start xcs, then start the bluetooth sensor box and go fly...

Actual behavior
---------------
xcs connects to the ble device, than claims "no Data" 

Do you have any idea what may have caused this?
-----------------------------------------------
Maybe changes in xcs GATT implementation?

Do you have an idea how to solve the issue?
-------------------------------------------
Do other ble sensors have trouble too?

fastdown101, 2023-11-19:
Update: I can debug, missed something with the docker build on my developer machine.
The flowing is a log created with: `adb logcat | grep -e xcsoar -e bluetooth`
Ble sensor device which should connect to xcs has the mac: `C0:49:EF:BD:3E:76`
I can not make sens of the log... whats is wrong?
Greetings, Daniel

`11-19 12:37:04.659  1804  2588 D MIUIInput: [MotionEvent] publisher action=0x0, 617526, channel '188fe1d org.xcsoar/org.xcsoar.XCSoar (server)'
11-19 12:37:04.676  2786  2887 D BtGatt.GattService: bleAppName = org.xcsoar#
11-19 12:37:04.681  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
11-19 12:37:04.683 16404 16404 I GH.WifiBluetoothRcvr: Connection action: android.bluetooth.device.action.ACL_CONNECTED, device C0:49:EF:BD:3E:76
11-19 12:37:04.699  1804  2588 D MIUIInput: [MotionEvent] publisher action=0x1, 617566, channel '188fe1d org.xcsoar/org.xcsoar.XCSoar (server)'
11-19 12:37:04.701  2786  2887 D BtGatt.GattService: bleAppName = org.xcsoar#org.xcsoar#
11-19 12:37:04.719  2786  2887 D BtGatt.GattService: bleAppName = org.xcsoar#org.xcsoar#org.xcsoar#
11-19 12:37:04.722  7180  8429 I bcje    : (REDACTED) bluetoothConnectionStateChanged. Action: %s
11-19 12:37:04.733  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
11-19 12:37:04.733  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
11-19 12:37:04.743  2786  2887 D BtGatt.GattService: bleAppName = org.xcsoar#org.xcsoar#org.xcsoar#org.xcsoar#
11-19 12:37:04.781  1804  3903 W BroadcastQueueInjector: Unable to launch app com.spotify.music/10233 for broadcast Intent { act=android.bluetooth.device.action.ACL_CONNECTED flg=0x5000010 (has extras) }: process is not permitted to  auto start
11-19 12:37:04.786  1804  2220 W BroadcastQueueInjector: Unable to launch app com.spotify.music/10233 for broadcast Intent { act=android.bluetooth.device.action.ACL_CONNECTED flg=0x5000010 (has extras) }: process is not permitted to  auto start
11-19 12:37:04.814  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
11-19 12:37:05.004  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
11-19 12:37:05.004  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
11-19 12:37:05.354  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
11-19 12:37:05.994  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
11-19 12:37:05.994  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
11-19 12:37:06.076  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
11-19 12:37:06.190  1219  3596 I vendor.qti.bluetooth@1.0-ibs_handler: DeviceSleep: TX Awake, Sending SLEEP_IND
11-19 12:37:06.190  1219  3596 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
11-19 12:37:06.352  1219  3411 D vendor.qti.bluetooth@1.0-wake_lock: Release wakelock is released 
11-19 12:37:07.049  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
11-19 12:37:07.049  1219  3591 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
11-19 12:37:07.050  1219  3591 D vendor.qti.bluetooth@1.0-wake_lock: Acquire wakelock is acquired 
11-19 12:37:07.050  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
11-19 12:37:07.131  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
11-19 12:37:07.131  1219  3591 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
11-19 12:37:07.282  1219  3411 D vendor.qti.bluetooth@1.0-wake_lock: Release wakelock is released 
11-19 12:37:08.039  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
11-19 12:37:08.039  1219  3591 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
11-19 12:37:08.040  1219  3591 D vendor.qti.bluetooth@1.0-wake_lock: Acquire wakelock is acquired 
11-19 12:37:08.040  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
11-19 12:37:08.122  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
11-19 12:37:08.123  1219  3591 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
11-19 12:37:08.274  1219  3411 D vendor.qti.bluetooth@1.0-wake_lock: Release wakelock is released 
11-19 12:37:09.030  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
11-19 12:37:09.030  1219  3591 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
11-19 12:37:09.031  1219  3591 D vendor.qti.bluetooth@1.0-wake_lock: Acquire wakelock is acquired 
11-19 12:37:09.031  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
11-19 12:37:09.113  1219  3591 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
11-19 12:37:09.114  1219  3591 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
11-19 12:37:09.264  1219  3411 D vendor.qti.bluetooth@1.0-wake_lock: Release wakelock is released 
[... cut ...]
11-19 12:37:14.643  1804  2220 W BroadcastQueueInjector: Unable to launch app com.spotify.music/10233 for broadcast Intent { act=android.bluetooth.device.action.ACL_DISCONNECTED flg=0x5000010 (has extras) }: process is not permitted to  auto start
11-19 12:37:15.505  1219  3596 I vendor.qti.bluetooth@1.0-ibs_handler: DeviceSleep: TX Awake, Sending SLEEP_IND
11-19 12:37:15.505  1219  3596 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
11-19 12:37:15.656  1219  3411 D vendor.qti.bluetooth@1.0-wake_lock: Release wakelock is released 

`

MaxKellermann, 2023-11-19:
Bisect it.

MaxKellermann, 2023-11-20:
No. `git bisect` to find the first commit that has the problem. So far, your bug report lacks vital information, and doing the bisect could maybe fill in for the information you did not submit.

fastdown101, 2023-11-23:
How do I get the correct build docker for different commits/tags?
For instance, I do the following, `git checkout v7.34` then `git submodule update --init --recursive` and finally `docker run --mount type=bind,source="$(pwd)",target=/opt/xcsoar -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest xcsoar-compile ANDROID`. But this fails, I guess because of the wrong container... How do I do this propperly?

lordfolken, 2023-11-23:
This is properly. Do you have an error?

MaxKellermann, 2023-11-23:
A proper reply to his questions won't help him, that's why I didn't bother to take the time to respond. If he has build failures but keeps the details secret, nobody can help him.

lordfolken, 2023-11-23:
I agree... painful.

The latest tag on the container image on the ghcr.io registry, always points to master. 

If you are building an old version you need to build the container yourself for that version, as there may have been ndk version changes or other dependencies in the meantime.

This can be done as described here:
https://github.com/XCSoar/XCSoar/blob/master/ide/docker/README.md

```bash
docker build \
    --file ide/docker/Dockerfile \
    -t xcsoar/xcsoar-build:latest ./ide/
```
Then use that image to run your build.

fastdown101, 2023-11-23:
Okay, this might take a long time to tackle...here is where I am stuck now:

`git status` gives me `HEAD detached at v7.34
nothing to commit, working tree clean`
did a `git submodule update --init --recursive`
then `docker build --file ide/docker/Dockerfile -t xcsoar/xcsoar-build:latest ./ide/`

should be okay, right?

`Sending build context to Docker daemon  147.5kB
Step 1/12 : FROM debian:bullseye
 ---> ca1a8403096e
Step 2/12 : ENV DEBIAN_FRONTEND=noninteractive
 ---> Using cache
 ---> f44ef18eb7b5
Step 3/12 : RUN apt-get update && apt-get install -y python3 vim ccache python3-pkgconfig gdb && apt-get clean
 ---> Using cache
 ---> 69ea663f17b1
Step 4/12 : COPY ./provisioning/install-debian-packages.sh /root/install-debian-packages.sh
 ---> Using cache
 ---> 3a14e7b59bc7
Step 5/12 : RUN /root/install-debian-packages.sh && apt-get clean
 ---> Using cache
 ---> f42b56a878c3
Step 6/12 : COPY ./provisioning/install-android-tools.sh /root/install-android-tools.sh
 ---> Using cache
 ---> 035b013fce19
Step 7/12 : RUN chmod 0755 /root/install-android-tools.sh /root/install-debian-packages.sh
 ---> Using cache
 ---> 91f272ab8e57
Step 8/12 : RUN /root/install-android-tools.sh && apt-get clean
 ---> Using cache
 ---> 841287c59e95
Step 9/12 : COPY ./docker/bin/* /usr/local/bin/
 ---> Using cache
 ---> 4caeb7367b34
Step 10/12 : WORKDIR /usr/local/bin
 ---> Using cache
 ---> 08321d07619f
Step 11/12 : RUN chmod 755 ./*
 ---> Using cache
 ---> 0b1c8d737819
Step 12/12 : WORKDIR /opt/xcsoar
 ---> Using cache
 ---> 0249de365f3d
Successfully built 0249de365f3d
Successfully tagged xcsoar/xcsoar-build:latest
`

trying to compile stuff

`docker run --mount type=bind,source="$(pwd)",target=/opt/xcsoar -it xcsoar/xcsoar-build:latest xcsoar-compile ANDROID`

but no, something is wrong... 

`make[1]: Entering directory '/opt/xcsoar'
make[1]: Entering directory '/opt/xcsoar'
make[1]: Entering directory '/opt/xcsoar'
make[1]: Entering directory '/opt/xcsoar'
make[1]: *** No rule to make target '/root/opt/android-ndk-r26b/toolchains/llvm/prebuilt/linux-x86_64/bin/../sysroot/usr/include/c++/v1/optional', needed by 'output/ANDROID/x86/dbg/src/io/MapFile.o'.  Stop.
make[1]: Leaving directory '/opt/xcsoar'
make: *** [build/android.mk:333: output/ANDROID/x86/dbg/bin/libxcsoar.so] Error 2
make: *** Waiting for unfinished jobs....
make[1]: *** No rule to make target '/root/opt/android-ndk-r26b/toolchains/llvm/prebuilt/linux-x86_64/bin/../sysroot/usr/include/c++/v1/optional', needed by 'output/ANDROID/armeabi-v7a/dbg/src/io/MapFile.o'.  Stop.
make[1]: Leaving directory '/opt/xcsoar'
make: *** [build/android.mk:333: output/ANDROID/armeabi-v7a/dbg/bin/libxcsoar.so] Error 2
make[1]: *** No rule to make target '/root/opt/android-ndk-r26b/toolchains/llvm/prebuilt/linux-x86_64/bin/../sysroot/usr/include/c++/v1/optional', needed by 'output/ANDROID/x86_64/dbg/src/io/MapFile.o'.  Stop.
make[1]: Leaving directory '/opt/xcsoar'
make[1]: *** No rule to make target '/root/opt/android-ndk-r26b/toolchains/llvm/prebuilt/linux-x86_64/bin/../sysroot/usr/include/c++/v1/optional', needed by 'output/ANDROID/arm64-v8a/dbg/src/io/MapFile.o'.  Stop.
make[1]: Leaving directory '/opt/xcsoar'
make: *** [build/android.mk:333: output/ANDROID/x86_64/dbg/bin/libxcsoar.so] Error 2
make: *** [build/android.mk:333: output/ANDROID/arm64-v8a/dbg/bin/libxcsoar.so] Error 2
`

lordfolken, 2023-11-23:
https://github.com/XCSoar/XCSoar/blob/v7.34/ide/provisioning/install-android-tools.sh according to this ndk 25c should be installed not r26b. 

Cam you rm rf ./output folder in your xcsoar dir?

fastdown101, 2023-11-23:
Yes, removing old stuff did the trick, it compiles now, thanks

fastdown101, 2023-11-23:
Works up to v7.39, now I have to dig deeper into the v7.40 which definitively does not work

fastdown101, 2023-11-24:
That is my findings:

`cbac070bfb50efd4ee89f1af519dc3aa32e4b84b fails`
`68bd709c17c1fdbb175c0ed2f90482f10c77c289 ok`

so `(https://github.com/XCSoar/XCSoar/commit/cbac070bfb50efd4ee89f1af519dc3aa32e4b84b)` breaks it

fastdown101, 2023-11-25:
I am a bit lost. Connection to the ble device seems okay. But data flow is somehow messed up? @MaxKellermann you moved Bluetooth stuff around? Any suggestions how to tackle the problem?
Thanks and greetings, 
Daniel

PS:
If someone owns a esp32 devkit like this one [ESP32-DEVKITC32E](https://mou.sr/3sP3xIU), you can debug the engine box with this simplified arduino sketch

```cpp
#include <ArduinoBLE.h>
#include "esp_bt.h"
#include <esp_task_wdt.h>

#include <array>
#include <cstdint>

// Watchdog timeout
#define WDT_TIMEOUT 2

std::array<uint8_t, 13> mem_ = {};                          // Memory for local sensor data.
uint8_t& status = reinterpret_cast<uint8_t&>(mem_[0]);      // General status.
uint16_t& t1 = reinterpret_cast<uint16_t&>(mem_[1]);        // MAX6675 #1.
uint16_t& t2 = reinterpret_cast<uint16_t&>(mem_[3]);        // MAX6675 #2.
int16_t& t3 = reinterpret_cast<int16_t&>(mem_[5]);          // BMP280 temp.
uint32_t& pressure = reinterpret_cast<uint32_t&>(mem_[7]);  // BMP280 pressure.
uint16_t& rpm = reinterpret_cast<uint16_t&>(mem_[11]);      // Revolutions per minute of the engine.

BLEService ppgmService("D2865ECA-2C07-4610-BF03-8AEEBEF047FB");
BLECharacteristic ppgmCharacteristic("D2865ECB-2C07-4610-BF03-8AEEBEF047FB", BLERead | BLEWrite | BLENotify, mem_.size(), true);

// *****************************************************************************************
void setup() {
  esp_task_wdt_init(WDT_TIMEOUT, true);  // Enable panic so ESP32 can be restart by watchdog (WD)

  Serial.begin(9600);  // Initialize serial communication for debugging
  while (!Serial);

  // BLE stuff
  if (!BLE.begin()) {
    Serial.println("starting BLE failed!");
    while (1);
  }
  esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P7);
  esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN, ESP_PWR_LVL_P7);
  Serial.print("ADV Power Level:");
  Serial.println(esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_ADV));
  Serial.print("SCAN Power Level:");
  Serial.println(esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_ADV));

  // More BLE stuff
  BLE.setLocalName("PPGM-V2");
  BLE.setAdvertisedService(ppgmService);
  ppgmService.addCharacteristic(ppgmCharacteristic);
  BLE.addService(ppgmService);
  BLE.advertise();
  BLE.setConnectable(true);
  Serial.println("Bluetooth® device active, waiting for connections...");

}

// *****************************************************************************************
void loop() {

  // Wait for a Bluetooth® Low Energy central.
  BLEDevice central = BLE.central();
  if (central) {
    Serial.print("Connected to central: ");
    Serial.println(central.address());

    // Set ble power to max for all possible connection handles.
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL0, ESP_PWR_LVL_P7);
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL1, ESP_PWR_LVL_P7);
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL2, ESP_PWR_LVL_P7);
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL3, ESP_PWR_LVL_P7);
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL4, ESP_PWR_LVL_P7);
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL5, ESP_PWR_LVL_P7);
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL6, ESP_PWR_LVL_P7);
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL7, ESP_PWR_LVL_P7);
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_CONN_HDL8, ESP_PWR_LVL_P7);

    esp_task_wdt_add(NULL);  // Add current thread to WDT watch.

    // While central is connected, check sensors every 1000ms and toggle indicator LED
    long previousMillis = 0;
    while (central.connected()) {
      long currentMillis = millis();
      if (currentMillis - previousMillis >= 1000) {
        updateSensorData();
        previousMillis = currentMillis;
        esp_task_wdt_reset();
      }
    }

    esp_task_wdt_delete(NULL);  // Delete current thread from WDT watch

    Serial.print("Disconnected from central: ");
    Serial.println(central.address());
  }
}

// *****************************************************************************************
void updateSensorData() {
  static uint16_t cnt = 0;

  status = 1;
  t1 = 182 + 273 + (cnt % 30);
  t2 = 510 + 273 + (cnt % 40);
  t3 = 22 + 273;
  pressure = 92000;
  rpm = 36 + (cnt % 10);

  ppgmCharacteristic.writeValue(mem_.data(), mem_.size(), false);  // Update characteristic 
  
  cnt++;
}
```

fastdown101, 2023-12-01:
Never mind... I will stick to the woking version. Issue can be close.

DanD222, 2023-12-01:
I would take it that Max is just busy with other things in his life/work at this particular moment in time, so.. closing the issue if it’s still valid doesn’t seem that great of an idea to me.

MaxKellermann, 2023-12-01:
> Never mind... I will stick to the woking version. Issue can be close.

You can do whatever you want, and if you stick to an older version forever, you'll never need my help again. Win-win.

fastdown101, 2023-12-01:
It is exactly this atitude Max, which makes me think to move on from xcs...

MaxKellermann, 2023-12-01:
Oh, now it is your attitude to make this personal. People who make technical discussions personal are an annoying waste of time, and I have learned to quickly ignore people who do that. Bye then, you're now blocked.

closed 2023-12-20 by MaxKellermann