XCSoar/XCSoar #1317: Windows build is broken

andrestotorica, 2023-12-19:
XCSoar versions having and not having the problem
-------------------------------------------------
HEAD.

It seems the problem was introduced with the update to Boost 1.83.
Build succeeds if reverting to 1.82 (`git revert b5d254f 10b2f7f a627941`) - I don't know if the program works properly in this case, it just manages to build.

Steps to reproduce the behavior
-------------------------------

* Checkout HEAD
* Build on the docker env: `docker run --mount type=bind,source="$(pwd)",target=/opt/xcsoar -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest xcsoar-compile WIN64`
  * Also happens on a freshly built docker image
  * Note that the [build-windows](https://github.com/XCSoar/XCSoar/actions/runs/7165100433/job/19506418337) workflow is also breaking with this error among others.

Actual behavior
---------------
Error is a redeclaration of compiler builtins `__builtin_ia32_crc32qi, __builtin_ia32_crc32hi, __builtin_ia32_crc32si`, on file `intrin.h` included from Boost headers.

I don't understand if the problem lies in the functions being declared somewhere else on other headers, or they shouldn't be declared at all as they are built ins.

Do you have an idea how to solve the issue?
-------------------------------------------

No. From the comment at the top of `intrin.h`, it seems this file is aimed to provide implementations of MS Visual Studio built-ins for compatibility in gcc. But it confuses me as the functions are also gcc built-ins. In any case I don't understand why Boost includes this header explicitly, and if this is ok.

A temporary workaround is removing `-Wredundant-decls` though it is not great.

Full error
---------

`  CXX     output/WIN64/dbg/src/json/Boost.o`
`<built-in>: error: redundant redeclaration of ‘unsigned int __builtin_ia32_crc32qi(unsigned int, unsigned char)’ in same scope [-Werror=redundant-decls]`
`In file included from ./output/src/boost_1_83_0/boost/json/detail/charconv/detail/fast_float/float_common.hpp:70,`
`                 from ./output/src/boost_1_83_0/boost/json/detail/charconv/detail/fast_float/fast_float.hpp:11,`
`                 from ./output/src/boost_1_83_0/boost/json/detail/charconv/impl/from_chars.ipp:14,`
`                 from ./output/src/boost_1_83_0/boost/json/src.hpp:58,`
`                 from src/json/Boost.cxx:17:`
`/usr/share/mingw-w64/include/intrin.h:59:23: note: previous declaration of ‘unsigned int __builtin_ia32_crc32qi(unsigned int, unsigned char)’`
`   59 |   extern unsigned int __builtin_ia32_crc32qi (unsigned int, unsigned char);`
`      |                       ^~~~~~~~~~~~~~~~~~~~~~`
`<built-in>: error: redundant redeclaration of ‘unsigned int __builtin_ia32_crc32hi(unsigned int, short unsigned int)’ in same scope [-Werror=redundant-decls]`
`/usr/share/mingw-w64/include/intrin.h:60:23: note: previous declaration of ‘unsigned int __builtin_ia32_crc32hi(unsigned int, short unsigned int)’`
`   60 |   extern unsigned int __builtin_ia32_crc32hi (unsigned int, unsigned short);`
`      |                       ^~~~~~~~~~~~~~~~~~~~~~`
`<built-in>: error: redundant redeclaration of ‘unsigned int __builtin_ia32_crc32si(unsigned int, unsigned int)’ in same scope [-Werror=redundant-decls]`
`/usr/share/mingw-w64/include/intrin.h:61:23: note: previous declaration of ‘unsigned int __builtin_ia32_crc32si(unsigned int, unsigned int)’`
`   61 |   extern unsigned int __builtin_ia32_crc32si (unsigned int, unsigned int);`
`      |                       ^~~~~~~~~~~~~~~~~~~~~~`
`cc1plus: all warnings being treated as errors`
`make: *** [build/compile.mk:111: output/WIN64/dbg/src/json/Boost.o] Error 1`

MaxKellermann, 2023-12-20:
This has nothing to do with Boost or with XCSoar, this problem is only in your mingw toolchain.

lordfolken, 2023-12-20:
So what is wrong with our mingw toolchain? It doesnt work in the github actions either.

andrestotorica, 2023-12-20:
Ok so we have 2 issues. The `intrin.h` issue is also present the GH workflow.

1- The `std::byte` issue does not happen on the docker build. Should we try to install that mingw version on the GH workflows image?
2- No idea how the `intrin.h` issue can be tackled gracefully. How open would you be to conditionally excluding `-Wredundant-decls` for Windows targets?

MaxKellermann, 2023-12-21:
> 1- The `std::byte` issue does not happen on the docker build. Should we try to install that mingw version on the GH workflows image?

How?

> 2- No idea how the `intrin.h` issue can be tackled gracefully. How open would you be to conditionally excluding `-Wredundant-decls` for Windows targets? The redeclaration in the header is just the prototype, no alternative implementation to the built in is given as far as I can see.

No. You can always compile with `WERROR=n` if you want, or add your custom flags to `CXXFLAGS` on the command line.

andrestotorica, 2023-12-23:
After investigating some more, my point # 1 above is wrong. The problem was not appearing on the docker image because I was just not building the tests.

I believe neither the GH actions (ubuntu:22.04), the Docker build (debian:bullseye) nor my local ubuntu:latest environments can do a full build, all with the same errors and warnings.

Do we know of an environment (or a gcc/mingw version) that can successfully do a `make everything TARGET=WIN64`?

How was the binary for the 7.40 release on 2-Nov built?

MaxKellermann, 2023-12-23:
I build everything on Debian Bookworm (with the mingw shipped by Debian) and I have no such errors.

andrestotorica, 2023-12-23:
Ok, excellent, I got same results. I had problems with the JDK 11 package not available on Bookworm, just replaced it with the JDK 17 equivalent.

I don't know what are the implications on the Android side for updating to JDK 17. If that is possible, the provisioning script and Dockerfile can be updated and problem would be solved.

MaxKellermann, 2024-01-04:
Closing because this is a toolchain bug, not a XCSoar bug.

closed 2024-01-04 by MaxKellermann

andrestotorica, 2024-01-09:
@MaxKellermann just curious, what setup do you have? Debian Bookworm + some workaround for the JDK dependency?