XCSoar/XCSoar #1318: Data folder writable from another app

llauner, 2023-12-25:
XCSoar version
--------------
7.40

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Data file location in a Public folder. Revert to the XCSoarData public folder, or use /Download

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
The idea is to use an external download app to automatically download and write the .txt, .cup files to the XCSoar data folder. This is no longer possible as the files are written storage/emulated/0/Android/data/org.xcsoar/files/ which is not writable from another application.
This used to work fine when files were stored into the public XCSoarData folder.
Please add this back, or add the Download folder as well so that we can keep using an app to automatically upload and update the airspace and cup files.
P.S: I know that XCSoar as a way to download files, but we have no control over what these files are and when / how they are maintained.

lordfolken, 2024-01-01:
Dublicate #992 see there for discussions and solutions.

closed 2024-01-01 by lordfolken

llauner, 2024-01-01:
This is not providing a solution:
It's saying you can't write to org.xcsoar/files 
It's saying you can do that **manually** with a file explorer

But it's not providing a way for an external file application to write data to the XCSoar file location.
Why would XCSoar not allow a public (writable from another app) folder to be used for irs files, just like the XCSoarData folder did ?

lordfolken, 2024-01-01:
> Why would XCSoar not allow a public (writable from another app) folder to be used for irs files, just like the XCSoarData folder did ?

This is a google android limitation enforced by appstore policies, not a choice of xcsoar. 
What would be needed to do so in an android native way, has been discussed in #992 at length. 

> But it's not providing a way for an external file application to write data to the XCSoar file location.
The way forward is to deprecate file access, and actually provide tooling to manage what inside the app folder. For example #1114

llauner, 2024-01-01:
SeeYou Navigator is reading its files (waypoints, airspace, ...) from the /Downloads folder.

lordfolken, 2024-01-08:
Why don't you contribute your files as urls to the https://github.com/xcsoar/xcsoar-data-content repo, and then just use XCSoar's built in File Manager to download/update the files?

The first one from your [app](https://github.com/llauner/AirspaceDownloader) is already in there. 
https://github.com/search?q=repo%3AXCSoar%2Fxcsoar-data-content%20planeur&type=code

llauner, 2024-01-08:
> Why don't you contribute your files as urls to the https://github.com/xcsoar/xcsoar-data-content repo, and then just use XCSoar's built in File Manager to download/update the files?
>
> The first one [...]

@lordfolken That's a very good suggestion, thank you !
I had no idea that this could be done this way...I'll have a look at it and add the file URLs to the repo !
Thanks

lordfolken, 2024-01-08:
It works for induvidual files.  Where I see a problem is with cupx, respectively pics archive. See #795

lordfolken, 2024-01-08:
I made a design proposal for a sane format instead of cupx, called xcd.
https://github.com/XCSoar/XCSoar/issues/1025#issuecomment-1230839276

If you need forensic tools to open a file, its really not well designed.

llauner, 2024-01-09:
Hi @lordfolken
You're proposition is great, do you have any timeline for the implemetation ?
There could also be another way to look at things. Even though cupx brings some difficulties in the handling, it's currently the only and very popular format where you can embed an image along with the waypoint description.
It's widespread, commonly adopted and has software (SeeYou) to generate it from a UI.
Wouldn't it make sens to XCSoar to support this popular and widespread format and enlarge its user base rather than trying to propose another one even if it's theoretically better ?
I'm not saying I have the answer, but it's just a thought...