XCSoar/XCSoar #1319: LX Nav S10 incorrect ballast synchronization

tatro77, 2023-12-26:
XCSoar versions having and not having the problem
-------------------------------------------------

Current version 7.38 

System information
------------------

 Stefly Nav 7.0, Android 11.

Steps to reproduce the behavior
-------------------------------

 It is not possible to match the water ballast setting of the LXNav S10 and XCsoar. 

Expected behavior
-----------------

Ideally the settings for water ballast should match and drain at the same rate.

Actual behavior
---------------

Currently  when 100L of water ballast are  set in  XCsoar (flight setup-custom), the LXNAV S10 shows 70L

Do you have any idea what may have caused this?
-----------------------------------------------

no

Do you have an idea how to solve the issue?
-------------------------------------------

no

tatro77, 2023-12-26:
[image]

lordfolken, 2023-12-26:
Please switch on debug on the serial port and attach the resulting log.

tatro77, 2023-12-27:
Hi. 
Here is the log. :)

tatro77, 2023-12-27:
[xcsoar.log](https://github.com/XCSoar/XCSoar/files/13778855/xcsoar.log)

lordfolken, 2023-12-27:
The ballast is communicated as an overload factor. 
$PLXV0, BAL,R/W,Overload factor*CS, where overload factor is defined as ratio between current mass and mass of glider, where polar was calculated

PLXV0,BA L,W,1.50

Please show us the configured plane profile.

tatro77, 2023-12-27:
[image]
[image]
[image]

lordfolken, 2023-12-27:
So according to src/ActionInterface.ccp overload is defined as:
https://github.com/XCSoar/XCSoar/blob/f0437208763b629de30d459f8f9ff0fb584477b5/src/ActionInterface.cpp#L105-L107

However the LX Manual gives a different formula:
[image]

@hjr Can you check please?

tatro77, 2023-12-29:
Thank you.
I believe the overload formula from the manual LX nav manual is correct. Overload =1.36  about 73kg when XCsoar shows 100l. I am not sure if I missed something.

lordfolken, 2023-12-30:
So this function is used centrally for several drivers. Therefore delicate testing is required. 

In Summary:

Vaulter: Uses dry mass, would break with the change (would break) 
Cai302: Uses Fraction (OK)
Borglet: Uses Cai302::PutBallast thefore fraction (OK)
LX1600: Uses reference Weight (currently broken)
LXNAV: Uses reference Weight (currently broken)
OpenVario: Uses overload and reference weight (currently broken)
XCVario: uses fraction, (OK)

src/Device/Driver/Vaulter.cpp:
 Wing Load:
     WL,<value>
     <value> is a factor of the additional weight added by water ballast in relation to the reference weight of the glider
     Example: $POV,C,WL,1.0*12 => No water ballast
              $POV,C,WL,1.1*13 => 10% more weight than reference weight       

So Vaulter would break changing this. 

CAI302 uses fraction, therefore not a problem. 

lx1600:  According to current lxnavigation documentation:

    char buffer[100];
    sprintf(buffer, "PFLX2,,%.2f,,,,", overload);
    PortWriteNMEA(port, buffer, env);

<load_factor> float Total glider mass divided by polar reference mass

Should be broken currently and the change would be appropriate

LXNAV:
  SetBallast(Port &port, OperationEnvironment &env, double overload)
  {
    char buffer[100];
    sprintf(buffer, "PLXV0,BAL,W,%.2f", overload);
    PortWriteNMEA(port, buffer, env);
  }

Overload Wrong

OpenVario:
bool
OpenVarioDevice::RepeatBallast(OperationEnvironment &env)
{
  if (!_overload_valid) {
    return InformUnavailable("WL", env);
  }
  
  char buffer[20];
  sprintf(buffer,"POV,C,WL,%1.3f",(float)_overload);
  PortWriteNMEA(port, buffer, env);
  return true;
}

Wing Load:
     WL,<value>
     <value> is a factor of the additional weight added by water ballast in relation to the reference weight of the glider
     Example: $POV,C,WL,1.0*12 => No water ballast
              $POV,C,WL,1.1*13 => 10% more weight than reference weight      

So also currently broken. 

XCVario: uses fraction therefore not affected.

MaxKellermann, 2024-01-04:
> However the [LX Manual](https://gliding.lxnav.com/wp-content/uploads/manuals/S8x-S10xManualEnglishVer908rev62.pdf) p.84 gives a different formula:

How is this formula different, and is this difference relevant for this bug?

(Please do not use the term "weight" - it's about mass, not weight. It's sad to see that even instrument manufacturers mix this up.)

The difference I see:
- XCSoar uses "dry mass", which is "empty mass" plus "actual crew mass"
- LXNAV uses "reference mass", which is "empty mass" plus "some unspecified crew mass"

So it's just how much crew is accounted for?

Right? Am I missing something?

kobedegeest, 2024-01-05:
> How is this formula different, and is this difference relevant for this bug?

The difference lies in the denominator. XCSoar divides by dry mass which makes no sense as you want to get a scaling factor to rescale the glide polar which is calculated at the reference mass. So besides the incorrect use of the word weight the LX formula is the correct one. You can't get a scaling factor without the use of the mass used for the original calculation/measurement.

MaxKellermann, 2024-01-05:
@kobedegeest, so that's a "yes"?

> XCSoar divides by dry mass which makes no sense

It makes no sense only with your personal definition of "scaling factor", but it does make sense with a different definition. When we discuss how to change XCSoar's formula, we need to consider the definition of each and every user of the term "scaling factor". XCSoar and LXNAV have a different definition, so this is really a bug, but as @lordfolken said we must change this formula only for LXNAV S10 and no other product (unless other products also have a different definition, which so far nobody has pointed out).

lordfolken, 2024-01-05:
> How is this formula different, and is this difference relevant for this bug?
>
> (Please do not use the term "weight" - it's about mass, not weight. It's sad to see that even instrument manufacturers [...]

Only when it is talking to the drivers when updating the weight.

 It also uses the polar reference_mass for the glide polar calculation:
https://github.com/XCSoar/XCSoar/blob/227840b644872fee3f445f6e6ec1c56349d04b62/src/Engine/GlideSolvers/GlidePolar.hpp#L78-L79

>     * LXNAV uses "reference mass", which is "empty mass" plus "some unspecified crew mass"
The reference mass is the mass at which the polar was measured. 

So the first number from polarstore.hpp
https://github.com/XCSoar/XCSoar/blob/227840b644872fee3f445f6e6ec1c56349d04b62/src/Polar/PolarStore.cpp#L179

325kg for the ls8(15m).

The overload factor is defined as the current total mass (crew + empty plane + water ballast) / (the polar reference mass)

So basically if you had a 255 empty plane + a 70kg  piloy = 325. So your overload factor towards the polar would be 1.

MaxKellermann, 2024-01-05:
> The overload factor is defined as the current total mass (crew + empty plane + water ballast) / (the polar reference mass)

Who exactly defines it that way, and which devices agree with this definition?

Take this document for example: https://www.lx-avionik.de/wp/download/manuals/LXS3ManualEnglishVer0101.pdf page 20
Instead of "reference mass", it uses the "minimum glider weight", which is defined as "Minimum glider weight is a weight of empty glider + minimum pilot weight". This is different from "reference mass" = "mass at which the polar was measured".

So now we already have two documents by one manufacturer (LXNAV) with different definitions of "overload factor". Sh*t!

lordfolken, 2024-01-05:
> @kobedegeest, so that's a "yes"?
>
>
> It makes no sense only with your personal definition of "scaling factor", but it does make sense with a different definition. When we discuss how to change [...]

I went through all the documentation i could find when i researched https://github.com/XCSoar/XCSoar/issues/1319#issuecomment-1872533910

The overload factor is always divided by the reference_mass, for all mentioned products, except for the vaulter driver. 
So currently broken are the following drivers:
* lxnav
* lxnavigation
* OpenVario

According to the vaulter manual, its the only one to use dry mass. 
[image]

MaxKellermann, 2024-01-05:
The LXNAV LX9000 manual also uses "minimum glider weight", not "reference mass".

lordfolken, 2024-01-05:
Uh... that is indeed ugly. 
S3 (minimal glider weight)
https://gliding.lxnav.com/wp-content/uploads/manuals/LXS3ManualEnglishVer0110rev7.pdf
S7 (minimal glider weight)
https://gliding.lxnav.com/wp-content/uploads/manuals/LXS7ManualEnglishVer0533rev17.pdf

MaxKellermann, 2024-01-05:
I don't think XCSoar should put a meaning in "overload factor". It's a bad metric, it's useless for calculations (you need to derive the ballast mass from it before you can do anything, so why not just use the ballast mass in the first place?), and there are (at least) three valid contradicting definitions for it.
Each device driver which needs an "overload factor" should calculate it whenever needed using the definition from that device's manufacturer.
We shouldn't care how many definitions there are, as long as we use the right one for each device.

lordfolken, 2024-01-05:
So according to the protocol specification 1.04 from LXNAV: 

[image]

[image]

lordfolken, 2024-01-05:
I´ll contact them and ask them whether this is a bug in the docs...

lordfolken, 2024-01-05:
> I don't think XCSoar should put a meaning in "overload factor". It's a bad metric, it's useless for calculations (you need to derive the ballast mass from it before you can do anything, so why not just use the ballast mass in the first place?)

Every's plane dry mass is different. For the polar it makes sense to take the weight at which a,b,c, is defined, and then transform it to the current total mass. How much of that is ballast / crew or else isn't really relevant.

MaxKellermann, 2024-01-05:
Of course, for glide path calculations, only the actual mass (or rather: weight! yes! .. which includes G load) of the glider is relevant, and knowing the amount of pilot mass vs water ballast doesn't matter here. But this misses the point I made, because the current mass of the glider is not a primary metric, but a calculated value. I was talking about primary metrics, and the ballast mass is a good primary metric (one that the user enters because they know how much water they filled into the wings); the overload factor is a very bad primary metric and also a very bad calculated metric. The overload factor should not be something that is entered by anybody or transferred to/from a device. It could be displayed if the user is curious about this value, but it's really not useful for anything.
In my opinion, a device that allows transferring the overload factor over its machine-readable protocol is not well-designed.

lordfolken, 2024-01-07:
I have access to an S80, which runs the same firmware as the S100. 

Some testing just on the vario side and observing the NMEA traffic. 

  |   |   |   |   |   |  
-- | -- | -- | -- | -- | -- | --
Reference Weight | Pilot Weight | Empty Weight |   | Ballast | xcsoar bal | LXWP2
300 | 0 | 200 |   | 40 |   | 0.8
300 | 40 | 200 |   | 0 |   | 0.8
300 | 0 | 100 |   | 140 |   | 0.8
300 | 100 | 200 |   | 0 |   | 1
300 | 120 | 200 |   | 0 |   | 1.06
300 | 130 | 200 |   | 0 |   | 1.1

Some observations
* the minimum value that is available for overload is 0.8
* the overload is clearly divided by the reference mass
* the vario doesn't allow a pilot below 40 kg or ballast of 40 kg
* setting reference weight = pilot + empty weight will fix the synchronization between xcsoar and s80/s100

lordfolken, 2024-01-07:
Rereading the S7 Manual:
[image]

So even if they call it minimum weight, they actually mean reference weight.

kobedegeest, 2024-01-08:
> I went through all the documentation i could find when i researched [#1319 (comment)](https://github.com/XCSoar/XCSoar/issues/1319#issuecomment-1872533910)
> 
> The overload factor is always divided by the reference_mass, for all mentioned products, except for the vaulter driver. So currently broken are the following drivers:
> 
> * lxnav
> * lxnavigation
> * OpenVario
> 
> According to the vaulter manual, its the only one to use dry mass. ![image](https://private-user-images.githubusercontent.com/407371/294479347-64f29ce4-60a9-4fcc-abef-4030c3015034.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ3MjY0MjQsIm5iZiI6MTcwNDcyNjEyNCwicGF0aCI6Ii80MDczNzEvMjk0NDc5MzQ3LTY0ZjI5Y2U0LTYwYTktNGZjYy1hYmVmLTQwMzBjMzAxNTAzNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwOFQxNTAyMDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kZmM2ODVlMDFlOGE4NzE2ZjdmMjk0NGI5NmVhMmEwYmMwZjYzM2RiODZhMWEzMGFiNTdlNmMwN2JlYzlkNzRkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.lrtvTB4DR12Bib4u9StFYRZLc8yl4A0r3PHneNRsOy8)

I don't think vaulter would break if the formula gets changed in XCSoar. On page 27 they give an example of how the polar in vaulter works (hardly a polar rather best LD): 
[image]

So how I understand if you set the V_LD and LD of your glider at reference mass of your polar used by XCSoar. Then the overload factor defined by LXnav would be the correct value to transmit. The manual also does not say that that value need to be more then 100% (but of course that is no guarantee). Current users with working setup would have to go and change their settings but they would have to do this every time the pilot mass changes anyway with current implementation.

lordfolken, 2024-01-08:
So yes, all the weights have to be entered manually on both sides. 

The Vaulter vario uses the following command to set the overload factor:
```
$PITV1,WL=1.35
```
According to the Protocol docs: "Set the wing loading factor (ratio of no-ballast to weight)"

lordfolken, 2024-01-08:
Oh, the fix for vaulter is easy. I use the fraction and add +1.

kobedegeest, 2024-01-09:
> Oh, the fix for vaulter is easy. I use the fraction and add +1.

Isn't fraction defined as ballast/max_ballast? At least according to the documentation of the CAI302 that is what fraction is defined as. How can fraction +1 be correct for vaulter? Unless for the rare case that you can exactly double your mass with water this will not give the correct value. Vaulter has no knowledge of max_ballast. It only knows best_LD and V_LD at some mass which it takes as 100%. If you are 35% heavier than this mass you send:
`$PITV1,WL=1.35`

What the user decides to be this 100% seems (soly based on the manual) not to strictly matter and then reference mass of the polar seems logical (since dry mass is not a constant as it changes based on the pilot).  And if this weight/ wingloading factor cannot be smaller than 100% it is similar to S3 and S7 varios and you have to use min take off mass.

lordfolken, 2024-01-09:
Your correct, as it isn't so simple. 

Vaulter vario's only know 1 parameter. Weight. Its the current weight of the glider including the pilot. 

 $PITV1,WL=1.00
Would be that weight without ballast.

So 
Pilot + Empty Plane + Ballast / pilot + empty plane
120 + 200 + 100 / 320 =  1.31

The S3 / S7 varios call it "minimum weight" incorrectly. They also mean reference mass, as per the description in the manual. So there is no difference between the LX S80 / S100 / S7 / S3 or LXX000 Devices.

lordfolken, 2024-01-09:
So basically, using the overload as newly defined is correct, and no special provision is needed for vaulter.

kobedegeest, 2024-01-10:
Out of curiosity and on the topic of syncing ballast. Is the polar info automatically synced between XCSoar and LX nav (like is the case for CAI302 for example) or does the user need to set this by himself manually? As currently XCSoar and LX nav input the polar data in a different way checking if both match is not as simple as checking if the numbers match. LX nav uses a,b,c coefs of the quadratic approximation and XCSoar uses 3 data points on the polar. (LXNav uses -W = a/10 000 *V^2 + b/100 *V + c, as a formula.)

It is also not documented anywhere, at the moment, that both devices need to be using a polar calculated at the same reference mass for ballast to be synced, which is a non trivial requirement imposed by how LXnav receives this data.

lordfolken, 2024-04-23:
So what is this then?
https://github.com/XCSoar/XCSoar/blob/258b352ecc99dbb61ca9fef53e5865f0031f50b5/src/Engine/GlideSolvers/GlidePolar.hpp#L76-L77

The reference mass is  the mass at which the glide polar was measured at. 
This includes the sum of an unknown empty weight of a glider with and unknown weight of a pilot.

lordfolken, 2024-04-23:
This is dead code. Its used in only one place.

lordfolken, 2024-04-27:
> So back then it did use empty mass but not anymore i think. (The code documentation should also say "ratio of **max** ballast to glider empty weight" to be correct)

What do you do with this info? 
 * you cannot use it for the ballast timer, as you cannot also drop the pilot
 * using reference mass is also wrong, as that is a mass that only exists for the polar and isn't based on current reality.
 * you cannot even use this info for the vaulter driver, as its refereeing to the "gross weight" being 100% and the water ballast the added percentage.  (so 100% = empty plane + pilot)

Also stuff like this scares me:
https://github.com/XCSoar/XCSoar/blob/9c217a0941c213e0d65d32c94113b6ff5a7a44c2/src/Engine/GlideSolvers/GlidePolar.cpp#L28

All of this just happens to work, because in the unit test, reference mass was set to the same as drymass:
https://github.com/XCSoar/XCSoar/blob/9c217a0941c213e0d65d32c94113b6ff5a7a44c2/test/src/TestGlidePolar.cpp#L32-L35

@kobedegeest i updated my pr in overload2, to address some of this. Of course the unit tests fail at the moment.

lordfolken, 2024-04-27:
This is the "default polar". 
https://github.com/XCSoar/XCSoar/blob/9c217a0941c213e0d65d32c94113b6ff5a7a44c2/src/Polar/PolarStore.cpp#L42
Imo .51 min sink? Did someone use the 18m polar, and rename it to 15m? Not even the manufacturer is that optimistic.

lordfolken, 2024-04-27:
Can't be right:
https://github.com/XCSoar/XCSoar/blob/9c217a0941c213e0d65d32c94113b6ff5a7a44c2/src/Polar/PolarStore.cpp#L179-L180

lordfolken, 2024-04-27:
> So just to fill in my gap in knowledge how does unit test work? How does it know the formulas are wrong/correct?

In short: You set the inputs to certain values, let the code run, and compare the outputs to what you (manually predetermined) what they should be. 
Should the results not match, or any part of the code influence that, you know that your change broke something. 
Theory says, that every-time you have bug, you should write a unit-test running on the pipeline to prevent that bug from happening again.
[A better more detailed explanation.](https://stackoverflow.com/questions/652292/what-is-unit-testing-and-how-do-you-do-it)