XCSoar/XCSoar #1321: Crash in task wind calculation - Angle::AsBearing() assertion (v7.40, Android)

xwhatsit, 2023-12-27:
XCSoar versions having and not having the problem
-------------------------------------------------

Debug build of v7.40 has the problem. Suspect I had the same issue in v7.24 (the only two versions I've been able to use). Hard to bisect as cannot (?) reproduce in sim.

System information
------------------

Android 13 phone, arm64. Self-build from docker image.

Using phone's internal GPS sensors, along with external BLE variometer (OpenVario driver).

Wind setting is from circling only (no pitot tube).

Steps to reproduce the behavior
-------------------------------

XCSoar crashes and closes after a few minutes thermalling in repeated tight circles in light/nil wind. Aircraft is a hang glider, so low speed (~40km/h circling speed) and small circles; would assume relatively few GPS datapoints per 360°. Haven't been able to reproduce it during simulation mode yet.

Do you have any idea what may have caused this?
-----------------------------------------------

adb logcat shows the following:

```
12-27 13:46:39.717 12582 12616 F libc    : src/Math/Angle.cpp:59: Angle Angle::AsBearing() const: assertion "fabs(value) < 100 * FullCircle().Native()" failed
12-27 13:46:39.718 12582 12616 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 12616 (Thread-5), pid 12582 (org.xcsoar)
12-27 13:46:39.974 13076 13076 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
12-27 13:46:39.974 13076 13076 F DEBUG   : Build fingerprint: 'samsung/a52sxqxx/a52sxq:13/TP1A.220624.014/A528BXXS5EWJ1:user/release-keys'
12-27 13:46:39.974 13076 13076 F DEBUG   : Revision: '4'
12-27 13:46:39.974 13076 13076 F DEBUG   : ABI: 'arm64'
12-27 13:46:39.974 13076 13076 F DEBUG   : Processor: '6'
12-27 13:46:39.974 13076 13076 F DEBUG   : Timestamp: 2023-12-27 13:46:39.802622334+1300
12-27 13:46:39.974 13076 13076 F DEBUG   : Process uptime: 126s
12-27 13:46:39.974 13076 13076 F DEBUG   : Cmdline: org.xcsoar
12-27 13:46:39.974 13076 13076 F DEBUG   : pid: 12582, tid: 12616, name: Thread-5  >>> org.xcsoar <<<
12-27 13:46:39.974 13076 13076 F DEBUG   : uid: 10402
12-27 13:46:39.974 13076 13076 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
12-27 13:46:39.974 13076 13076 F DEBUG   : Abort message: 'src/Math/Angle.cpp:59: Angle Angle::AsBearing() const: assertion "fabs(value) < 100 * FullCircle().Native()" failed'
12-27 13:46:39.974 13076 13076 F DEBUG   :     x0  0000000000000000  x1  0000000000003148  x2  0000000000000006  x3  0000007022ccd0e0
12-27 13:46:39.974 13076 13076 F DEBUG   :     x4  0000008080808080  x5  0000008080808080  x6  0000008080808080  x7  8080808080000000
12-27 13:46:39.974 13076 13076 F DEBUG   :     x8  00000000000000f0  x9  00000073575dab60  x10 0000000000000001  x11 000000735761a6f0
12-27 13:46:39.974 13076 13076 F DEBUG   :     x12 0101010101010101  x13 000034863618004b  x14 002a61afb298ea9f  x15 0000000000000060
12-27 13:46:39.974 13076 13076 F DEBUG   :     x16 0000007357682d70  x17 000000735765e430  x18 000000701eb04000  x19 0000000000003126
12-27 13:46:39.974 13076 13076 F DEBUG   :     x20 0000000000003148  x21 00000000ffffffff  x22 0000007022ccd288  x23 b4000071aec596d0
12-27 13:46:39.974 13076 13076 F DEBUG   :     x24 0000000000000110  x25 b40000713eacdb90  x26 b400007340daf878  x27 0000000000000001
12-27 13:46:39.974 13076 13076 F DEBUG   :     x28 0000000000000000  x29 0000007022ccd160
12-27 13:46:39.974 13076 13076 F DEBUG   :     lr  000000735760b628  sp  0000007022ccd0c0  pc  000000735760b654  pst 0000000000001000
12-27 13:46:39.974 13076 13076 F DEBUG   : backtrace:
12-27 13:46:39.974 13076 13076 F DEBUG   :       #00 pc 0000000000052654  /apex/com.android.runtime/lib64/bionic/libc.so (abort+168) (BuildId: ef11d8d2511bfd3cab1588a6cb2014bb)
12-27 13:46:39.974 13076 13076 F DEBUG   :       #01 pc 0000000000052a4c  /apex/com.android.runtime/lib64/bionic/libc.so (__assert2+40) (BuildId: ef11d8d2511bfd3cab1588a6cb2014bb)
12-27 13:46:39.974 13076 13076 F DEBUG   :       #02 pc 000000000075124c  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #03 pc 0000000000751378  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #04 pc 00000000006d9dc0  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #05 pc 00000000006c2f74  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #06 pc 00000000006c1ac0  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #07 pc 00000000006acf88  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #08 pc 00000000006a9f6c  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #09 pc 00000000006aa1dc  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #10 pc 00000000006a8dfc  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #11 pc 000000000069343c  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #12 pc 000000000068ce50  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #13 pc 000000000053249c  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #14 pc 0000000000747430  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #15 pc 00000000007468f4  /data/app/~~zrvgzGohn9-Pt4J5L1d3cg==/org.xcsoar-TVZ-aN_nD5Yr8KJnzfKnXQ==/lib/arm64/libxcsoar.so
12-27 13:46:39.974 13076 13076 F DEBUG   :       #16 pc 00000000000ba5a8  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+208) (BuildId: ef11d8d2511bfd3cab1588a6cb2014bb)
12-27 13:46:39.974 13076 13076 F DEBUG   :       #17 pc 0000000000053e9c  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+68) (BuildId: ef11d8d2511bfd3cab1588a6cb2014bb)
```

addr2line gives me the following locations in the backtrace:

 - Frame 02: /opt/xcsoar/src/Math/Angle.cpp:59
 - Frame 03: /opt/xcsoar/src/Math/Angle.cpp:109
 - Frame 04: /opt/xcsoar/src/Engine/GlideSolvers/GlideState.cpp:67
 - Frame 05: /opt/xcsoar/src/Engine/Task/Solvers/TaskSolution.cpp:67
 - Frame 06: /opt/xcsoar/src/Engine/Task/Solvers/TaskMacCready.cpp:23
 - Frame 07: /opt/xcsoar/src/Engine/Task/Ordered/OrderedTask.cpp:914
 - Frame 08: /opt/xcsoar/src/Engine/Task/AbstractTask.cpp:191
 - Frame 09: /opt/xcsoar/src/Engine/Task/AbstractTask.cpp:226
 - Frame 10: /opt/xcsoar/src/Engine/Task/TaskManager.cpp:261
 - Frame 11: /opt/xcsoar/src/Computer/TaskComputer.cpp:62
 - Frame 12: /opt/xcsoar/src/Computer/GlideComputer.cpp:87
 - Frame 13: /opt/xcsoar/src/CalculationThread.cpp:93
 - Frame 14: /opt/xcsoar/src/thread/WorkerThread.cpp:55
 - Frame 15: /opt/xcsoar/src/thread/Thread.cpp:118

Frame 04 has the effective wind angle calculation code:
```C++
    effective_wind_angle = wind.bearing.Reciprocal() - vector.bearing;
```

Unfortunately at this time I don't have an xcsoar.log; I launched XCSoar again to check some settings and xcsoar-old.log was replaced with a non-crashing log. However, when previously viewing this it is a "normal" log up until the point where the same assert fails ('src/Math/Angle.cpp:59: Angle Angle::AsBearing() const: assertion "fabs(value) < 100 * FullCircle().Native()" failed').

If this is related to the comment in Angle::AsBearing():
```

MaxKellermann, 2023-12-28:
This is a very old bug which, as you read in my code comment, has so far resisted any attempt to reproduce it with a debugger. Your bug report is useful, but unfortunately it is not enough information to really fix it.

xwhatsit, 2023-12-28:
Thanks, no problem. I will close this and run a release build from now on :)

Before I close it, I'll attach a little more information I noticed when looking at the tracklogs. I'm not sure it adds much but I've debugged enough painful issues "in the field" without gdb to know that occasionally some random fact will trigger someone's imagination:

 - I was simultaneously running XCTrack on the same device, which also uses the internal GPS; as such I have a "full tracklog" to compare
 - Comparing the tracklogs, just before XCSoar crashed each time there would be a very low number of points on that circle; maybe only 3 or 4 points
 - This is maybe interesting as:
     - I would assume that the GPS sensor continued to output new updated positions that XCSoar didn't capture at the same update rate it was using earlier, and
     - Maybe the angle calculation code is more likely to produce NaNs if there are sudden sharp angles or reversing in direction?

It does not always crash when the circles "go triangular" as you can see in the last screenshot. I also assume (can't check as it's sadly closed-source) that XCTrack may be applying some filtering to the datapoints as they are not absolutely identical.

Obviously the datapoint that caused the abort is not logged. Nonetheless see attachments if this is of any use in future (these also include the waypoints for the planned task which may be relevant).

Thanks once again, maybe this is of use one day to someone.

[image]
[image]
[image]

IGCs attached as zip bundle, as Github does not allow direct attachment:

[2023-12-27_tracklogs.zip](https://github.com/XCSoar/XCSoar/files/13789905/2023-12-27_tracklogs.zip)

closed 2023-12-28 by xwhatsit