XCSoar/XCSoar #1322: XCSoar connectivity to LXNAV S100 intermittent

bbaer1955, 2024-01-02:
XCSoar versions having and not having the problem
-------------------------------------------------

 N/A

System information
------------------

Steps to reproduce the behavior
-------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

bbaer1955, 2024-01-02:
Apologies, it seems I messed up the template above .
XCSoar V 7.41
Galaxy A51 Android 13

I successfully paired my new S100 with my Samsung Galaxy A51 running XCSoar, but it has now failed. In XCSoar /Devices I see error message:

read failed, socket might closed or timeout, read ret: -1

Yes, Bluetooth is enabled in the S100, and the antenna is seated firmly.
When it was working I managed to download an igc file via XCSoar and also sent a task to the S100.

So I cleared Bluetooth cache, unpaired the devices, then paired the devices again and bingo it worked again. However today the Galaxy shut down on the grid due to overheating and now I get the error message above again. I unpaired again, but now can't get the phone to see the S100. 

This problem is intermittent; sometimes it allows me to downlad .igc files, then inext time it will fail,
Connectivity to the S100 in general use also fails for no apparent reason occasionally also.
Another LXNAV device (Flarm Power Mouse) seems to have rock solid BT connectivity.

bbaer1955, 2024-01-03:
This morning I found another mode of failure. I had the S100 BT working and in XCSoar under config/devices selected it then selected the 'manage'  option. Then once in manage I selected the 'edit' option (or some similar name) and then it failed with the
same error message as above. I turned BT off, XCSoar off and S100 off and went away for an hour. When I came back I turned them all on again and once again it was working. I went into /manage/edit again and this time it did NOT fail so I left it at that.

MaxKellermann, 2024-01-03:
I cannot reproduce your problem because I don't have an S100. At this point, a logcat would be useful to see more verbose error messages.

bbaer1955, 2024-01-04:
Three logcat's from 3 different faiIures this morning. I also ran debug in Devices but am so far unable to find where these logs were stored! Also, my phone does NOT enable root access so not sure how useful the logcats will be. HTH.
[2024-01-04_09.19.zip](https://github.com/XCSoar/XCSoar/files/13826417/2024-01-04_09.19.zip)
[2024-01-04_09.22.zip](https://github.com/XCSoar/XCSoar/files/13826418/2024-01-04_09.22.zip)
[2024-01-04_09.28.zip](https://github.com/XCSoar/XCSoar/files/13826419/2024-01-04_09.28.zip)

bbaer1955, 2024-01-04:
Here is the log:
[xcsoar.log](https://github.com/XCSoar/XCSoar/files/13826455/xcsoar.log)

bbaer1955, 2024-01-04:
I installed ADB on my Win10 laptop and enabled READ-LOG access to SysLog app via ADB; if you need me to gather logcat(s) again. I'm using SysLog app on my Android phone to gather logcat.

MaxKellermann, 2024-01-04:
The xcsoar.log is useless because it only contains what you already see inside the app; it doesn't contain any further information because XCSoar doesn't have it and thus can't log it.

Your logcats are useless because you weren't using XCSoar and didn't trigger the problem.

MaxKellermann, 2024-02-28:
In your logcat, I found the error message you quoted:
```
01-06 09:14:35.274 W/ObexServerSockets19(14856): Accept exception for ServerSocket: Type: TYPE_L2CAP Channel: 4103
01-06 09:14:35.274 W/ObexServerSockets19(14856): java.io.IOException: read failed, socket might closed or timeout, read ret: -1
01-06 09:14:35.274 W/ObexServerSockets19(14856):        at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:952)
01-06 09:14:35.274 W/ObexServerSockets19(14856):        at android.bluetooth.BluetoothSocket.waitSocketSignal(BluetoothSocket.java:905)
01-06 09:14:35.274 W/ObexServerSockets19(14856):        at android.bluetooth.BluetoothSocket.accept(BluetoothSocket.java:664)
01-06 09:14:35.274 W/ObexServerSockets19(14856):        at android.bluetooth.BluetoothServerSocket.accept(BluetoothServerSocket.java:180)
01-06 09:14:35.274 W/ObexServerSockets19(14856):        at android.bluetooth.BluetoothServerSocket.accept(BluetoothServerSocket.java:165)
01-06 09:14:35.274 W/ObexServerSockets19(14856):        at com.android.bluetooth.ObexServerSockets$SocketAcceptThread.run(ObexServerSockets.java:345)
01-06 09:14:35.274 W/bt_stack(14856): le_scanning_manager :: OnResume() - call AckResume
01-06 09:14:35.274 D/ObexServerSockets19(14856): shutdown(block = false)
01-06 09:14:35.274 W/bt_stack(14856): le_address_manager :: ack_resume()
01-06 09:14:35.275 W/ObexServerSockets18(14856): Accept exception for ServerSocket: Type: TYPE_RFCOMM Channel: 19
01-06 09:14:35.275 W/ObexServerSockets18(14856): java.io.IOException: read failed, socket might closed or timeout, read ret: -1
01-06 09:14:35.275 W/ObexServerSockets18(14856):        at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:952)
01-06 09:14:35.275 W/ObexServerSockets18(14856):        at android.bluetooth.BluetoothSocket.waitSocketSignal(BluetoothSocket.java:905)
01-06 09:14:35.275 W/ObexServerSockets18(14856):        at android.bluetooth.BluetoothSocket.accept(BluetoothSocket.java:664)
01-06 09:14:35.275 W/ObexServerSockets18(14856):        at android.bluetooth.BluetoothServerSocket.accept(BluetoothServerSocket.java:180)
01-06 09:14:35.275 W/ObexServerSockets18(14856):        at android.bluetooth.BluetoothServerSocket.accept(BluetoothServerSocket.java:165)
01-06 09:14:35.275 W/ObexServerSockets18(14856):        at com.android.bluetooth.ObexServerSockets$SocketAcceptThread.run(ObexServerSockets.java:345)
```
But it was from an Android system process. This is outside of XCSoar's control, and thus XCSoar can't be changed to avoid your  problem between the Android operating system and the S100. Sorry.

closed 2024-02-28 by MaxKellermann