XCSoar/XCSoar #1323: Saved tasks show as empty tasks when loading

paynterf, 2024-01-04:
XCSoar versions having and not having the problem
-------------------------------------------------

New user, so don't know

System information
------------------

Steps to reproduce the behavior
-------------------------------
Open XCSoar
Create a task
Close XCSoar (this is supposed to save the task)
Open XCSoar
Use Nav->Manage -> Manage -> Browse to show a list of tasks
Click on the name of the task just saved
See 'Task is empty (FAI badges/records)'

Note:  I could find the task file in the XCSoarData folder, and it looks like a valid XML structure, but all the 'name=' fields just have '1' as a value - not the correct waypoint names.

------------------------------- TASK FILE ON DISK ------------------------------------
<Task pev_start_window="0" pev_start_wait_time="0" fai_finish="0" finish_min_height_ref="1" finish_min_height="30" start_max_height_ref="1" start_max_height="5100" start_max_speed="46.3" start_score_exit="1" start_requires_arm="0" aat_min_time="3000" type="1">
	<Point type="1">
		<Waypoint altitude="687" comment="1" id="238" name="1">
			<Location latitude="46.3823" longitude="14.2162"/>
		</Waypoint>
		<ObservationZone length="1000" type="1"/>
	</Point>
	<Point type="1">
		<Waypoint altitude="1788" comment="1" id="110" name="1">
			<Location latitude="46.4964" longitude="13.8883"/>
		</Waypoint>
		<ObservationZone type="1"/>
	</Point>
	<Point type="1">
		<Waypoint altitude="1414" comment="1" id="87" name="1">
			<Location latitude="46.3247" longitude="14.4642"/>
		</Waypoint>
		<ObservationZone type="1"/>
	</Point>
	<Point type="1">
		<Waypoint altitude="502" comment="1" id="1" name="1">
			<Location latitude="46.3566" longitude="14.1752"/>
		</Waypoint>
		<ObservationZone type="1"/>
	</Point>
</Task>
------------------------------- TASK FILE ON DISK ------------------------------------

paynterf, 2024-01-04:
Hmm, I cut/pasted the contents of the saved task, but it doesn't show up in the submitted issue.  I have (hopefully) attached the file here instead
[another - Copy.txt](https://github.com/XCSoar/XCSoar/files/13824991/another.-.Copy.txt)

MaxKellermann, 2024-01-04:
You don't even know the XCSoar version number, huh??

paynterf, 2024-01-04:
Hmm, I did put that info in the original submission, but must have cut it off before sending it.  The version is 7.40-PC.

The 'New user, so don't know' comment related to the "XCSoar versions having and not having the problem" part of the submission template.  Since 7.40-PC was the only version I tried, I didn't know what earlier versions might or might not have the issue.

I would rather have posted this to the XCSoar forum than GitHub, because I'm almost certain this is a user problem rather than a code issue.  Unfortunately I can't post to XCSoar as my original account from several years ago has been blocked for some reason, and a new account I created for user 'TA1' is still inactive after a week. Your name was mentioned as someone who could address that problem; if you could I would be ever so grateful.

TA

lordfolken, 2024-02-09:
@paynterf is this still an issue?
If you suspect a bug, then you should try multiple versions from https://download.xcsoar.org/releases/

P.s.: Forum registration should be possible again, even with large provider email (gmail,t-offline, etc.)

lordfolken, 2024-02-09:
Actually i'm closing this. if its still an issue, please reopen.

paynterf, 2024-02-09:
I haven't gone back and checked - but I will do so.  BTW, thanks for fixing
the registration issue.  Would you mind fixing the current problem with the
forum, in that it appears that nobody can post new topics or reply to old
ones?

TIA,

Frank

lordfolken, 2024-02-09:
created issue #1344

lordfolken, 2024-02-19:
This is still an issue. #1346

closed 2024-02-19 by lordfolken