XCSoar/XCSoar #1325: .CUP file with 499 or more lines fails to load

paynterf, 2024-01-07:
System information
------------------

Steps to reproduce the behavior
-------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

[AA2_truncated.zip](https://github.com/XCSoar/XCSoar/files/13855045/AA2_truncated.zip)

lordfolken, 2024-01-08:
I cannot reproduce on current head Linux.
Have you chosen a filter, such as a name or similar?

lordfolken, 2024-01-08:
https://github.com/XCSoar/XCSoar/blob/4320c10145507e3b9b6dcdc7bd45c57efa28e49d/src/Dialogs/Waypoint/WaypointList.cpp#L256-L257

paynterf, 2024-01-08:
Yep - that certainly would do the trick! 🤪

Any thoughts about providing an error message for us poor users?

Or maybe looking at the reason or the 500 limit? I can't imagine that
modern tablets would have a problem with 5000 lines, much less 500.

lordfolken, 2024-01-08:
And now the users want error messages. What has the world come to?  ;-)

The 500 limit is more a human performance thing, than technological. 
If you had to infinitely scroll in a bouncy cockpit.. 

You made me discover another issue though #1328 :/

closed 2024-01-08 by lordfolken

paynterf, 2024-01-08:
And now the users want error messages. What has the world come to? ;-)

The horrors! ;)

While I've got your ear <eye>  I don't suppose  you could tell me why I
can't log onto or re-register to the XCSoar forum?

The expected validation email message never arrives (and I've tried three
different username/email address combinations).

I'm pretty sure this isn't your bailiwick, but I'm hoping you know whose it
is ;)

Regards,

Frank