XCSoar/XCSoar #1328: XCSoar cannot load cup files, larger than 1000 wp

lordfolken, 2024-01-08:
XCSoar versions having and not having the problem
-------------------------------------------------

First bad commit: 

81fb9ce1de0b808341c2472dddfe8b3d1f56c281

System information
------------------

UNIX target

Steps to reproduce the behavior
-------------------------------
Start with clean .xcsoar directory
Load a large wp file, for example 
DE-WPT-National-XCSoar.cup

Expected behavior
-----------------

Load the file

Actual behavior
---------------

Debug build crash with assertion failure:

Starting XCSoar XCSoar v7.40-Linux~git#8c76ec37e4
GLX config: RGB=8/8/8 alpha=8 depth=24 stencil=8
GL vendor: AMD
GL version: 4.6 (Compatibility Profile) Mesa 22.3.6
GL renderer: AMD Radeon Graphics (renoir, LLVM 15.0.6, DRM 3.49, 6.1.0-13-amd64)
GL extensions: GL_ARB_multisample GL_EXT_abgr GL_EXT_bgra GL_EXT_blend_color GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_copy_texture GL_EXT_subtexture GL_EXT_texture_object GL_EXT_vertex_array GL_EXT_compiled_vertex_array GL_EXT_texture GL_EXT_texture3D GL_IBM_rasterpos_clip GL_ARB_point_parameters GL_EXT_draw_range_elements GL_EXT_packed_pixels GL_EXT_point_parameters GL_EXT_rescale_normal GL_EXT_separate_specular_color GL_EXT_texture_edge_clamp GL_SGIS_generate_mipmap GL_SGIS_texture_border_clamp GL_SGIS_texture_edge_clamp GL_SGIS_texture_lod GL_ARB_framebuffer_sRGB GL_ARB_multitexture GL_EXT_framebuffer_sRGB GL_IBM_multimode_draw_arrays GL_IBM_texture_mirrored_repeat GL_ARB_texture_cube_map GL_ARB_texture_env_add GL_ARB_transpose_matrix GL_EXT_blend_func_separate GL_EXT_fog_coord GL_EXT_multi_draw_arrays GL_EXT_secondary_color GL_EXT_texture_env_add GL_EXT_texture_filter_anisotropic GL_EXT_texture_lod_bias GL_INGR_blend_func_separate GL_NV_blend_square GL_NV_light_max_exponent GL_NV_texgen_reflection GL_NV_texture_env_combine4 GL_S3_s3tc GL_SUN_multi_draw_arrays GL_ARB_texture_border_clamp GL_ARB_texture_compression GL_EXT_framebuffer_object GL_EXT_texture_compression_s3tc GL_EXT_texture_env_combine GL_EXT_texture_env_dot3 GL_MESA_window_pos GL_NV_packed_depth_stencil GL_NV_texture_rectangle GL_ARB_depth_texture GL_ARB_occlusion_query GL_ARB_shadow GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_mirrored_repeat GL_ARB_window_pos GL_ATI_fragment_shader GL_EXT_stencil_two_side GL_EXT_texture_cube_map GL_NV_copy_depth_to_color GL_NV_depth_clamp GL_NV_fog_distance GL_NV_half_float GL_APPLE_packed_pixels GL_ARB_draw_buffers GL_ARB_fragment_program GL_ARB_fragment_shader GL_ARB_shader_objects GL_ARB_vertex_program GL_ARB_vertex_shader GL_ATI_draw_buffers GL_ATI_texture_env_combine3 GL_ATI_texture_float GL_EXT_depth_bounds_test GL_EXT_shadow_funcs GL_EXT_stencil_wrap GL_MESA_pack_invert GL_NV_primitive_restart GL_ARB_depth_clamp GL_ARB_fragment_program_shadow GL_ARB_half_float_pixel GL_ARB_occlusion_query2 GL_ARB_point_sprite GL_ARB_shading_language_100 GL_ARB_sync GL_ARB_texture_non_power_of_two GL_ARB_vertex_buffer_object GL_ATI_blend_equation_separate GL_EXT_blend_equation_separate GL_OES_read_format GL_ARB_color_buffer_float GL_ARB_pixel_buffer_object GL_ARB_texture_compression_rgtc GL_ARB_texture_float GL_ARB_texture_rectangle GL_ATI_texture_compression_3dc GL_EXT_packed_float GL_EXT_pixel_buffer_object GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_mirror_clamp GL_EXT_texture_rectangle GL_EXT_texture_sRGB GL_EXT_texture_shared_exponent GL_ARB_framebuffer_object GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXT_packed_depth_stencil GL_ARB_vertex_array_object GL_ATI_separate_stencil GL_ATI_texture_mirror_once GL_EXT_draw_buffers2 GL_EXT_draw_instanced GL_EXT_gpu_program_parameters GL_EXT_gpu_shader4 GL_EXT_texture_array GL_EXT_texture_compression_latc GL_EXT_texture_integer GL_EXT_texture_sRGB_decode GL_EXT_timer_query GL_OES_EGL_image GL_AMD_performance_monitor GL_EXT_texture_buffer_object GL_AMD_texture_texture4 GL_ARB_copy_buffer GL_ARB_depth_buffer_float GL_ARB_draw_instanced GL_ARB_half_float_vertex GL_ARB_instanced_arrays GL_ARB_map_buffer_range GL_ARB_texture_buffer_object GL_ARB_texture_rg GL_ARB_texture_swizzle GL_ARB_vertex_array_bgra GL_EXT_texture_swizzle GL_EXT_vertex_array_bgra GL_NV_conditional_render GL_AMD_conservative_depth GL_AMD_depth_clamp_separate GL_AMD_draw_buffers_blend GL_AMD_seamless_cubemap_per_texture GL_AMD_shader_stencil_export GL_ARB_ES2_compatibility GL_ARB_blend_func_extended GL_ARB_compatibility GL_ARB_debug_output GL_ARB_draw_buffers_blend GL_ARB_draw_elements_base_vertex GL_ARB_explicit_attrib_location GL_ARB_fragment_coord_conventions GL_ARB_provoking_vertex GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_shader_stencil_export GL_ARB_shader_texture_lod GL_ARB_tessellation_shader GL_ARB_texture_buffer_object_rgb32 GL_ARB_texture_cube_map_array GL_ARB_texture_g
Display dpi=96,96
OpenGL: mda=true npot=false stencil=0x0
Startup dialog
Loading profiles
Loaded profile from /home/folken/.xcsoar/default.prf
Loading language file
Loading input events file
Loading Topography File...
ReadWaypoints
/usr/include/c++/12/span:278: constexpr std::span<_Type, _Extent>::element_type& std::span<_Type, _Extent>::operator[](size_type) const [with _Type = std::byte; long unsigned int _Extent = 18446744073709551615; reference = std::byte&; size_type = long unsigned int]: Assertion '__idx < size()' failed.
Aborted

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2024-01-08:
@twhitehead any idea?

twhitehead, 2024-01-08:
I am having a look.

twhitehead, 2024-01-08:
My bad. I used
```
field_start = &input[input_offset - input_shift + field_quoted];
```
but this can technically be one-past the end when `input` (the current block of input read in) ends on an opening quote as the field starts one-past the end of the opening quote. Switching to
```
input.data() + input_offset - input_shift + field_quoted;
```
removes the implicit bounds check in using `[]` and fixes the issue.

twhitehead, 2024-01-08:
Updated the code comments too now (noting the special cases, like one past-the-end and zero-length).

closed 2024-01-19 by MaxKellermann

twhitehead, 2024-03-06:
Can't be this issue as it was about a mistake in commit 81fb9ce1de0b808341c2472dddfe8b3d1f56c281, and that isn't is anything but the latest v7.42 (which also contains the fix)
```
$ git  tag --contains 81fb9ce1de0b808341c2472dddfe8b3d1f56c281
v7.42
```

Rensiepensie, 2024-03-07:
> Can't be this issue as it was about a mistake in commit [81fb9ce](https://github.com/XCSoar/XCSoar/commit/81fb9ce1de0b808341c2472dddfe8b3d1f56c281), and that isn't is anything but the latest v7.42 (which also contains the fix)
> 
> ```
> $ git  tag --contains 81fb9ce1de0b808341c2472dddfe8b3d1f56c281
> v7.42
> ```

Yes! v7.42 problem with my .dat files is fixed (also)

commit 18140d4c (2024-01-19): Use data() instead of &operator[] as can be one-past end (fix #1328)

The field_start can be one-past-the-end if the end is an opening
quote. This case causes &operater[] to choke on the bounds check.