XCSoar/XCSoar #1330: Display of multiple images= with details.txt

llauner, 2024-01-09:
XCSoar versions having and not having the problem
-------------------------------------------------
Which XCSoar version are you using when seeing the problem? 7.40pc + 7.41 Android
Which XCSoar version is the last one you know that did not show the problem? Never tried with earlier version

System information
------------------
Android 11
Android 9

Steps to reproduce the behavior
-------------------------------

Use a waypoint cup file.
Create a waypoint details.txt file including at least 1 waypoint with multiple images
[my_waypoint]
image=Pics/1.jpg
image=Pics/2.jpg

Expected behavior
-----------------

When looking at a waypoint details, the first image appears. When click on the right arrow, the second image should appear.
Going further right would display the rest of the details. Going backward should take you to the pages in reverse order.

Actual behavior
---------------

Second image does not appear. It appears if you start going "backward" by clicking on the left arrow.

Do you have any idea what may have caused this?
-----------------------------------------------

No

Do you have an idea how to solve the issue?
-------------------------------------------

Navigation problem in the details pages ?

ubx, 2024-01-09:
Could not reproduce the problem. Here is my cup file:

```
﻿name,code,country,lat,lon,elev,style,rwdir,rwlen,rwwidth,freq,desc,userdata,pics
Arquata (Kat B 2023),ARQU4707,IT,4244.383N,01316.374E,770m,3,70.0,220m,45m,,"Vorsicht! Feld, Breite 45m, Leicht ansteigend",,1_4707_osm.jpg;2_2164.jpg
```
and the waypoint details file:

```
[ARQU4707]
image=pics/1_4707_osm.jpg
image=pics/2_2164.jpg
```

llauner, 2024-01-09:
Well..that's very embarrassing as now I can't reproduce it either !
I'll try to make further investigation and I'll come back to you if I find out what was causing this problem.

Thanks !

closed 2024-01-09 by llauner