XCSoar/XCSoar #1332: XCSoar 7.40 @ Kobo: Iso8859-1 Charset forWaypoint

RolfWilhelm, 2024-01-19:
XCSoar versions having and not having the problem
-------------------------------------------------

 7.23 @ Kobo did not show the problem. 7.40 @ PC also does not show the problem
7.39@ Kobo: Installer was currently not available at https://download.xcsoar.org/releases/7.39/KOBO/KoboRoot.tgz

System information
------------------

Kobo Touch 2.0 /  Kobo Clara HD

Steps to reproduce the behavior
-------------------------------

XCSoar.cup shows the problem XCSoar.cup does not (charset corrected)
[XCSoar.cup.txt](https://github.com/XCSoar/XCSoar/files/13987502/XCSoar.cup.txt)
[XCSoar2.cup.txt](https://github.com/XCSoar/XCSoar/files/13987503/XCSoar2.cup.txt)
[XCSoar2.cup charset Diff.txt](https://github.com/XCSoar/XCSoar/files/13988151/XCSoar2.cup.charset.Diff.txt)

Expected behavior
-----------------

Reading the cup correctly file as before and within the PC build

Actual behavior

---------------

The problem: Reading the waypoint list interrupts at the first Iso8859-1 char. 
The Waypoint list consists of the already read items plus the waypoints of the current task.
The home airfield is not set (even if available in the current task)
 

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------
Enabling the Iso8859-1 charset in the next kobo build

MaxKellermann, 2024-01-19:
Was this fixed by 227840b644872fee3f445f6e6ec1c56349d04b62 ?

RolfWilhelm, 2024-01-19:
Ich denk mal nicht.

Was mich am meisten wundert ist, daß es in der PC-Version funktioniert.
In der Kobo-Version 7.23 war das auch ok.
Durch Wandlung des Charsets war das Problem beseitigt.

Zugriff auf die Vorgänger-Version 7.39 etc. habe ich zur Zeit keine.

Gruß
Rolf

Von: Max Kellermann
Gesendet: Freitag, 19. Januar 2024 12:12
An: XCSoar/XCSoar
Cc: RolfWilhelm; Author
Betreff: Re: [XCSoar/XCSoar] XCSoar 7.40 @ Kobo: Iso8859-1 Charset forWaypoint(Issue #1332)

Was this fixed by 227840b ?

RolfWilhelm, 2024-01-19:
Kleiner Nachtrag:
Kobo Mini, Kobo Aura zeigen das gleiche Verhalten.

Auf meinem Samsung Android läuft die Version 7.41 und da wird auch beim 1. Iso8859-1 char das Lesen der Wegpunktliste abgebrochen

Gruß
Rolf
 
Gesendet: Freitag, 19. Januar 2024 12:12
An: XCSoar/XCSoar
Cc: RolfWilhelm; Author
Betreff: Re: [XCSoar/XCSoar] XCSoar 7.40 @ Kobo: Iso8859-1 Charset forWaypoint(Issue #1332)

Was this fixed by 227840b ?

RolfWilhelm, 2024-01-21:
Hallo Max

Ergänzung: Umwandlung in UTF-8 löst das Problem.
Es bleibt: Die PC-Version kann beides!

Ich kann mit allem Leben.

Gruß
Rolf

Von: Max Kellermann
Gesendet: Freitag, 19. Januar 2024 12:12
An: XCSoar/XCSoar
Cc: RolfWilhelm; Author
Betreff: Re: [XCSoar/XCSoar] XCSoar 7.40 @ Kobo: Iso8859-1 Charset forWaypoint(Issue #1332)

Was this fixed by 227840b ?

MaxKellermann, 2024-02-26:
> Ich denk mal nicht.

WTF!!

closed 2024-02-26 by MaxKellermann

RolfWilhelm, 2024-02-26:
Hallo Max,

für mich ist das kein Problem ich habe die Datei auf UTF-8 umgestellt.
Als XCSoar noch nicht UTF-8 konnte hatte ich explizit auf Iso8859-1 umgestellt (viele Jahre her).

Es gab keine Fehlermeldung!!
Aufgefallen ist es nur weil kein Heimatplatz mehr vorhanden war.
In der PC-Version funktioniert es weiterhin.

Falls es für niemand sonst ein Thema ist kannst du den Report auch schließen.

Gruß
Rolf

Von: Max Kellermann
Gesendet: Montag, 26. Februar 2024 13:44
An: XCSoar/XCSoar
Cc: RolfWilhelm; Author
Betreff: Re: [XCSoar/XCSoar] XCSoar 7.40 @ Kobo: Iso8859-1 Charset forWaypoint(Issue #1332)

Ich denk mal nicht.
WTF!!