XCSoar/XCSoar #1336: Version 7.x no longer accepts 'ft' in the waypoint elevation data in .cup files

johnmurtari, 2024-01-29:
XCSoar versions having and not having the problem
-------------------------------------------------
Which XCSoar version are you using when seeing the problem?  7.41
Which XCSoar version is the last one you know that did not show the problem?  6.x

System information
------------------
Samsung Android Tablet

Steps to reproduce the behavior
-------------------------------
Add a waypoint to a cup file and use 'ft' for elevation.  Use the waypoint in a task, no elevation data for details.

Expected behavior
-----------------
In the earlier versions, the correct elevation, in ft, was displayed and used.

Actual behavior
---------------
No elevation data is shown.

Do you have any idea what may have caused this?
-----------------------------------------------
It appears in a recent change only 'f' is accepted in the elevation field.

Do you have an idea how to solve the issue?
-------------------------------------------
It would be nice to allow backward compatibility so that 'ft' is accepted.

johnmurtari, 2024-01-29:
I had originally posted the question on the XCSoar forum, https://forum.xcsoar.org/viewtopic.php?f=2&t=4304 ,
also in the users mailing list.   A person in the users mailing list identified the issue:

 
It should be "1710f," not "1710ft,"
See ParseAltitude() in src/Waypoint/WaypointReaderSeeYou.cpp
However the issue https://github.com/XCSoar/XCSoar/issues/900 still exists.

Viele Grüße,
Ronald Niederhagen

lordfolken, 2024-01-29:
According to the standard the acceptable unit is m or ft. Mixing in the same file is allowed.

closed 2024-02-17 by lordfolken

johnmurtari, 2024-02-17:
Glad to hear it!  Many thanks for the quick work.
I know we're on 7.40 right now, any idea when it would be in a release?

commit af912705 (2024-02-17): Waypoint/WaypointReaderSeeYou: parse ft with 2 letter suffix

closes #1336