XCSoar/XCSoar #1339: xcsoar hosting - relocating from AWS

codexx666, 2024-01-30:
hi everyone,

currently the XCSoar website is running on a AWS instance.

It's running fine there now for some years, but the costs are approx 40 /.50 EUR per month.
Yes, I have looked into all saving options now.
I think there are cheaper options around and also some of these options offer more than the AWS instance.

Due to the pricey setup i want to relocate the website to another hosting environment - maybe a root server at NetCUP or IONOS.

But before that I'll need to now
- what do we actually need in respect of OS, performance, network, diskspace,... ???
- who has access to the DNS for xcsoar?
A relocation requires updating the IP to the new host

- what do we need to do to relocate the xcsoar website to a new host?

I really want to do this as quickly as possible.

Regards
Flip

lordfolken, 2024-01-30:
> what do we actually need in respect of OS, performance, network, diskspace,... ???
  * We currently have 1 GB of memory, 2 Cpus and 150GB of disk. 
  * we have about 1-2TB of transfer a month. 
  * Disk space would be good if its around 200-300GB
  * IPV6 address would be great (in addtion to ipv4)

The machine is very light weight. The website is static, the download folder aswell. The only thing dynamic is the mariadb for the forum, and mail service.
  
> who has access to the DNS for xcsoar?
> A relocation requires updating the IP to the new host
I haven´t  been able to find that out, and the original owners where non-responsive to my mails. @MaxKellermann do you have control of xcsoar.org?

> what do we need to do to relocate the xcsoar website to a new host?
# spin up a wm, perferrably debian
# add a few users and ssh keys
# rsync stuff between the two machines
# play ansible books which are in the root of the vm, which do 95% of the configuration
# switchover dns to the new vm

lordfolken, 2024-01-30:
> Why root server? Simply static shared hosting would suffice, wouldn't it?

apart from the forum, yes, static hosting would do the job.

lordfolken, 2024-01-30:
But honestly, a debian box i can manage... fightiing with VPS solutions is another matter.

codexx666, 2024-01-31:
I think a root server is preferable because it offers more flexibility. 
Also I would expect that most "static hosting" solutions don't offer SSH features or use of ansible.

Ok, I'll will check available root server solutions and check the pricing / features.
In regards of the current AWS VM specs and the requirements I assume that simple root server offers are more than sufficient.

example
https://www.netcup.de/bestellen/produkt.php?produkt=2892
8GB RAM, AMD EPYC™ 7702 , 160GB disk - approx 11EUR / month.
That's way less than AWS and way more power :-)

lordfolken, 2024-01-31:
That netcoup server would be good. Currently we have about 55GB Disk usage and i expect that to double.

MaxKellermann, 2024-01-31:
Downloads on cheap shared hosting, and the remaining traffic to be hosted somewhere else should suddenly be very small.

lordfolken, 2024-01-31:
Hetzner's shared hosting is 9.16 for shared hosting. https://www.hetzner.com/webhosting/level-9/
Netcup is 9,81€ for that server if you pay by 12months. 
I would prefeer the vm. 

But again, the show stopper is dns entries on xcsoar.org.

hjr, 2024-01-31:
I made good experience with the Hetzner's, would recommend them.

lordfolken, 2024-01-31:
Again I'd move it in a split second. But I need control of xcsoar.org.

codexx666, 2024-02-05:
I'll try to contact John W. - hopefully he can get us in contact with Scott for the DNS update

lordfolken, 2024-02-05:
Https://dd.com.au/contact

codexx666, 2024-02-07:
Ok, John has replied and has access to the DNS.

I will now have a look at some possible options (Netcup, Ionos, Hetzner, ... ?) and report back here.
Based on that info, we can make a decision where to move xcsoar and order the rootserver/vps/... 

next step
Deploy the website to the new environment, check everything is working fine.

final step
update the DNS entry and complete the move.

ok? ;-)

lordfolken, 2024-02-07:
Give me an account and ssh, I can do the deployment.

OBrown92, 2024-02-12:
I have had very good experiences with uberspace. Good docs and really good support, as well as very cheap and flexible.

lordfolken, 2024-02-12:
Ueberspace would be right for a nerd like me. 

However the maps currently are close to 60gb. This will increase with upcoming topology data.

Ueberspace is priced at 3/month Euro per 10gb increment. 10gb are included.

OBrown92, 2024-02-12:
Yes, that's right, the service is not designed for storage.

lordfolken, 2024-03-10:
@codexx666 hey there, any next steps that I need to initiate?

codexx666, 2024-04-03:
@lordfolken - new server is finally ready. I'll send you all the relevant info via email.

codexx666, 2024-05-06:
update:
the AWS instance will be terminated latest end of May

@lordfolken is working on the migration and so far the new VM is working fine.

lordfolken, 2024-06-01:
Content was moved on the 30TH. The old server has a port forward to the new one via http/https.

The DNS can be changed.
We might also want to remove the old mxserver in the the MX records.

There is also still the matter of https://cloud.xcsoar.net I guess that is still on a server from Max?

lordfolken, 2024-06-02:
please also see #1470

lordfolken, 2024-06-25:
cc: @jwharington @codexx666 (thx for the call today!) :-)

So the records have been changed to as follows:

```
xcsoar.org	A	Simple	-	No	45.142.179.218
	300	-	-	-
	xcsoar.org	AAAA	Simple	-	No	2a03:4000:4a:cf:1443:a0ff:fe58:2ee2
	300	-	-	-
	xcsoar.org	MX	Simple	-	No	10 mail.xcsoar.org.
	300	-	-	-
	xcsoar.org	NS	Simple	-	No	ns-896.awsdns-48.net.
ns-111.awsdns-13.com.
ns-1782.awsdns-30.co.uk.
ns-1362.awsdns-42.org.
	172800	-	-	-
	xcsoar.org	SOA	Simple	-	No	ns-896.awsdns-48.net. awsdns-hostmaster.amazon.com. 1 7200 900 1209600 86400
	900	-	-	-
	xcsoar.org	TXT	Simple	-	No	"v=spf1 a:mapgen.xcsoar.org ip4:45.142.179.218 ip6:2a03:4000:4a:cf:1443:a0ff:fe58:2ee2 ?all"
	300	-	-	-
	_dmarc.xcsoar.org	TXT	Simple	-	No	"v=DMARC1; p=reject; rua=mailto:dmarc@xcsoar.org; ruf=mailto:postmaster@xcsoar.org; fo=1"
	300	-	-	-
	bugs.xcsoar.org	A	Simple	-	No	45.142.179.218
	300	-	-	-
	bugs.xcsoar.org	AAAA	Simple	-	No	2a03:4000:4a:cf:1443:a0ff:fe58:2ee2
	300	-	-	-
	cloud.xcsoar.org	A	Simple	-	No	45.142.179.218
	300	-	-	-
	cloud.xcsoar.org	AAAA	Simple	-	No	2a03:4000:4a:cf:1443:a0ff:fe58:2ee2
	300	-	-	-
	download.xcsoar.org	A	Simple	-	No	45.142.179.218
	300	-	-	-
	download.xcsoar.org	AAAA	Simple	-	No	2a03:4000:4a:cf:1443:a0ff:fe58:2ee2
	300	-	-	-
	forum.xcsoar.org	A	Simple	-	No	45.142.179.218
	300	-	-	-
	forum.xcsoar.org	AAAA	Simple	-	No	2a03:4000:4a:cf:1443:a0ff:fe58:2ee2
	300	-	-	-
	mail.xcsoar.org	A	Simple	-	No	45.142.179.218
	300	-	-	-
	mail.xcsoar.org	AAAA	Simple	-	No	2a03:4000:4a:cf:1443:a0ff:fe58:2ee2
	300	-	-	-
	mapgen.xcsoar.org	A	Simple	-	No	87.102.254.100
	300	-	-	-
	mapgen.xcsoar.org	AAAA	Simple	-	No	001:4060:ccc:ba51:5054:ff:fec7:f7bf
	300	-	-	-
	www.xcsoar.org	A	Simple	-	No	45.142.179.218
	300	-	-	-
	www.xcsoar.org	AAAA	Simple	-	No	2a03:4000:4a:cf:1443:a0ff:fe58:2ee2
	300	-	-	-
```

@codexx666 I think you can terminate the old vm.. maybe after one day.

closed 2024-06-25 by lordfolken

codexx666, 2024-06-28:
Ok, AWS instance is stopped. 
xcsoar.org seems still to be working :-)