XCSoar/XCSoar #1340: OpenAir extended format: use AY indication for colouring with AC UNCLASSIFIED

llauner, 2024-02-02:
XCSoar version: 7.41

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Currently the airspace colour is based on the Airspace Class: AC
Following the Extended OpenAir format parsing #1152 the colouring should be based on the airspace type (AY) when AC is set to  UNCLASSIFIED.

For exemple, this Airspace is declared as 
[image]

And is displayed using the "Unknown" color:
[image]

It should be displayed using the "Prohibited" color.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
Fully implement the OpenAir extended format so that the AY type indication is taken into account in the airspace colour

ubx, 2024-02-04:
Is there an 'official' description of the OpenAir extended format ? Or do we have to agree with [this implementation](https://github.com/openAIP/openaip-openair-parser?tab=readme-ov-file#node)?

llauner, 2024-02-04:
> Is there an 'official' description of the OpenAir extended format ? Or do we have to agree with [this implementation](https://github.com/openAIP/openaip-openair-parser?tab=readme-ov-file#node)?

This is the "official" documentation:
http://www.winpilot.com/UsersGuide/UserAirspace.asp

ubx, 2024-02-04:
> > Is there an 'official' description of the OpenAir extended format ? Or do we have to agree with [this implementation](https://github.com/openAIP/openaip-openair-parser?tab=readme-ov-file#node)?
> 
> This is the "official" documentation: http://www.winpilot.com/UsersGuide/UserAirspace.asp

I'm familiar with this document, but it doesn't provide much information about the extended format (Extensions suggested by the Naviter Company?). To implement the OpenAir extended format, we need to understand its specifications.

kobedegeest, 2024-02-05:
The AY and extended format is introduced by openAIP i belief. Info can be found here [https://www.openaip.net/docs](https://www.openaip.net/docs). 

And i agree that looking at the type can be useful (even if AC field is given). Example a "restriced airspace" all have class R in the AC field but can mean different things RMZ or TMZ, ... I also find marking for example "CTR" differently useful.

llauner, 2024-02-05:
> And i agree that looking at the type can be useful (even if AC field is given). Example a "restriced airspace" all have class R in the AC field but can mean different things RMZ or TMZ, ... I also find marking for example "CTR" differently useful.

Well, that's the point, they shouldn't have R in the AC field, as airspace class is A,B,C,D,E,F,G only and airspace type can be R
http://www.winpilot.com/UsersGuide/UserAirspace.asp
Hence the problem, when you have AC UNCLASSIFIED (because it's not A...G), and AY R, it shows up as unclassified when it should show up as Restricted.

ubx, 2024-02-05:
The [OpenAIP documantation](https://www.openaip.net/docs) says:

`AY: Specifies the type of the airspace, e.g. "CTR". In the extended format, the AC tag is exclusively used to specify the airspace ICAO class. The AY tag is required and must be placed directly after the AC tag. `

A search for AC tags following AY tags in some commonly used airspace files ([xcsoar-data-content](https://github.com/XCSoar/xcsoar-data-content)[](https://github.com/XCSoar/xcsoar-data-content/tree/master/data/remote/airspace/country)) shows:

- 2023_07_Airspace_Germany2023.txt has total AC with following AY 295, without AY 45
- SFVS-FSVV_CH-Airspace23_eDABS_APR2.txt has total AC with following AY 295, without AY 45
- france.txt   has total AC with following AY 1288, without AY 0
- australia_class_all_23_11_30.txt has total AC with following AY 0, without AY 1877

The questions are:

- How to distinguish between the **OpenAir** and **OpenAir extended** format?
- If we implemented a fully OpenAir extended parser, what to do with records not following the above specs?

llauner, 2024-02-05:
Well, I guess this as to be "flexible" at least in a first approach, and allow that some AC will not have AY, and in other cases there will be an AY after the AC...I know it's easy to say, but a pain to implement...

lordfolken, 2024-02-06:
The way I understand this, is that the format can be chosen per record.

Both formats agree on the basic airspace classes. As long as its one of a-f, you check for the presence of AY.

ubx, 2024-02-06:
Can we agree on the following:
if AC has an ICAO airspace class and is followed by AY, then we have an extended record, with AY representing the airspace class?

lordfolken, 2024-02-06:
Basically this from the openaip parser:

If any of these

> if AC has an ICAO airspace class and is followed by AY, then we have an extended record, with AY representing the airspace class?

if any of these values appear in AC:
  extendedFormatClasses: ['A', 'B', 'C', 'D', 'E', 'F', 'G', 'UNCLASSIFIED'],

Then we check for presence of AY.

ubx, 2024-02-06:
> UNCLASSIFIED
Why? It is nowhere specified, just an invention of the parser.

llauner, 2024-02-06:
> > UNCLASSIFIED
> > Why? It is nowhere specified, just an invention of the parser.

Well, it looks like UNCLASSIFIED is widely used now.
Which makes sense when the ICAO class for a given airspace is not A,B,C,D,E,F,G

As indicated above, it's also listed:
>OpenAIP lists them in the headers of their files

reskume, 2024-02-25:
This does not help directly with solving this issue but I wanted to shed some insights on this "extended format" that we use with openAIP. Actually, this was proposed by Naviter, as already mentioned in one comment above. We just picked up the idea and use it as another export/import format for openAIP and extended it with one tag, the "AI" tag. Reason behind this was that we didn't want to loose metadata that is already present in the system when exporting to openAIR, e.g. using only AC instead of AC+AY. Also, the system may also includes multiple frequencies for each airspace and we are now able to at least dump the "primary" one including the ground station callsign with using AF+AG tags.
The AF/AG and AC/AY combination can be found in various country airspace files (e.g.  Switzerland, Austria (although Austrocontrol leaves the AY blank if not defined) ...). The only thing that openaAIP "added", besides extending the AY types with the internal types we use, was the AI tag so that we are able to track changes to different versions of a country file if necessary. Personally I consider the AI tag to be a very useful addition because it helps a lot with doing automated diffs and imports in our system. Unfortunately, it's not really used anywhere although we provide the [OpenAIR Fixer Tool](https://github.com/openAIP/openaip-openair-fix-format) that can add this line to every openAIR file. I guess it's because the format is specifically consumer oriented.

ubx, 2024-05-28:
[image]

[You can try it out](https://github.com/XCSoar/XCSoar/actions/runs/9261950576/artifacts/1542425080)

closed 2024-05-30 by lordfolken

lordfolken, 2024-06-05:
Ty for reviewing in depth.
Please open seperate issues.
We can Tag and discuss which oft these are release critical and fix them in RC1