XCSoar/XCSoar #1342: Crash on start: android cache dir not found

fds2610, 2024-02-03:
XCSoar versions having and not having the problem
-------------------------------------------------

V. 7.41
F-Droid
LineageOS20
No Gapps installed

System information
------------------

LineageOS model gaugin (Mi10Tlite)

Steps to reproduce the behavior
-------------------------------

Installiert APP
Start APP
Error message

Expected behavior
-----------------

APP should display Start screen

Actual behavior
---------------

APP displays error message
Java.lang.Exceptuon: No Android cache directory

lordfolken, 2024-02-03:
Is your storage full?

fds2610, 2024-02-03:
thanks for your reply. Nope. 52% used of 64 GB

lordfolken, 2024-02-03:
Can you provide a logcat?
Also the xcsoar.log would be interesting.

lordfolken, 2024-02-09:
No activity. Closing.

Nati-AS, 2024-05-11:
Same problem here, in Android 13 TPhone wirh Google Apps.
No XCSoar folder, No log file found

First Screen when opening the app only shows this Message. I tried apk files from Webpage as well as fdroid.

lordfolken, 2024-05-11:
Again. I need to see a logcat.

Nati-AS, 2024-05-14:
Thanks for reopening the issue.

As I didn't know (and couldn't find) how to filter adb output properly here is a filtered logcat made by logcat reader app. I hope everything important is inside.
Especially I get the error:
Failed to set project id on /data/media/0/Android/data/org.xcsoar/: Operation not supported on transport endpoint

If this does not help I can provide another file or more info
[logcat_05-14-2024_21-05-22.txt](https://github.com/XCSoar/XCSoar/files/15313015/logcat_05-14-2024_21-05-22.txt)

lordfolken, 2024-05-14:
Failed to set project id on /data/media/0/Android/data/org.xcsoar/: Operation not supported on transport endpoint

I supect this means that android cannot set the selinux context on the directory for xcsoar, because the filesystem doesn't seem to support it. 

Xcsoar then cannot create any files or directories, because the permissions are incomplete. 

Are you by chance using an sdcard?

Nati-AS, 2024-05-14:
> Xcsoar then cannot create any files or directories, because the permissions are incomplete

got the same thought, but I have no idea why. I didn't experience any similar problems with other apps.

> Are you by chance using an sdcard?

No, no sd card in the phone

lordfolken, 2024-05-15:
Can you try the attached pull request, please? This moves all storage to the primary path.

lordfolken, 2024-05-21:
@Nati-AS can you try out the pr?

closed 2024-06-10 by lordfolken