XCSoar/XCSoar #1344: Forum: No posts / replies possible

lordfolken, 2024-02-09:
[image]

The textbox isn't properly specified, and not accessible.

lordfolken, 2024-02-09:
maintenance page set for the moment.

lordfolken, 2024-02-09:
phpBB upgraded to 3.3.X
still looking at performance issues.

paynterf, 2024-02-11:
Nice to know I'm not the only one seeing this ;)

lordfolken, 2024-02-12:
It refuses to work with any smaller php memory limit than 512mb. The debug output suggests a maximum of 98mb usage. 
Phpfpm seems to be stuck for 5-7s due to memory unmap, according to strace. Possibly because the box us trashing due to 1gb of memory. 
@afp any ideas?

paynterf, 2024-02-12:
Move it to another box with > 1GB temporarily, to rule in/out RAM as the culprit?

lordfolken, 2024-02-13:
I deployed it in a local vm with 4gb of memory. Everything works, except that its very slow.

paynterf, 2024-02-13:
Nice!  So definitely a RAM issue?

lordfolken, 2024-02-14:
No still slow.

lordfolken, 2024-02-19:
Andreas Pfaller sent me the following:

just for info: I fixed some issues with forum.xcsoar.org

+ Increased php memory limit in /etc/php/8.2/apache2/php.ini
+ enabled module log_forensic
+ modified /etc/apache2/sites-available/forum.xcsoar.org-ssl.conf
  - renabled SSL and forensic logging
  - added the "access-rules" include
+ added the required include files for the access-rules config to /etc/apache
+ patched phpbb (with the patch which was already present in 
  /root/xcsoar-forum-phpbb-session.diff on the server) to reduce
  memory used by phpbb ridiculous caching method. Because of this
  the first change in this list could probably be reverted, I have not
  tried it yet.

This last patch really fixes the performance.

lordfolken, 2024-02-19:
```
just for info: I fixed some issues with forum.xcsoar.org

+ Increased php memory limit in /etc/php/8.2/apache2/php.ini
+ enabled module log_forensic
+ modified /etc/apache2/sites-available/forum.xcsoar.org-ssl.conf
  - renabled SSL and forensic logging
  - added the "access-rules" include
+ added the required include files for the access-rules config to /etc/apache
+ patched phpbb (with the patch which was already present in 
  /root/xcsoar-forum-phpbb-session.diff on the server) to reduce
  memory used by phpbb ridiculous caching method. Because of this
  the first change in this list could probably be reverted, I have not
  tried it yet.
```

closed 2024-02-19 by lordfolken