XCSoar/XCSoar #1345: WIN64: TestAirspaceParser.exe fails

ubx, 2024-02-10:
XCSoar versions having and not having the problem
-------------------------------------------------
WIN64, latest from master (9149c338f94a03f5e596265759e3cf34b07e5e4b)

System information
------------------
[see here]( https://github.com/ubx/XCSoar/actions/runs/7854201754/job/21434955765)

Steps to reproduce the behavior
-------------------------------
On a Linux box: 

`make TARGET=WIN64 DEBUG=y USE_CCACHE=y V=2 everything`

`wine output/WIN64/bin/TestAirspaceParser.exe`

Expected behavior
-----------------
All tests should PASS

Actual behavior
---------------
```
not ok 37 - airspace.GetRadioFrequency() == RadioFrequency::FromMegaKiloHertz(130, 125)
#     Failed test (test/src/TestAirspaceParser.cpp:TestOpenAir() at line 116)
```
and
```
not ok 79 - airspace.GetRadioFrequency() == RadioFrequency::FromMegaKiloHertz(130, 125)
#     Failed test (test/src/TestAirspaceParser.cpp:TestTNP() at line 224)

```

[and here]( https://github.com/ubx/XCSoar/actions/runs/7854201754/job/21434955765)

Do you have any idea what may have caused this?
-----------------------------------------------

File encoding ?

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2024-02-10:
Jup, saw that when fixing the toolchain. Care to bisect?
There is also #1336 and #1332

ubx, 2024-02-11:
```
78766892c2eb3e52b07b4225b18d423949c309db is the first bad commit
commit 78766892c2eb3e52b07b4225b18d423949c309db
Author: Max Kellermann <max.kellermann@gmail.com>
Date:   Sun Oct 8 16:11:45 2023 +0200

    Airspace/Parser: migrate from TLineReader to BufferedReader

 build/test.mk                         |   8 +-
 src/Airspace/AirspaceGlue.cpp         |  18 +-
 src/Airspace/AirspaceParser.cpp       | 312 +++++++++++++++++-----------------
 src/Airspace/AirspaceParser.hpp       |   6 +-
 test/src/RunAirspaceParser.cpp        |   6 +-
 test/src/RunAirspaceWarningDialog.cpp |  12 +-
 test/src/RunMapWindow.cpp             |   9 +-
 test/src/TestAirspaceParser.cpp       |   7 +-
 8 files changed, 189 insertions(+), 189 deletions(-)
```

output/WIN64/bin/TestAirspaceParser.exe NOK
output/PC/bin/TestAirspaceParser.exe NOK
output/UNIX/bin/TestAirspaceParser  OK

closed 2024-02-14 by lordfolken