XCSoar/XCSoar #1346: Problem with Creating a Task

hag3fe, 2024-02-12:
XCSoar versions having and not having the problem
-------------------------------------------------

 XCSoar version 7.40 windows 64
Which XCSoar version is the last one you know that did not show the problem? --

System information
------------------

 Desktop PC, Windows 11

Steps to reproduce the behavior
-------------------------------

 I created a task and stored it. 
      Loading was not possible. Failure message was: see screenshot

Expected behavior
-----------------

 File should be loaded and work

Actual behavior
---------------

 In the file in which the task was stored, the name of WP wasn't included. The name is "1". All other data of the waypoints look good, including the index of the waypoint.
--> on the map the task is shown correctly. See screenshot
--> I tried to use all

Do you have any idea what may have caused this?
-----------------------------------------------

 no idea, I found some bad datasets in the waypoint-file, but this was not the root-cause
--> it doesn't matter wether the name of the waypoint is set between quotion marks or not

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2024-02-14:
(WIN64) Indeed, creating a task creates a file with just "1" for name, comment,type etc.

```
<Task pev_start_window="0" pev_start_wait_time="0" fai_finish="0" finish_min_height_ref="1" finish_min_height="0" start_max_height_ref="1" start_max_height="0" start_max_speed="0" start_score_exit="1" start_requires_arm="0" aat_min_time="0" type="1">
	<Point type="1">
		<Waypoint altitude="306" comment="1" id="2249" name="1">
			<Location latitude="46.887" longitude="15.8003"/>
		</Waypoint>
		<ObservationZone type="1"/>
	</Point>
	<Point type="1">
		<Waypoint altitude="1899" comment="1" id="4724" name="1">
			<Location latitude="47.0289" longitude="8.68888"/>
		</Waypoint>
		<ObservationZone radius="500" type="1"/>
	</Point>
	<Point type="1">
		<Waypoint altitude="489" comment="1" id="1561" name="1">
			<Location latitude="48.2286" longitude="11.4228"/>
		</Waypoint>
		<ObservationZone type="1"/>
	</Point>
</Task>

```

lordfolken, 2024-02-14:
so... as far as i could gather, the SetAttribute funciton recieves a c_str() as argument. 
On Linux this is a char, however on windows it passes a w_char.  
https://github.com/XCSoar/XCSoar/blob/903981c7a9e9e829ef06f755b2ceaf778c307586/src/Task/Serialiser.cpp#L58-L62

However the implemenation in
https://github.com/XCSoar/XCSoar/blob/903981c7a9e9e829ef06f755b2ceaf778c307586/src/XML/DataNodeXML.cpp#L49-L53

Seems to only accept char, and there is no w_char implementation.

lordfolken, 2024-02-14:
I did add some LogFormat() outputs in Serialiser.cpp. 
Outputting the value data.name.c_st().
On windows the LogFormat wants %ls a placeholder, while on linux its %s. 

The Values are there as the windows log output shows:
[2024-02-14T23:12:13Z] name: Dit 1Sp Dittinge
[2024-02-14T23:12:13Z] comment: DITTINGEN
[2024-02-14T23:12:13Z] name: Grenchen GLD
[2024-02-14T23:12:13Z] comment: ATIS 121.105 GLD 127.580

lordfolken, 2024-02-15:
Yes, 7.38 works fine as indicated.
But the changes since then are quite extensive and do make sense.
There is just one place where the sender and receiver do not agree on the data format.

ubx, 2024-02-19:
First bad commit is: `4d709b0b60e875fdd9bd69d463b859bb98f78bda`

MaxKellermann, 2024-02-20:
The problem is that `const wchar_t*` gets implicitly casted to `bool`, which is a pretty fragile API. 1189da0cb4fb9867fd5962590feffba15e417c24 prevents this implicit cast (and breaks the build).

PR #1350 (2024-02-22): test/TestTaskSave: add a test for saving and loading a task
May help to fix  #1346
PR #1353 (2024-02-22): Taskwriterwin
closes #1346
closed 2024-02-22 by lordfolken