XCSoar/XCSoar #1360: Flarm Radar and Thermal Assistant background have wrong scale

Turbo87, 2024-03-02:
XCSoar versions having and not having the problem
-------------------------------------------------

> Which XCSoar version are you using when seeing the problem?

v7.41

> Which XCSoar version is the last one you know that did not show the problem?

I'm unsure. I know that last summer this wasn't an issue, but when I gave an XCSoar workshop two weeks ago this was broken.

System information
------------------

- Android on OnePlus 8
- Windows with HighDPI screen

Steps to reproduce the behavior
-------------------------------

Easiest way to reproduce this is with an NMEA log with flarm targets and/or while circling.

Actual behavior
---------------

[image]

[image]

Expected behavior
-----------------

It looks like the Flarm targets in the foreground and the blue part of the thermal assistant are scaled correctly, but the background for some reason is not. That makes the foreground not fit the number scale on the background.

Do you have any idea what may have caused this?
-----------------------------------------------

unfortunately not

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2024-03-13:
##  Testing
One can test with FeedNMEA
make ./output/UNIX/bin/FeedNMEA
./output/UNIX/bin/FeedNMEA tcp 4353 < test/data/driver/FLARM/pflaf01.nmea

and a device configured to TCP Client and host localhost

## Reproduce
Imo everything is drawn at the correct scale. The distance seems to match the circles even at high dpi:

https://github.com/XCSoar/XCSoar/assets/407371/85af371b-a32f-46b2-b428-eb65cad2ee91

Don´t get me wrong, the dialog is unintuitive and broken
  * the circle don't indicate that the zoom level has changed 
  * to few circles in relation to screen size
  * the zoom level is set to the last alarm, but not reset

lordfolken, 2024-03-14:
So i have created a pr for the flarm side of things:
https://github.com/XCSoar/XCSoar/pull/1382

Will have a look at the thermal assistant tomorrow.

lordfolken, 2024-03-14:
So i found the bug. updated my PR.

Turbo87, 2024-03-14:
Interesting. looks like this was introduced about 6 months ago by https://github.com/XCSoar/XCSoar/commit/808a32721f41530aa6b44f6396978c509efcc891