XCSoar/XCSoar #1361: WiFi not working on Kobo Touch

ulipo, 2024-03-02:
XCSoar versions having and not having the problem
-------------------------------------------------
v7.42 - wifi not working
v7.41 - OK

System information
------------------
Kobo Touch

Steps to reproduce the behavior
-------------------------------
Turn on wifi, enter the "wifi" page and do a scan

Expected behavior
-----------------
there should be a list of all avaliable, saved and connected networks

Actual behavior
---------------
The list shows just the saved networks and as unreachable, nothing connected.

Do you have any idea what may have caused this?
-----------------------------------------------
It seems to me that turning on the wifi takes slightly less time on 7.42 than on 7.41 and older versions

Do you have an idea how to solve the issue?
-------------------------------------------

ulipo, 2024-03-02:
I think I have pinpointed the commit to blame: 620f912d6afcfffc250ac5051bd11e1ddf8e2e49 
Compiling before that point wifi works, after that commit it stops working.
I see that there are changes in WPASupplicant.cpp but I'm not skilled enough to understand what's wrong :-(

ulipo, 2024-03-03:
I can confirm that the fix works on my Kobo Touch and also on my "production" Kobo GloHD.

closed 2024-03-04 by MaxKellermann

Rensiepensie, 2024-08-23:
Yes also I had this problem. And with this update my Kobo GloHD Wifi Connection works again. But now FTP connection does not function anymore as before.
Any changes in FTP server with this update?
Login Fail
PASS *********************
Antwoord:	530 Login failed

FTP login method anonymous. Before a username and password was not necessary.

PR #1362 (2024-03-04): Fix WiFi not working on Kobo
Fixes #1361