XCSoar/XCSoar #1365: IOS/MacOSX Targets fail with Checksum.hpp:17:1: note: candidate template ignored: couldn't infer template argument '_src:auto'

lordfolken, 2024-03-05:
Current master fails to build on ios and macos targets. IOS doesn't report an error interestingly. 

```
src/Device/Util/NMEAReader.cpp:53:39: error: no matching function for call to 'NMEAChecksum'
  const uint8_t calculated_checksum = NMEAChecksum({start, asterisk});
                                      ^~~~~~~~~~~~
./src/NMEA/Checksum.hpp:43:1: note: candidate function not viable: cannot convert initializer list argument to 'std::string_view' (aka 'basic_string_view<char>')
NMEAChecksum(std::string_view src) noexcept
^
./src/NMEA/Checksum.hpp:17:1: note: candidate template ignored: couldn't infer template argument '_src:auto'
NMEAChecksum(std::convertible_to<const char *> auto &&_src) noexcept
^
1 error generated.
gmake: *** [build/compile.mk:104: output/OSX64/dbg/src/Device/Util/NMEAReader.o] Error 1
gmake: *** Waiting for unfinished jobs....
gmake: 
```

HugoKNL, 2024-03-15:
Since iOS 17.4 sideloading apps is allowed on iPhones. It’d be great if iOS would be supported again! I hope someone can make it work. Many people I know wish for this app to work on their iPhones. (The pipeline build run can be found in the Github Actions section).

lordfolken, 2024-03-15:
> Since iOS 17.4 sideloading apps is allowed on iPhones. It’d be great if iOS would be supported again! I hope someone can make it work. Many people I know wish for this app to work on their iPhones. (The pipeline build run can be found in the Github Actions section).

Are you willing, or have access to someone,  that can do this?  Personally I do not have Apple hardware and its not a priority for me. However im willing to support such an effort infrastructure wise.

Xcsoar is not free, its price is the effort you bring to the table.

HugoKNL, 2024-03-15:
I don’t have enough knowledge and skills to do this unfortunately and nor do any gliding enthousiasts I know. 

I do understand it’s not free and appreciate the amount of work you put into this project. I know many people who love XCSoar

I’ll keep checking if people I meet have C++ knowledge and skills and can contribute. (One name pops to my mind who I will ask soon, not sure if he can help though).

lordfolken, 2024-04-23:
Repoening this as, the current approach is a workaround, that needs revising.

yorickreum, 2025-04-29:
Thanks for merging. This issue can be closed now imho.

closed 2025-06-17 by lordfolken