XCSoar/XCSoar #1370: Release Capability

lordfolken, 2024-03-06:
The project isn't in the position to make a release and push it to end users right now.

Current Situation:

On F-droid there is a release of xcsoar, but its basically a 3rd party compilation, that seems to be triggered by releases on github. However currently its stuck on V7.41

Then there is download.xcsoar.org/releases for all current releases. (Kobo,Linux,Android etc)

There is a play store release by Max tied to his personal account.

At this moment the steps to create a release aren't defined or documented.

zinuzoid, 2024-03-07:
Regards Play store, I manage my fork (XCSoar JET) in Play store so let me know if I can help. 

First step is probably to create a company publisher account (XCSoar) them we can transfer the app from personal account to the new one.

lordfolken, 2024-03-07:
@zinuzoid your help would be very welcome. 

I tried applying for an org account on the google play store. This would need a D-U-N-S number. This number seems to be available only to legal entities.

zinuzoid, 2024-03-12:
@lordfolken I can try to create the org account. I just want a confirmation that we're committed on this route before I proceed and pay the account fee? (otherwise I would pay for nothing)

This route means: we'll use the new play store developer account going forward by either
- Migrate the current app from personal account to the new org account
- Or start from scratch (probably lose all stats)

lordfolken, 2024-03-12:
> @lordfolken I can try to create the org account. I just want a confirmation that we're committed on this route before I proceed and pay the account fee? (otherwise I would pay for nothing)
I would have preferred so, but in lack of a DUNS number this doesn't seem possible. Do you have access to one?

> This route means: we'll use the new play store developer account going forward by either
> 
>     * Migrate the current app from personal account to the new org account
Max has left the project and removed his play store entry for xcsoar. So this might already be too bad. 
I don't have the signing keys for his release, so I'm afraid that its going to be option 2:

>     * Or start from scratch (probably lose all stats)

zinuzoid, 2024-03-12:
I've got DUNS number and paid the fee, but now they needed a hard document for verification. One of any
- Certificate of incorporation
- VAT registration certificate
- Certificate of registration of charity
🤔

Another less bad option is I can turn this account (newly created) back to individual and we share the credential, not the best but still an option.

lordfolken, 2024-03-13:
I think the shared credential would be the way to go. I would add this as a github secret, and we can then use it in a pipeline that hopefully pushes to firebase/play store.

lordfolken, 2024-03-13:
> Since we're gonna start publish app from scratch again and likely need to change Android package name (anyway?) Have we thought about official fork and new a new repo and project name like (XCSoar gen2, XCSoar N, ...)

I was hoping for continuation. 

The app package id has to be unique in google play. Even if its no longer published, if there have been any installs, you cannot recycle the app package id, with a new signing key. 

According to play store, you can transfer an app between accounts: 
https://support.google.com/googleplay/android-developer/answer/6230247?sjid=7457013487326523018-EU
So in theory it could be transferred even if unpublished at this time.
MaxKellermann would have to do this, and for this we need an account first. 

> This might need a separate issue to start more conversation. 

If he doesn't agree or doesn't react once we make that request to him, we will have to think about a new name and a fork. 

> Anyway I'm contacting the support to converting the account back to an individual one, should be available in a few days.

lordfolken, 2024-03-13:
https://github.com/marketplace/actions/upload-android-release-to-play-store this would be the github action that i'd use.

OBrown92, 2024-03-14:
We use fastlane for app releases. Fastlane also has a github action, might be worth a look. Storing the keys on Github is a good idea at first, but they are not so easy to see afterwards, so maybe a shared password safe would be an idea.

lordfolken, 2024-03-14:
Fine with me also. Do you have an example repo somewhere?

OBrown92, 2024-03-14:
Not really, it's all closed source and I not tried the github action so far 😆 I just want to mention it, because it has a good community behind it.

zinuzoid, 2024-03-15:
I've got the individual account. Let me know where to put the credential and any other keys.

lordfolken, 2024-03-15:
Great. 
So i'm reading through fastlane docs https://docs.fastlane.tools/getting-started/android/setup/

Basically the setup process would require a service account. This is described in the above link. 
This we would add to the github secrets. (I sent you an invitation to the ORG)

lordfolken, 2024-03-15:
maybe send the credential to me aswell folken@kabelsalat.ch

zinuzoid, 2024-03-16:
I've invited your email to the account as admin and the new account credential.
Also the instruction to transfer the app from other account.

lordfolken, 2024-03-19:
So unfortunatly i didnt get a reaction from @MaxKellermann concerning the app in the playstore.
So we need to rename the package ID, and publish with it. This will also break the fdroid release.

lordfolken, 2024-03-19:
I also saw that we can no longer upload apk in the playstore, so an application bundle is needed now.

zinuzoid, 2024-03-19:
> So we need to rename the package ID, and publish with it. This will also break the fdroid release.

Ok, should we first make decision if we want an official fork or stay in this repo?

> I also saw that we can no longer upload apk in the playstore, so an application bundle is needed now.

What do you mean by that? (I just upload my fork a few days ago)

lordfolken, 2024-03-19:
> > So we need to rename the package ID, and publish with it. This will also break the fdroid release.
> 
> Ok, should we first make decision if we want an official fork or stay in this repo?

I'm one of two owners of this repo/org. 
We don't really need to fork. Basically we cloud rename XCSoar to XCSoarNG XCSoar2 or so, and that would be it.

Users will have to reinstall and we can't provide updates to existing users regardless of what we do now.

A new package id will ensure that there is a unique target and no conflicts. 

> > I also saw that we can no longer upload apk in the playstore, so an application bundle is needed now.
> 
> What do you mean by that? (I just upload my fork a few days ago)

According to play console help: https://support.google.com/googleplay/android-developer/answer/9859348?visit_id=638464818898434802-1241153029&rd=1
"With a release, you can manage your app's [Android App Bundle](https://developer.android.com/guide/app-bundle/) (or APK for apps created before August 2021) and then roll out your app to a specific track."
Applications newer than 2021 need to provide a app bundle and not an apk.

lordfolken, 2024-03-25:
@zinuzoid do you have some guidance as to how you upload an apk? because i'm not able to do so. Is your XCSoar fork from that time 2021?

zinuzoid, 2024-03-26:
> We don't really need to fork. Basically we cloud rename XCSoar to XCSoarNG XCSoar2 or so, and that would be it.

Ok that's clear. I respect that.

> @zinuzoid do you have some guidance as to how you upload an apk? because i'm not able to do so. Is your XCSoar fork from that time 2021?

Yeah, unfortunately my fork was created pre 2021 so I still can do normal .apk file. I simply make `ANDROIDFAT` apk target locally and upload it via browser.

And because we don't use Gradle, that make lot of things not work out of the box :-/
We'll now need to make the build pipeline output the .aab file instead. (As a benefit .aab file can generate many .apk variants that dedicated to each device instead of what we have "ANDROIDFAT" target which included all possible device variants)

lordfolken, 2024-06-05:
I'm fixing the release upgrade in my private repo. https://github.com/lordfolken/xcsoar

zinuzoid, 2024-06-16:
I've been playing with Github workflow to sign apk from Github secret in https://github.com/zinuzoid/XCSoar/commit/36be420d9b63df36cad3ce30c341018391b615af , if you think it's useful, I can tidy up and do pull request.
But if you have a bigger plan to support `.aab` and rewrite bunch of stuff then this might not needed.

lordfolken, 2024-07-28:
Still a few topics... 
 - the debian package doesn´t seem to work.. it fails with an assertion on start
 - still some naming issues for the artefacts
 - signing for android, testing now.

lordfolken, 2024-09-02:
Fixed the uploads for the testing packages. The user was configured with shell /bin/true, so rsync couldn't work.

Turbo87, 2025-01-06:
what's the state of this? is anything still blocking us from releasing at least for a subset of targets?

lordfolken, 2025-01-06:
This issue may be a bit overloaded. 

The original problem was that the release process wasn't public. 

Since then I invested about 100h into a github workflow that is capable of building releases:
https://github.com/XCSoar/XCSoar/blob/master/.github/workflows/build-native.yml

This is automated to the point that you can tag a release in github, v7.44 and it will build a release, with the attached artifacts. What works is Windows, Android (signed with a github stored key), debian, KOBO. 

https://github.com/XCSoar/XCSoar/releases

Tagging a release will also trigger a new build on f-droid buildserver, https://gitlab.com/fdroid/fdroiddata/-/blob/master/metadata/org.xcsoar.yml

There is still some documentation needed, as there are a couple of places where the version number has to be increased in the code. Also the android build number needs to be incremented. 

What isn't solved is the whole play store issue. 

* Basically the "org.xcsoar" manifest name is locked to the signing keys of the previous developer. I asked for a transfer, but no reaction. 
* This means reonboarding on Play store from scratch
* New applications cannot be submitted in an APK format anymore. They need to be now an https://developer.android.com/guide/app-bundle/ The infra to build such a thing is currently non-existent. 

I maybe should close this issue and generate some new ones depending on the topic.

Turbo87, 2025-01-06:
> * New applications cannot be submitted in an APK format anymore

ugh, okay, I wasn't aware of that 😞 

> * The infra to build such a thing is currently non-existent.

not in general, or not in this project specifically?

lordfolken, 2025-01-06:
> > * The infra to build such a thing is currently non-existent.
> 
> not in general, or not in this project specifically?
Not in this project. 
There seems to be bundletool https://github.com/google/bundletool/releases to build these releases. 
As far as i can see, this whole thing is an archive with architecture seperation, optional data bundles and resources split, so that when a client downloads it, it only downloads that client's architecture and required components.

lordfolken, 2025-01-06:
i found this: https://github.com/iricartb/go-apk2aab-tool

nkappler, 2025-04-01:
hi, any news on this?
I just noticed the app was gone from the playstore.
Do you need help with publishing?

lordfolken, 2025-04-01:
Well if you want to sink your teeth into things, one step to get play store release, would be to modify the current build scripts to produce an aab bundle.

andrestotorica, 2025-04-26:
Hi, I have been exploring the build into App Bundles. It's at an early stage, but wanted to check in to avoid any duplication of effort. 

Anyone interested can check my progress at [my branch](https://github.com/andrestotorica/XCSoar/tree/bundle_build). I'll keep it updated as I move forward.

So far, I've migrated the construction of the base package from using AAPT to the more modern AAPT2, which can output the protobuf format needed for creating a bundle at a later stage. This is not finished tough as the build fails downstream, will look into that next. Once fixed, I can continue actually generating a bundle.

I'll check in here again when I have something good enough to consider merging.

lordfolken, 2025-04-28:
Not yet being worked on. Wanted to get 744 out the door first. I like the approach.

andrestotorica, 2025-05-15:
This is done and sent [PR 1760](https://github.com/XCSoar/XCSoar/pull/1760). Makefile should be backwards compatible, generating the same APKs as before, but via creating an App Bundle - which can be used for Play Store upload when compiled with `TARGET=ANDROIDFAT`.

This is selectively enabled by passing `ANDROID_BUNDLE_BUILD=y` to the make command -see updated build-native.yml-. Once we have deploy tested we can make it the default and clean up the old (now heavily repeated) code in the Makefile.

I've held back stuff that changes the old behavior. However I'd like to simplify the targets for easier usage for devs who want to test on Android but are not familiar with the build (it did confuse me at first). As this may break people's setups I defer to hearing comments before moving on. 
My proposal:
* `TARGET=ANDROID`: the default target, should build the full Bundle and fat APK. This would be good both for distribution and general testing on any device.
* `TARGET=ANDROIDFAT`: remove, as the default takes its place.
* `TARGET=ANDROIDxxx`: for single arch builds. The only reason to invoke this I can think of is to reduce build time for quick testing. A dev can use it as an optimization if he/she understands the limitations. Otherwise, just use the default.
* Of course, these changes updated on the documentation

lordfolken, 2025-10-19:
So something, suboptimal should show up here as soon as the reviews complete (closed testing phase):
https://play.google.com/store/apps/details?id=org.xcsoar.play

I'm unsure about half of the mirad of settings in this play console, so expect it to break.

lordfolken, 2025-11-18:
https://play.google.com/apps/testing/org.xcsoar.play <- This is the closed testing stage. If you are not on the list, please email me your email address.

lordfolken, 2026-03-12:
I think we have most of it together. The workflow will generate a published version when there is v7.xx tag. 
Also the recently implemented ci-pipeline for android is working.