XCSoar/XCSoar #1371: Airspace list has poor text contrast

Turbo87, 2024-03-07:
XCSoar versions having and not having the problem
-------------------------------------------------

> Which XCSoar version are you using when seeing the problem?

v7.41

> Which XCSoar version is the last one you know that did not show the problem?

I'm not sure, but I don't remember seeing this issue last year.

Steps to reproduce the behavior
-------------------------------

Click on an airspace

Expected behavior
-----------------

I expect the text to be the same color for all airspaces

Actual behavior
---------------

The airspace color is used for the text, which makes it very hard to read for some airspaces.

[image]

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2024-03-07:
I agree. Colored text should be antialiased or as in this case simply black.

lordfolken, 2024-03-07:
This is a windows only bug. Linux / Android display black text.

lordfolken, 2024-03-07:
@August2111 could you have a look?

Turbo87, 2024-03-08:
Huh, indeed. I could've sworn I saw that on Android too but I must've dreamt that 😅

August2111, 2024-03-09:
... but I found this issue in very old XCSoar versions too, the 1st I have checked was v6.8.17 ;-)
So I think the problem isn't so big, because up to now I saw only one system with a windows-openvario in a glider...

lordfolken, 2024-03-09:
I guess it was well hidden as it almost looked like a feature.

While I no longer see windows based systems in gliders, I know that the VR e-sport users would surley be happy.

closed 2024-03-10 by lordfolken