XCSoar/XCSoar #1387: potential issue with airspace parser

0870887, 2024-03-22:
XCSoar versions having and not having the problem
-------------------------------------------------
7.42

System information
------------------
Samsung Galaxy A53 5G
Android 14

Steps to reproduce the behavior
-------------------------------
add notam file (attached) to be used as airspace data

Expected behavior
-----------------
database is processed without errors

Actual behavior
---------------
Error "not enough polygon points"

Do you have any idea what may have caused this?
-----------------------------------------------
Might be caused by the database file, but I don't think so because could not identify obvious errors. Also, it did work with past xcsoar versions, but unfortunately I cannot tell at which version the break occurred (haven't used that function for about a year). It is also not a single event, I have numerous database files that ends in error messages.

Do you have an idea how to solve the issue?
-------------------------------------------

[image]
[image]

[notams.20240325.txt](https://github.com/XCSoar/XCSoar/files/14721002/notams.20240325.txt)

lordfolken, 2024-03-22:
Hi there!

The airspace at that location in this file:
*
AC NOTAM
AN NOTAM Banner/Target towing will 12Mar 06:38-03Apr 18:57 Sched:DAILY SR-SS<br> D0367/24 TEST FLT OF ACFT TOWING MEASURING EQPT APRX 150FT BELOW THE PLANE. RTE BTN 543000N 0073000E AND 540000N 0073000E (15NM WNW HELGOLAND DVOR/DME DHE).
AL 200ALT
AH 2500ALT
DP 54:30:00 N 007:30:00 E
DP 54:00:00 N 007:30:00 E
*

From the [standard](http://winpilot.com/usersguide/userairspace.asp)
*    DP coordinate                     ; add polygon pointC

A [Polygon](https://en.wikipedia.org/wiki/Polygon) by definition must have more than 2 points.  

Since you are about the 3rd user suspecting an error in XCSoar when reading that message, how can we change that error message, to make it clear that the error is in the file?

ubx, 2024-03-22:
My 2c comment:  potential issue, those who can read the error message are clearly at an advantage.

lordfolken, 2024-03-23:
The file is from notaminfo.com and the export is geared towards xcsoar and alikes. 
I wrote the guy on the contact form. Let's see if he responds.

lordfolken, 2024-04-29:
I have received no response or reaction. 
I'm sorry there is little that XCSoar can do here. 
Please try to lobby with dave@notaminfo.com to change their output.

closed 2024-04-29 by lordfolken