XCSoar/XCSoar #1392: Fresh install of app V7.42 fails to start SIM or FLY, it just closes (fresh install, and on factory reset lenovo tablet)

mccartes, 2024-04-02:
XCSoar versions having and not having the problem
-------------------------------------------------

System information
------------------

Steps to reproduce the behavior
-------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Do you have any idea what may have caused this?
-----------------------------------------------

ubx, 2024-04-02:
Please submit a comprehensive bug report using the provided template, including all relevant questions and details.

mccartes, 2024-04-02:
I did, but I did not understand the markup language.  It should be readable now.

lordfolken, 2024-04-02:
#1235 duplicate?

mccartes, 2024-04-02:
No, that one is in pan mode.  I never get the map screen in the first place.

lordfolken, 2024-04-03:
Attach a logcat please.

mccartes, 2024-04-03:
what is a logcat ?

lordfolken, 2024-04-03:
Logcat, is a log output from android.

The conclusion from the other bug report is, that there is something wrong with the graphics driver stack in that device.
Without access to hardware, this will be hard to debug. Unfortunately this device is in none of the device testing sites (lambdatest, testgrid.io...) i tried. 

Until such time, i can only dis recommend that tablet for XCSoar.

mccartes, 2024-04-03:
where on the device do I find the Logcat ?

lordfolken, 2024-04-03:
https://www.xda-developers.com/how-to-take-logs-android/

ggajoch, 2024-04-29:
Hey, on my device symptoms are the same. Below is my bug report:

If I can help with testing on the device just let me know.

XCSoar versions having the problem
-------------------------------------------------

7.42

XCSoar versions not having the problem
-------------------------------------------------

7.41

System information
------------------

Device: Aquila GPS from PureGliding 
[Website](https://puregliding.fr/)
[Manual](https://puregliding.fr/wp-content/uploads/Ressources/Aquila%20Quick%20Guide.pdf)

Android info:
Model number: m515
Android version: 6.0
Kernel version: 3.18.22
Build number: MRA58K test-keysV1.920210108

I've cleaned XCSoarData folder.

Steps to reproduce the behavior
-------------------------------

Start app, then click fly or sim. 

Expected behavior
-----------------

App launches normally.

Actual behavior
---------------

- app start to load
- loading bar gets to "Initialising display" 
- crashes with "Unfortunately, XCSoar has stopped"

I'm attaching logcat from the crash:
[xcsoar_7.42_crash.log](https://github.com/XCSoar/XCSoar/files/15144701/xcsoar_7.42_crash.log)

lordfolken, 2024-04-29:
What makes you think that this is the same problem?

lordfolken, 2024-04-29:
The xcsoar.log is emtpy.

ggajoch, 2024-04-29:
@lordfolken the version and symptoms are the same.
I've tried again with the XCSoar logs - and I can't see it being created. I've updated my comment above.

Should I open new issue for my device?

closed 2024-09-02 by lordfolken