XCSoar/XCSoar #1396: Vario Gauge - Graphical Issues

lordfolken, 2024-04-06:
[image]

Issues:
 - [ ] Labels NetAvg / Gross / MC are maligned
 - [ ] The arc numbers overflow the box
 - [ ] The values under NetAvg/Gross/MC do not resize, the labels themselves do.

I think a design along the butterfly vario, would be dooable and look good.

[image]

DanD222, 2024-04-07:
I remember seeing the vario scale replaced by vector lines a while ago on the forum, might be worth looking into as well? The vario scale looks a lot better this way IMO.

https://forum.xcsoar.org/viewtopic.php?p=17530#p17530

[0001-Replace-bitmap-vario-scale-by-drawn-lines.patch.txt](https://github.com/XCSoar/XCSoar/files/14896109/0001-Replace-bitmap-vario-scale-by-drawn-lines.patch.txt)

lordfolken, 2024-04-07:
None of this is a  bitmap. I think that patch was merged long ago.

DanD222, 2024-04-07:
Here’s the photo from the forum:
[image]

It doesn’t look very similar to the current implementation to me.

> None of this is a bitmap

You mean the two vario scales in your first post?

lordfolken, 2024-04-07:
Yes. Neither scala is a bitmap. Its just maths and then rendered. The bitmap scale was removed some time ago.
Ill have a look at the patch? Maybe there is some inspiration there

DanD222, 2024-04-08:
OK, thanks. Maybe see how it looks without the arc? (that doesn’t look all that great IMO)

lordfolken, 2024-04-08:
@DanD222 please review.

DanD222, 2024-04-08:
Some first thoughts.. 

- Looking at the B&W WIP screenshot in your first post, I think that the horizontal line at 0 lift/sink could be made shorter, so as to be the same length as the other lift/sink marks.

- It seems to me that the lines at -3 and -4 are oriented towards a different point in the centre of the now imaginary arc than the lines at -1 and -2.

- I'm not sure if it's doable to have the grey scale lift/sink numbers be "forward" of the white arrow?

- I'm thinking whether the green arc shouldn't be more on the outside od the lift marks (possibly move the lift marks + labels more inwards so that the green bar doesn't run out of space)? Alternatively, are we able to have the green arc "behind" the white lift marks?

- I'm also thinking about is whether the arrows shouldn't be touching/referenced to the right edge of the lines/green bar, instead of the current left edge. This might necessitate moving the numbers inwards a bit.

- I would also consider making the NetAvg hollow arrow a bit thicker.

I don't use the vario display myself, so I'm not all that familiar with it, and had to think about what's displayed for a moment. I'm not sure if this is within the scope of what you're trying to improve, but:

- The MC arrow is visually almost the most prominent element, while being the least useful one (compare to the tiny "Little gray pointer (reverse) with tip" on the Butterfly). I initially thought the MC arrow was the instantaneous/gross vario indicator.

Is that something we can also adjust perhaps?  Could we colour-code the values in the centre of the vario scale the same way as the arrows on the vario (i.e. NetAvg, Gross, MC), so that there are visual cues as to which indicator is displaying which value, providing there's enough contrast so as to be able to differentiate between them on a B&W e-ink display?

elgandoz, 2024-04-09:
Sorry to jump in, I would like to throw my 2 cents here... I quickly realised a mockup to test an improved display on the gauge, a bit more concentrical / symmetrical (like in the forum pic), using a more detailed scale (like the butterfly vario, or a classic analog one - see belove pic)
[image]

The scale would work for both m/s and kt and it could be a more readable.
Also, just like the showcased varios, I believe there is no need to add the minus ["-"] sign in the negative scale, it just add noise in a small area.

[image]

Given also the tiny space and the dimension of the scale numbers on a real screen, we could also consider to remove the scale numbers since the values are already visible within the Gross infobox at the center
[image]

I'm not sure how easy this would be in code, I'm just trying to suggest how to improve the readability.

closed 2026-10-16 by lordfolken