XCSoar/XCSoar #1403: WeGlide Copilot Data implementation

XCNav, 2024-04-15:
XCSoar version
--------------

7.42 Android 11/13

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

As there already exists a connection to the WeGlide servers, it would be perfect to add the WeGlide Copilot Data into the MapDisplay.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
In the WeGlide App you can choose different actual information layers as well based on OGN/Flarm data if you have a subscription. As you can already log on the WeGlide server, this should be accessible cvia XCS.
That would make the use of a second phone or switching between XCS and WeGlide App obsolete.

lordfolken, 2024-04-16:
Is this data accessible via api? 
Do you have documentation for this? 
Do you have someone that can implement this?

XCNav, 2024-04-16:
No, not yet. I addressed this to one of the  WeGlide guys with a link to this issue.

XCNav, 2024-06-04:
Not yet, please keep it open, will work on it this week!

LeNinjaHD, 2025-07-28:
Is there any news on this?

XCNav, 2025-08-10:
> @XCNav have you contact weglide regarding this?

I contacted Johannes from We glide with my concern but got no response. Just started a second approach!

LeNinjaHD, 2025-09-10:
Alright, i got a response from WeGlide! They are willing to provide their APIs if a XCSoar Maintainer contacts them. They told me however that they are a bit sceptical, beacause they provided API in the past which never made it into the App.

LeNinjaHD, 2026-03-12:
> I just want you to realize what you are asking me to do here: This is the ai's estimation of this kind of effort:
> 
> client-side estimate
> 
> If backend provides simple HTTP tiles/grids (recommended v1)
> ~2–4 weeks of focused work
> Network client + scheduler (net/client/... style like tim): 2–4 days
> Cache + expiry + retry/backoff: 2–4 days
> Map overlay renderer integration (MapWindow layer, zoom/opacity): 3–6 days
> Settings UI (enable/disable, opacity, legend, age display): 2–4 days
> Cross-backend rendering fixes (OpenGL + memory canvas + GDI): 2–5 days
> Testing/polish (offline behavior, stale data, performance): 3–5 days

I thought WeGlide might be interested in helping implementing this, however I have heard that they are working on their own app at the moment, so unfortunately this wont happen I believe.

yorickreum, 2026-03-12:
> They want to talk to a XCSoar maintainer first about the scope.

Are they aware of this issue here? Would be a good place to discuss the scope imo.

lordfolken, 2026-03-12:
I wrote them an email. Lets see.

lordfolken, 2026-09-29:
Checking whether this is still current. Not closing.

WeGlide wanted to agree scope with a maintainer before Copilot data can be drawn on the map. I wrote to them in March and nothing further has landed here. If you have a current API note or a sample payload, please add it to this issue.

dspreitz, 2026-10-06:
Any updates on this?

lordfolken, 2026-10-06:
After initial reaction from them to the V7.45 announcment, that they would be happy to have it.
I asked them on the 20/08/2026, 11.02 pm
Then again on the 31/08/2026, 9.10 pm
And again just now. 2026-10-07 

 i asked them for:

* einen Zugang zu euer api für copilot
* eine api doku (zbsp. swagger)
* einen technischen beschreib wie es funktionert. 
* einen kontakt für rückfragen

If someone knows them, can that somebody try to facilitate a contact?

dspreitz, 2026-10-11:
> 26\. If someone knows them, can that somebody try to facilitate a contact?

@lordfolken Did you get feedback from Johannes from WeGlide?

lordfolken, 2026-10-11:
@dspreitz Johannes did, he ask whether we could implement https://www.rfc-editor.org/info/rfc8628/ device flow for authentication.
This means qr code support and so on. 
We could make this generic, and implement in weglide, skylines, etc. This way we would also avoid saving passwords in xcsoar profiles.

lordfolken, 2026-10-19:
So i have attached a far from bug free implementation. This will need a lot of testing. 
It includes two layers vario/cloudbase. (More are not available on the api endpoint...)

The authentication is via QRCode and Device Flow, it should work with a copilot subscription. 

Feedback and patches welcome. 

[image]