XCSoar/XCSoar #1409: [Android] Fullscreen: camera cutout area not used

elgandoz, 2024-04-26:
XCSoar versions having and not having the problem
-------------------------------------------------

Which XCSoar version are you using when seeing the problem?
7.42, 7.43-test
Which XCSoar version is the last one you know that did not show the problem?
TBC

System information
------------------
Pixel 6, Android 15 beta
Samsung S10e, Android 12

Steps to reproduce the behavior
-------------------------------

- Open XCSoar, with default settings
- Enter either FLY or SIM mode
- Observe black area covering the camera cutout

[image]

Expected behavior
-----------------
Maybe this is the expected behavior, but I personally think it would be much better having all the possible space available on screen, with the compromise of a tiny "blind spot".
This is what most of the Android app are doing

Actual behavior
---------------
The whole camera cutout height is "reserved" with a black bar.

Do you have any idea what may have caused this?
-----------------------------------------------
Probably Android "immersive mode" has not been adapted since the pre-cutout phones era.

Do you have an idea how to solve the issue?
-------------------------------------------
I'm not Android dev, but according to the specs we should add the instruction in the style.xml:
- https://developer.android.com/develop/ui/views/layout/display-cutout

I can try to draft a PR

elgandoz, 2024-04-26:
Partially related to #1410

lordfolken, 2024-05-06:
i tried LAYOUT_IN_DISPLAY_CUTOUT_MODE_SHORT_EDGES, i tried enabling notch true in the manifest. Sofar no luck.

bennsch, 2024-09-18:
It worked for me, but only if it is  applied **before** calling window.addFlags()

WindowUtil.java:
```java

static void enterFullScreenMode(Window window) {

  if (Build.VERSION.SDK_INT >= 28) {
    /* layoutInDisplayCutoutMode was added in API 28 (Android 9.0) */
      window.getAttributes().layoutInDisplayCutoutMode
       = WindowManager.LayoutParams.LAYOUT_IN_DISPLAY_CUTOUT_MODE_SHORT_EDGES;
  }

  window.addFlags(FULL_SCREEN_WINDOW_FLAGS);
  enableImmersiveMode(window);
}

```
[image]

Screenshot taken on Android emulator with simulated cutout (Developer Options -> Display Cutout).
I have also tested it on a physical Samsung S10e device and it works as well.  (Android 12)

I'd say this should be configurable by the user. For example, if the user selects "Full screen: ON", there should be another option for "Draw over cutouts"

lordfolken, 2024-09-18:
A proper implementation would check the layout, and only enable "notch view"  when no infoboxen are in the way and only the map is impacted. 
Organic maps handles this well.

groundhogxc, 2024-09-23:
> A proper implementation would check the layout, and only enable "notch view" when no infoboxen are in the way and only the map is impacted. 

Not sure that is necessary. I think it should be left up to the users to decide whether a) they want to just not use the two infoboxes in the center (on the screenshot above), or b) use a layout in which only the map is affected...

Maybe there are people who really want two values at the top, and do not care about the partially covered boxes...

bennsch, 2024-09-23:
A nice blend of both approaches would be to offer the following options for the "Fullscreen" setting:

- ```OFF```   
- ```ON (always avoid cutouts)```   
- ```ON (never avoid cutouts)```
- ```ON (avoid cutouts if necessary)```  _check if only map is impacted by cutout_

That would leave it up to the user to decide whether their phone's cutout is covering to much of the infoboxes or not.

groundhogxc, 2024-09-23:
That is true. But it would add yet another configuration option.

And in my view that is not necessary, since all the possible outcomes: 
- map covered by cutout
- map not covered by cutout (because black bar)
- infobox convered by cutout
- infobox not covered by cutout (because black bar)

are already available to the user now, just by choosing the appropriate setting of "Fullscreen" and choice of the appropriate infobox layout. So personally I would not add it, but if there are other opinions I am not opposed.
(or maybe I am overlooking another scenario)

closed 2026-01-01 by lordfolken