XCSoar/XCSoar #1410: [Android] Gesture navigation pill not transparent

elgandoz, 2024-04-26:
XCSoar versions having and not having the problem
-------------------------------------------------

Which XCSoar version are you using when seeing the problem?
 - **7.42**, **7.43-test**

Which XCSoar version is the last one you know that did not show the problem?
 - _TBC_

System information
------------------
Pixel 6, Android 15 beta

Steps to reproduce the behavior
-------------------------------
- Using a Pixel device with Android 12+ or any AOSP 12+ based OS without heavy customisation (no Samsung OneUI).
  - the device must be using the "Gesture navigation" (default)
- Open XCSoar, with default settings
- Observe the space reserved by the gesture navigation
- Enter either FLY or SIM mode
- go to **Config > System > Look > Screen Layout > Fullscreen** and set it to **Off**.
- Observe the space reserved by the gesture navigation and the status bar visibility

[image]

Expected behavior
-----------------
Modern Android apps display the gesture navigation pill with a transparent overlay.

Actual behavior
---------------
The background of the gesture pill is solid, it takes some space away from the screen and it's visually unpleasant.

Do you have any idea what may have caused this?
-----------------------------------------------
Likely XCSoar has not been adapted to the Android 12+ changes concerning the newer Android gesture navigation

Do you have an idea how to solve the issue?
-------------------------------------------
I have a clue, but I don't know if I have enough knowledge: https://developer.android.com/develop/ui/compose/layouts/insets
We need to make sure the fix won't interfere with the regular 3 button nav.

Nevertheless I'll try to have a go and provide a PR

elgandoz, 2024-04-26:
I believe on Pixel-like Androids it would be nice being able to display the status bar (in order to see the time and other system infos and notifications) but also taking advantage of all the available space. It will also be more aesthetically pleasing.
If this wouldn't be possible for any reason, the gesture pill should be at least  follow the "Dark mode" setting set in the Screen Layout.
[image]

elgandoz, 2024-04-26:
Partially related to #1409

lordfolken, 2024-04-26:
https://developer.android.com/develop/ui/views/layout/immersive i guess its this.

elgandoz, 2024-04-26:
> https://developer.android.com/develop/ui/views/layout/immersive i guess its this. 

Yeah, that's the same link I put in the issue description. It seems easy enough, I can have a go.
I also remember @zinuzoid did something similar in his fork, although the latest rebase suffers from the same issue described here.

zinuzoid, 2024-04-26:
> I also remember @zinuzoid did something similar in his fork, although the latest rebase suffers from the same issue described here.

You probably refers to https://github.com/zinuzoid/XCSoar/commit/68041206954705552f4f776002baef0a42011592
It's super hacky and I'm very ashamed of that 😅

lordfolken, 2024-04-29:
IMO The transparent pill only makes sense for applications with content that scrolls.

In XCSoar's case we would cover some infobox small values with the pill, rendering those difficult to read.
You can simulate the effect, in fullscreen mode by swiping up. 

I found very few of my my apps that make use of this. (ok i'm selective, and 90% of it is FOSS) (i run calyxos currently on Android 14 on a Pixel 6) 
The sole exception being Fossify Gallery And that one for some reason only shows things transparently when scrolling down...

I see more potential in #1409 and would propose to put this issue to not planned.

elgandoz, 2024-04-30:
Yes, unfortunately the gesture pill is not widely implemented (despite it has been introduced for a while), with even some own google apps not using it.

I agree that the pill would be an annoyance, although it would be great for someone to be able to display the Android statusbar: it allows to see battery, time, and mobile data connection (for livetracking) without using several infoboxes.

Ideally, Fullscreen:On would make use of the whole display (including cutout area) while Fullscreen:Off would display only the statusbar (only if using gesture navigation bar, otherwise display everything as is), which is what XCSoar JET was doing, but I'm not even sure if that's currently possible.

Let's close this issue and focus on #1409 , we could still re-open it if someone finds a solution.

closed 2024-04-30 by elgandoz