XCSoar/XCSoar #1416: crash during initialising display

ggajoch, 2024-04-29:
XCSoar versions having the problem
-------------------------------------------------

7.42

XCSoar versions not having the problem
-------------------------------------------------

7.41

System information
------------------

Device: Aquila GPS from PureGliding 
[Website](https://puregliding.fr/)
[Manual](https://puregliding.fr/wp-content/uploads/Ressources/Aquila%20Quick%20Guide.pdf)

Android info:
Model number: m515
Android version: 6.0
Kernel version: 3.18.22
Build number: MRA58K test-keysV1.920210108

I've cleaned XCSoarData folder.

Steps to reproduce the behavior
-------------------------------

- remove XCSoarData folder
- start XCSoar
- click fly or sim

Expected behavior
-----------------

App launches normally.

Actual behavior
---------------

- app start to load
- loading bar gets to "Initialising display" 
- crashes with "Unfortunately, XCSoar has stopped"

Installing 7.41 makes it work fine. Re-installing 7.42 makes it crash again - i.e. this is repeteable. 

I'm attaching logcat from the crash, first run after cleaning data folder and re-installation of the XCSoar.
[xcsoar.log](https://github.com/XCSoar/XCSoar/files/15155908/xcsoar.log)

roscosoarer, 2024-05-04:
I am not enjoying 7.42 as it looks like it loads but then immediately fails to blank screen.  

I have XCSoar 7.42 downloaded on a ~$100 Lenovo M7, android version 11.  I think Android Version 12 is now required.  So I tried to back install 7.41 but Lenovo offers 'update' that I cannot reject.   

7.41 worked fine although the app and data are now buried in an obscure system... Android directory whereas I used to be able to manage airspace, turnpoint, etc files on an SD card.  I can only access that directory via a USB attached PC, not directly via tablet 'file manager' etc.  [and 7.41 does not recognise my existing checklist.txt file that is perhaps too large?]

Hopefully a patch will be developed for Android 11 and the XCSoardata files will again be manageable on an SD card.  

[Meanwhile I will revert to my old Samsung tablet, XCSoar 6.? Android 4 but no WeGlide functions and other enhancements]   Sigh.
  
XCSoar is fantastic software though that I have been using for 10 years as can be seen on WeGlide and predecessors.  Keep it free. [I have been trying to get similar easy functionality on LXNav 8000 stuff but XCSoar has a much better GUI.]

lordfolken, 2024-05-05:
> Sure, I've went through bisect and this is the result:
> 
> ```
> 3a58000d8ff2c831ea75080b2e1403736546d884 is the first bad commit
> commit 3a58000d8ff2c831ea75080b2e1403736546d884
> Author: Max Kellermann <max.kellermann@gmail.com>
> Date:   Fri Mar 1 21:22:44 2024 +0100
> 
>     build/resource.mk: rename 96dpi icons to *_96.png
> 
>     Each rendered variant should have such a suffix.
> 
>  build/android.mk                  |  2 +-
>  build/resource.mk                 | 10 +++++-----
>  tools/GenerateWindowsResources.pl |  2 +-
>  tools/LinkResources.pl            |  2 +-
>  4 files changed, 8 insertions(+), 8 deletions(-)
> ```

Are you sure about that commit? It only renames icons.

lordfolken, 2024-05-05:
> I am not enjoying 7.42 as it looks like it loads but then immediately fails to blank screen.
This is a bugtracker not a forum.

ubx, 2024-05-06:
I'm able to reproduce the problem with XCSoar (latest from Head) on a virtual device, Android 6.0 x86_64.

Screenshot
[image]

xcsoar.log

```
[2024-05-06T11:39:18Z] Starting XCSoar XCSoar v7.43-Android~git#04eab3f9ec
[2024-05-06T11:39:18Z] Failed to initialise Bluetooth: java.lang.Exception: No Bluetooth adapter found
[2024-05-06T11:39:18Z] EGL vendor: Android
[2024-05-06T11:39:18Z] EGL version: 1.4 Android META-EGL
[2024-05-06T11:39:18Z] EGL extensions: EGL_KHR_get_all_proc_addresses EGL_ANDROID_presentation_time EGL_KHR_swap_buffers_with_damage EGL_KHR_image_base EGL_KHR_gl_texture_2D_image EGL_KHR_fence_sync EGL_KHR_create_context EGL_ANDROID_image_native_buffer EGL_ANDROID_recordable 
./src/java/Ref.hxx:139: void Java::TrivialRef<_jclass *>::Set(JNIEnv *, T) [T = _jclass *]: assertion "value == nullptr" failed
```

logcat

```
05-06 13:48:50.024  1686  1698 I ActivityManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=org.xcsoar/.XCSoar (has extras)} from uid 10008 on display 0
05-06 13:48:50.029  3290  3290 D XCSoar  : ABI=x86_64
05-06 13:48:50.029  3290  3290 D XCSoar  : PRODUCT=sdk_phone_x86_64
05-06 13:48:50.029  3290  3290 D XCSoar  : MANUFACTURER=unknown
05-06 13:48:50.029  3290  3290 D XCSoar  : MODEL=Android SDK built for x86_64
05-06 13:48:50.029  3290  3290 D XCSoar  : DEVICE=generic_x86_64
05-06 13:48:50.029  3290  3290 D XCSoar  : BOARD=unknown
05-06 13:48:50.029  3290  3290 D XCSoar  : FINGERPRINT=Android/sdk_phone_x86_64/generic_x86_64:6.0/MASTER/4174734:userdebug/test-keys
05-06 13:48:50.032  3290  3290 D XCSoar  : Successfully added bootstrap class: ioio.lib.impl.SocketIOIOConnectionBootstrap
05-06 13:48:50.033  3290  3290 D XCSoar  : Successfully added bootstrap class: ioio.lib.android.accessory.AccessoryConnectionBootstrap
05-06 13:48:50.033  3290  3290 E BluetoothAdapter: Bluetooth binder is null
05-06 13:48:50.033  3290  3290 D XCSoar  : No runtime support for: ioio.lib.android.bluetooth.BluetoothIOIOConnectionBootstrap. Not adding.
05-06 13:48:50.033  3290  3290 D XCSoar  : Successfully added bootstrap class: ioio.lib.android.device.DeviceConnectionBootstrap
05-06 13:48:50.033  3290  3290 V DevConnBootstrap: onCreate()
05-06 13:48:50.054  1279  2145 D gralloc_ranchu: gralloc_alloc: Creating ashmem region of size 3686400
05-06 13:48:50.076  2622  2649 D EGL_emulation: eglMakeCurrent: 0x7ffe45347960: ver 3 1 (tinfo 0x7ffe4523f2c0)
05-06 13:48:50.076  2622  2649 E Surface : getSlotFromBufferLocked: unknown buffer: 0x7ffe3b5b5cc0
05-06 13:48:50.100  1279  1726 D gralloc_ranchu: gralloc_alloc: Creating ashmem region of size 3686400
05-06 13:48:50.101  3290  3475 D XCSoar  : Context.getExternalFilesDirs()='/storage/emulated/0/Android/data/org.xcsoar/files'
05-06 13:48:50.101  3290  3475 D XCSoar  : Context.getExternalFilesDirs()='/storage/121A-2908/Android/data/org.xcsoar/files'
05-06 13:48:50.102  3290  3475 D XCSoar  : Environment.getExternalStoragePublicDirectory()='/storage/emulated/0/XCSoarData' (not accessible)
05-06 13:48:50.102  3290  3475 I XCSoar  : Starting XCSoar XCSoar v7.43-Android~git#04eab3f9ec
05-06 13:48:50.103  3290  3290 D         : HostConnection::get() New Host Connection established 0x7ffe4ce5aba0, tid 3290
05-06 13:48:50.103  3290  3475 E BluetoothAdapter: Bluetooth binder is null
05-06 13:48:50.103  3290  3475 I XCSoar  : Failed to initialise Bluetooth: java.lang.Exception: No Bluetooth adapter found
05-06 13:48:50.107  3290  3475 D         : HostConnection::get() New Host Connection established 0x7ffe48dd14e0, tid 3475
05-06 13:48:50.107  3290  3475 I XCSoar  : EGL vendor: Android
05-06 13:48:50.108  3290  3475 I XCSoar  : EGL version: 1.4 Android META-EGL
05-06 13:48:50.108  3290  3475 I XCSoar  : EGL extensions: EGL_KHR_get_all_proc_addresses EGL_ANDROID_presentation_time EGL_KHR_swap_buffers_with_damage EGL_KHR_image_base EGL_KHR_gl_texture_2D_image EGL_KHR_fence_sync EGL_KHR_create_context EGL_ANDROID_image_native_buffer EGL_ANDROID_recordable 
05-06 13:48:50.109  3290  3475 E XCSoar  : Initialisation error
05-06 13:48:50.109  3290  3475 E XCSoar  : java.lang.Exception: eglChooseConfig() failed
05-06 13:48:50.109  3290  3475 E XCSoar  : 	at org.xcsoar.NativeView.runNative(Native Method)
05-06 13:48:50.109  3290  3475 E XCSoar  : 	at org.xcsoar.NativeView.run(NativeView.java:160)
05-06 13:48:50.109  3290  3475 E XCSoar  : 	at java.lang.Thread.run(Thread.java:818)
05-06 13:48:50.112  1686  1705 I ActivityManager: Displayed org.xcsoar/.XCSoar: +86ms
05-06 13:48:50.128  1279  1370 D gralloc_ranchu: gralloc_alloc: Creating ashmem region of size 3686400
```
Perhaps this could assist in pinpointing the issue.

lordfolken, 2024-05-06:
I tried this as well when looking at this problem. 
But the submitter says that it doesn't work since 3a58000d8ff2c831ea75080b2e1403736546d884 
The last version that works for me in the emulator is 7.22. Not 7.41

lordfolken, 2024-05-07:
I fixed the first issue, the ALPHA channel for EGL was missing in the initialization for android.

lordfolken, 2024-05-07:
The second problem is if the VDPI value is lower than 120, then this logic is not triggered:
https://github.com/XCSoar/XCSoar/blob/b1244d8dc61e808a3fab3f9d14668aa777e91835/src/ui/canvas/Icon.cpp#L49-L59

ggajoch, 2024-05-10:
As I got the device back, I've tested that:
 - `3a58000d8f`: does not work (as before)
 - `1ef8b0ae91`: works correctly (as before)
 - `ce2cf3d456` from your merge restest #1422 shows the same crash

Attached is logcat from commit `ce2cf3d456`:
[xcsoar.log](https://github.com/XCSoar/XCSoar/files/15277950/xcsoar.log)

lordfolken, 2024-05-11:
Yes i know that it still crashes, despite the EGL_ALPHA_SIZE. The VDPI of your device is around 102. A higher DPI device would work without a problem.

lordfolken, 2024-08-29:
can you please test the attached PR? @ggajoch

closed 2024-09-02 by lordfolken