XCSoar/XCSoar #1419: Download tasks from Soaringspot

LWEdslev, 2024-05-02:
XCSoar version
--------------
7.42 android 

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
I would like to click a button and then the task for the selected competition would be downloaded, enabled and declared (if logger is connected).

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
On a competition morning, it is nice if I don't have to type the task in by hand

lordfolken, 2024-05-03:
Tiere is @kedder s openvario menu that does that.

DanD222, 2024-05-03:
“compman”, specifically https://github.com/kedder/openvario-compman, but that’s intended for Openvario use outside of XCSoar.

Another vector to get a task into XCSoar is to link your WeGlide account with XCSoar, declare your competition task to WeGlide, and then download that in XCSoar.

But yes, it’d still be nice if XCSoar could download a Soaringspot task on its own.

lordfolken, 2024-05-04:
Any takers?

LWEdslev, 2024-05-05:
I could try but need some help with questions since I haven't done any contributions to XCSoar and have limited C++ experience

LWEdslev, 2024-05-06:
I though only POST in the API needed an API key but apparently the GET requests also do.
["In order to access the data you need an AppID and a secret key. You can generate those by going to the “Edit Competition > API Key” page."](https://kb.naviter.com/en/kb/public-api-for-soaring-spot/)
Any ideas on how to handle this?

DanD222, 2024-05-06:
Only advice I can give is maybe have a look at the Compman code to see if you find anything useful there.

LWEdslev, 2024-05-06:
Hmmm, seems like compman doesn't use the API but just goes to the soaringspot homepage and does some web scraping.

LWEdslev, 2024-05-06:
I have written an e-mail to Naviter about how we should proceed with their API

LWEdslev, 2024-06-25:
I am implementing some of it with the XML `ConstDataNodeXML : ConstDataNode`. I need to go through the children and use them as `ConstDataNodeXML` but instead `ConstDataNode` is returned. What is the best way to use it as `ConstDataNodeXML`? `static_cast`?

LWEdslev, 2024-06-25:
How would I go about adding a third-party lib like gumbo to XCSoar?

lordfolken, 2024-06-25:
Build/thirdparty*

Web scraping in xcsoar? Please no. 
The ammount of breakage there is impossible to follow up on. 

Either there is a clear api , or there is no support.

LWEdslev, 2024-06-25:
Ok, seems only glideandseek provides a clear (JSON) API. It does however require a URL for the results page of a competition (such as https://www.soaringspot.com/en_gb/jegc2021/results/club) so the user would have to copy-paste that into XCSoar. Then the latest task could be downloaded for that competition and be parsed into a `.tsk` format. I'll give it a whirl

styblope, 2024-06-25:
My idea is to move whatever parsing/scraping to user-land, i.e. to Lua scripts while exposing the XCSoar's `.tsk` file format ingestion with as new `load_task` function that extends the [Task Lua API](https://xcsoar.readthedocs.io/en/latest/lua.html#task). The implementation is straightforward, using existing `Task/Deserialiser/LoadTask` function and some wrap code to commit the new read task for usage (see my working prototype of the [task load function](https://gist.github.com/styblope/fced9f3ee475eafc887a6745083d2073) and the corresponding [user lua script](https://gist.github.com/styblope/8acb7a0d4f51a344175c0ee1bf2573a4)).
I think exposing the `task_load` function in Lua is legit as it only allows Lua to read XCSoar's it's own task data from outside (irrelevant of the content provider), all with a very low additional code maintenance cost.
The downside to the above approach is there are no (not yet or never planned?) Lua UI interaction functions available, thus all variable inputs must be placed into the script. Yet, such as for a one-time competition, this approach proves good enough. I can submit a draft PR if we want to elaborate on this concept.

LWEdslev, 2024-06-27:
Specified in the glideandseek API [documentation for the task data interface](https://glideandseek.com/about/interfaces/task-data-interface) is that a field is returned specifying "RT" or "AAT" but this is actually not returned, [example](https://api.glideandseek.com/v2/task?url=https://www.soaringspot.com/en_gb/2024-usa-18m-15m-club-national-championship-hobbs-2024/results/18-meter). Comparing the response for an AST and an AAT there seems to be no other indication of which is which. This makes sense for the usage glideandseek has but task type and possible AAT duration is obviously a must for this feature.
Any ideas?

LWEdslev, 2024-06-27:
I don't think the feature has much value if the user must enter that data in the script.
It is also not always obvious for a user which Class they should enter, for example: for such a competition as `british-team-training-issoudun-le-fay-2024` where the class is specified as `Club` but in the URL it is `unknown`.
So fetching that data is also difficult to do automatically without webscraping. 
Maybe the user could paste an URL for the competition and class and we parse that into the tokens we need but still then there is some risk of breakage. That is also the case for the regex matching for the `.tsk` files in the script which seems to be the only way to get the files out from `soarscore`.
However if that is an acceptable approach (which I get that it probably isn't) there could be a feature where the user paste a URL (for a comp with class) into some field in the settings and that could be parsed into the tokens (comp and class).
All in all with glideandseek not providing the necessary data there seems to be no clear API.
> Either there is a clear api , or there is no support.

styblope, 2024-06-27:
The only stable serial format for XCSoar tasks is it it's own `.tsk` format (and possibly also `.cup` format), thus I only propose to expose ingestion of this format from Lua to cheaply open up for custom integrations until any better native solution is devised and agreed upon. None of the mentioned 3rd pty projects (Soarscore, Glideandseek, Soaringspot) imho provide a dependable task source or API which we want to hard-code inside the XCSoar binary. WeGlide extensive integration is perhaps an exception and maybe a reaction to Soaringspot/Naviter not willing to provide a public competition task API for general good.

DanD222, 2025-08-28:
So, I've asked my son if he'd like to do an Android app that does what Compman does as a project; the result is here: https://github.com/DanielDe8/xcomps

closed 2025-08-28 by LWEdslev