XCSoar/XCSoar #1421: Auto-MacCready (final glide mode) jumps from 2,3 -> 6,7

alex8735, 2024-05-06:
The Auto-MacCready (final glide mode) feature is showing a bug. After reaching a value above 2,3 it will jump to 6,7. 

XCSoar versions having the problem
-------------------------------------------------
testded on 7.41 but already observed last summer 2023

XCSoar versions not having the problem
-------------------------------------------------
Unknown, but this was working correctly some time ago.

System information
------------------

The issue has been seen on various android devices

Steps to reproduce the behavior
-------------------------------

Have auto MacCready in final glide mode and achieve an auto-MacCready >2,3
The behaviour is consistent. It can be reproduced every time.

Expected behavior
-----------------

Auto-MacCready (final glide mode) should calculate values above 2,3, counting up, decimal by decimal as the final glide improves

Actual behavior
---------------

Auto-MacCready (final glide mode) will work as advertised for values 2,3 and lower. As soon as avalue above 2,3 is calculated it will jump to a value of 6,7 and remain until 2,3 or lower is achieved again. The limit of 6,7 seems to be the maximum allowed MC value, which is okay.

Do you have any idea what may have caused this?
-----------------------------------------------

No

Do you have an idea how to solve the issue?
-------------------------------------------

No

lordfolken, 2024-05-06:
Logs, igc files, anything?

lordfolken, 2024-05-06:
Also your plane profile would be interesting.

alex8735, 2024-05-06:
I just tested the windows version with an igc replay and it does not show the issue. I will do some further trouble-shooting to further isolate the issue before reporting back with my findings.

lordfolken, 2024-05-06:
One limiting factor is the max cruise speed in the plane profile. Mc will not go past that speed.

lordfolken, 2024-06-02:
Please reopen when you have more info.

closed 2024-06-02 by lordfolken