XCSoar/XCSoar #1424: wrong arrival altitude in Map Labels

bomilkar, 2024-05-09:
XCSoar versions having the problem
-------------------------------------------------

The issue was introduced in 7.34

XCSoar versions not having the problem
-------------------------------------------------

Up to 7.33 it worked OK.

System information
------------------

Not platform dependent.

Steps to reproduce the behavior
-------------------------------

Start XCSoar in Sim mode.
Increase the altitude by clicking on an altitude InfoBox until you see a Waypoint in reach.
Then change Wind manually and observe how the content of "WP AltD" InfoBox changes compared to the altitude shown on the Map Label.

Expected behavior
-----------------

Up to 7.33 "WP AltD" InfoBox and Map Labels are in sync.
[image]

Actual behavior
---------------

[image]

Do you have any idea what may have caused this?
-----------------------------------------------

It seems as if since 7.33 there is a sign error in calculating the arrival altitude in the map label.

<html>

<body>

  | InfoBox | Map Label
-- | -- | --
head Wind | 422 | 469
no Wind | 448 | 447
tail Wind | 470 | 421

</body>

</html>

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2024-05-09:
Polar, saftey mc, and stf factor are the same?
We need a unit test for this.

lordfolken, 2024-05-09:
Probably all the same by judging the wp altd.

bomilkar, 2024-05-09:
Polar, saftey mc, and stf factor are the same.
This becomes obvious at 0 wind: wp altd and altitude in label are the same.

bomilkar, 2024-05-09:
> We need a unit test for this.

What does that mean?

bomilkar, 2024-05-11:
The fix might be trivial, but I simply can't find the part in the code, which calculates the arrival altitude in the map labels.
So unless somebody helps me find it, I won't be able to fix it.

lordfolken, 2024-05-12:
So the labels show you the altitude you have above glide ratio at mc 0. 
The infobox will factor in saftey mc (from settings) and current mc. If you set this to 0 then its the same. 

But as soon as you add the wind component, it increases where its supposed to decrease. 

The reachability is calculated here:

(https://github.com/XCSoar/XCSoar/blob/b1244d8dc61e808a3fab3f9d14668aa777e91835/src/Renderer/WaypointRenderer.cpp#L107)

bomilkar, 2024-05-15:
I don't want to give up hunting the bug, but I don't want to predict by when I'll be done. The code is too convoluted for my limited tool set and my abilities. Sorry!
So, I'd like to ask someone else to step in. @lordfolken maybe you??
The bug is too severe (in my opinion) to leave it there for much longer. (However, I'm surprised nobody noticed it much sooner!)

lordfolken, 2024-05-15:
Dont worry, im angry at this bug. 
"Tactical glide computer" my ass. We cant even get the basics right. 
And I also invested last Sunday into this issue.

ryanetz, 2024-05-17:
Just want to thank you guys for taking this issue seriously. I don’t have the right technical skills to fix this, but my hunch is that somewhere in the label arrival altitude calculation  a + got replace with a - in the wind factor calculation. 
Looking forward for the fix. Meanwhile I ignore the green labels with wind more than 5 knots…

Ramy (who initially reported this issue)

lordfolken, 2024-05-17:
So the commit that introduces the bug is: e020359a9c34e1e4f7727716c24d640b429e8647
This starts to use the function GetDelta()
https://github.com/XCSoar/XCSoar/blob/e020359a9c34e1e4f7727716c24d640b429e8647/src/Engine/Route/RouteLink.hpp#L72-L73

While one could flip those first/second arguments...  GetDelta is much older, and was introduced in 2015: ca1ab2c3549b8e83c403734dd32165791ba4a0e4
GetDelta() is used in CrossProduct() and DotProduct() which are functions used all over the place. 

So I think its best to revert e020359a9c34e1e4f7727716c24d640b429e8647 and not change GetDelta().

closed 2024-05-19 by lordfolken

ryanetz, 2024-06-26:
Thanks again for fixing this critical issue. When will this critical fix be published to users? The issue significantly increases the risk of not reaching landable place in windy conditions as XCSoar will show it as reachable. The windier the conditions the bigger the error. Please publish the fix. Also not clear how the update can be downloaded from the Google store now that only the Jet version is available?

commit 8244b669 (2024-05-19): Revert "Engine/Route/Link: simplify CalcSpeedups()"

This reverts commit e020359a9c34e1e4f7727716c24d640b429e8647.
The GetDelta() function calculates delta between second - first
arguments. The old implementation for CalcSpeedups calculated
the difference however with first - second argument.

This led to bug report #1424 that noticed that the reachable labels
on the map increased altitude with increasing headwind, instead of
the opposite.

Fixes #1424