XCSoar/XCSoar #1425: FLARM altitudes are incorrect when parsing GGA NMEA sentences

twpayne, 2024-05-13:
XCSoar versions having the problem
-------------------------------------------------

All for at least the last 14 years, including the current `master` branch (commit b1244d8dc61e808a3fab3f9d14668aa777e91835 at the time of writing).

XCSoar versions not having the problem
-------------------------------------------------

n/a

System information
------------------

n/a

Steps to reproduce the behavior
-------------------------------

These lines:

https://github.com/XCSoar/XCSoar/blob/b1244d8dc61e808a3fab3f9d14668aa777e91835/src/Device/Parser.cpp#L616-L628

state that FLARM uses MSL altitudes and subtract the geoid separation as computed from EGM96.

This is not correct. FLARM always uses elevation above the WGS84 ellipsoid. The subtraction should be removed.

Expected behavior
-----------------

The altitude is correct.

Actual behavior
---------------

The altitude is incorrect by the value of the geoid separation.

Do you have any idea what may have caused this?
-----------------------------------------------

n/a

Do you have an idea how to solve the issue?
-------------------------------------------

Remove the `else` branch of this `if` statement completely.

lyusupov, 2024-05-16:
@kobedegeest 

**it is obviously a [PEBKAC](https://en.wiktionary.org/wiki/PEBKAC) type of issue invented by the submitter** -

[image]

closed 2024-05-16 by twpayne

twpayne, 2024-05-16:
If your response to bug reports is ad hominem attacks on the bug reporter ... good luck with your project.

DanD222, 2024-05-16:
> If your response to bug reports is ad hominem attacks on the bug reporter ... good luck with your project.

The last comment was not from an XCSoar developer (I am not one either), so please don't be offended.

lordfolken, 2024-05-16:
@twpayne out of interest, what made you think the altitude indicated in XCSoar to be wrong?

lyusupov, 2024-05-17:
@lordfolken 

the statement 

> ... FLARM always uses elevation above the WGS84 ellipsoid ... 

has actually some sense for FLARM **radio**  data emitted by its built-in sub-1GHz ISM band transmitter.

But it has nothing to do with serial data protocol (**FTD-012**)  that FLARM uses for communication with external hardware & software, such as XCSoar.

lordfolken, 2024-05-17:
@lyusupov the [guy](https://github.com/twpayne) has written a lot of gis and geospatial related tools. I suspect he is quite knowledgeable on the topic. ;-)
I suspect he went bug hunting for a reason, and while this particular location probably isn't the cause, i"d like to know what made him hunt for this.

lyusupov, 2024-05-17:
@lordfolken 

Responsible bug reporters are typically very careful before they can make such statements

`All for at least the last 14 years, including the current master branch (commit https://github.com/XCSoar/XCSoar/commit/b1244d8dc61e808a3fab3f9d14668aa777e91835 at the time of writing).`

**With no arguments at all!**

twpayne, 2024-05-20:
Note that all FLARM radio communication uses ellipsoid height, i.e. my goal is ensure that all instruments are reporting the correct ellipsoid height.

I raised this issue when debugging an issue in a flight instrument that uses XCSoar as a reference implementation. In this particular case, the instrument was reporting an altitude roughly 100m different to other instruments at my testing location. At my testing location the geoid separation is 47m and I suspect that this was being subtracted when it should be added, which neatly accounts for the 100m difference.

From further reading on GGA NMEA sentences, I understand that the GGA sentence includes both geoid (AMSL) altitude and geoid separation. When both geoid altitude and geoid separation are positive, I believe that things look something like this (excuse the crude ASCII art):

```
        -oo-+ <----- aircraft
          ^
          :
          : <------- altitude AMSL (i.e. above geoid, say 400m)
          :
          v
----____----____---- geoid (AMSL)
          ^
          : <------- geoid separation (say 47m)
          v
______--------______ elipsoid
```

So therefore, the aircraft's height above the ellipsoid is 400+47 = 447m.

My understanding of the current XC Soar code is that it subtracts the geoid separation, i.e. reports an ellipsoid altitude of 400-47 = 353m.

Note that I do not have access to the official NMEA specification for GGA sentences. My understanding of the altitudes is based on what I've found online.