XCSoar/XCSoar #1440: Color Infoboxes that are Navigation related to the current Target magenta

lordfolken, 2024-05-21:
Both the Track and the related Infobox values should be the same color to make it clear 
Magenta is used for marking navigational targets in most navigation software.

[image]

lordfolken, 2024-08-24:
can you make a pr? @kobedegeest

closed 2025-03-31 by lordfolken

bomilkar, 2025-04-03:
Numbers of several InfoBoxes (including "Next waypoint", "Next altitude", "Next distance", and maybe more) is **always** magenta.
What's the point? Why not keep it black?

styblope, 2025-05-26:
I share the experience of @bomilkar (magenta everywhere) and have mixed feeling about the choice of magenta. I would be nice to let users choose about the color or at least have an option to return to the default.