XCSoar/XCSoar #1442: When Pressing on Infoboxes that have only one option -> open the dialog directly

lordfolken, 2024-05-21:
For example WP AltD Infobox
Short press opens Details of the Waypoint directly.
Long Press opens the current menu that allows also to change the infobox

lordfolken, 2024-05-25:
> I assume this should be some automatic check and not hard code for each individual info box.

Ideally yes. 

> Any hint or idea where this logic should go? I guess there currently is a check if there is an option or not where short press does nothing and long press allows to change the infobox but i do not know where that is handled.

The current box is implemented by srac/Dialogs/dlgInfoBoxAccess.cpp

kobedegeest, 2026-01-19:
@davespragg @Scumi is there a reason for the empty Panel in InfoBoxContentTransponderCode infobox, is there plan for a setup dialog there similar to the radiofrequency one?