XCSoar/XCSoar #1472: Kobo Compliation failure - src/OV/System.cpp - dbus/dbus.h: No such file or directory

lordfolken, 2024-06-02:
```
-j16 TARGET=KOBO USE_CCACHE=y DEBUG=n check everything
  LINK    output/KOBO/bin/TestDriver
  LINK    output/KOBO/bin/xcsoar-ns
  STRIP   output/KOBO/bin/KoboMenu
  STRIP   output/KOBO/bin/PowerOff
  LINK    output/KOBO/bin/test_reach
  LINK    output/KOBO/bin/test_troute
  LINK    output/KOBO/bin/TestTrace
  LINK    output/KOBO/bin/FlightTable
  LINK    output/KOBO/bin/BenchmarkProjection
  LINK    output/KOBO/bin/BenchmarkFAITriangleSector
  LINK    output/KOBO/bin/DumpTextInflate
  LINK    output/KOBO/bin/DumpHexColor
  CXX     output/KOBO/opt/src/OV/System.o
  CXX     output/KOBO/opt/src/OV/OpenVarioMenu.o
  CXX     output/KOBO/opt/test/src/test_route.o
  CXX     output/KOBO/opt/test/src/RunXMLParser.o
  CXX     output/KOBO/opt/test/src/ReadMO.o
  CXX     output/KOBO/opt/test/src/RunMD5.o
  CXX     output/KOBO/opt/test/src/RunSHA256.o
  CXX     output/KOBO/opt/test/src/ReadGRecord.o
  CXX     output/KOBO/opt/test/src/VerifyGRecord.o
  CXX     output/KOBO/opt/test/src/AppendGRecord.o
  CXX     output/KOBO/opt/test/src/FixGRecord.o
  CXX     output/KOBO/opt/test/src/AddChecksum.o
  CXX     output/KOBO/opt/test/src/LoadTopography.o
  CXX     output/KOBO/opt/src/Operation/ConsoleOperationEnvironment.o
In file included from src/OV/System.cpp:5:
./src/lib/dbus/Connection.hxx:7:10: fatal error: dbus/dbus.h: No such file or directory
    7 | #include <dbus/dbus.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make: *** [build/compile.mk:104: output/KOBO/opt/src/OV/System.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from src/OV/OpenVarioMenu.cpp:23:
./src/lib/dbus/Connection.hxx:7:10: fatal error: dbus/dbus.h: No such file or directory
    7 | #include <dbus/dbus.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make: *** [build/compile.mk:103: output/KOBO/opt/src/OV/OpenVarioMenu.o] Error 1
```

lordfolken, 2024-06-02:
Seems to work on ubuntu via github actions...

ubx, 2024-06-02:
Same problem on my Linux box, but it works with:

` make -j1 TARGET=KOBO USE_CCACHE=y DEBUG=n check everything`

Is it a sync problem?

lordfolken, 2024-06-02:
Yes, i think so. My theory is as follows:

DBUS is built for the OV menu, but the depends is missing in 
https://github.com/XCSoar/XCSoar/blob/master/build/kobo.mk#L41
so in parallel builds, some components might be attempted to be built too fast.

lordfolken, 2024-06-02:
> Yes, i think so. My theory is as follows:
> 
> DBUS is built for the OV menu, but the depends is missing in https://github.com/XCSoar/XCSoar/blob/master/build/kobo.mk#L41 so in parallel builds, some components might be attempted to be built too fast.

Doesn't look like its the case.

ubx, 2024-06-03:
> Seems to work on ubuntu via github actions...

I see only this command for the KOBO build in the actions:

`make -j$(nproc) TARGET=KOBO DEBUG=n USE_CCACHE=y V=2 output/KOBO/KoboRoot.tgz`

Did I miss something?

lordfolken, 2024-08-23:
You are correct. The compile command was wrong.

closed 2024-08-23 by lordfolken