XCSoar/XCSoar #1487: Defaulted equality comparison may not work as expected

bomilkar, 2024-06-07:
XCSoar versions having the problem
-------------------------------------------------

7.42 in combination with compiler version g++-10 .

XCSoar versions not having the problem
-------------------------------------------------

The issues was introduced when specific C++20 language features were used.

System information
------------------

Affected are platforms with older compilers, for instance g++-10. But also platforms with g++-12 are affected.

Steps to reproduce the behavior
-------------------------------

Issue #1253 demonstrates an issue in one place. Other places might be affected, too.

Expected behavior
-----------------

Actual behavior
---------------

Do you have any idea what may have caused this?
-----------------------------------------------

Some problems have been reported that are due to shortcomings in certain compilers. For example, it turned out that Issues #1253  is compiler version specific: g++ version 10.2 does not implement the default “operator==” construct in the same way as g++ version 12. This becomes clear if you replace the default “operator==” with the explicit “operator==” declaration.

One additional issue is the declaration of the inequality operator „operator!=“ together with the declaration of the „operator==“ be it defaulted or not. The reference manual says “per the rules for operator==, this will also allow inequality testing.” In other words declaring the equality operator implicitly declares the inequality operator. The reference manual is not explicit what happens if both operators are declared.

These are the affected files:
src/Geo/Flat/FlatGeoPoint.hpp
src/Geo/GeoPoint.hpp
src/Math/Point2D.hpp
src/PageSettings.hpp

Do you have an idea how to solve the issue?
-------------------------------------------

To address this second issue I recommend to remove the declarations of the inequality operator where it comes together with the corresponding equality comparison. This will remove ambiguity.

To address the first issue I recommend to replace the defaulted equality comparison by the explicit declaration. (If the defaulted equality comparison causes bad code.) We can fix it either here in the xcsoar code or we can create a patch which should be applied on platforms which are still on g++-10. One affected platform is OpenVario. Are there any others (we should care about)?

Details to the “default comparisons” are here: https://en.cppreference.com/w/cpp/language/default_comparisons

lordfolken, 2024-06-07:
@August2111 heads up to you, as this affects ov builds.

bomilkar, 2024-06-07:
> Can we revert these commits? Documentation also mentions GCC 10.

I wouldn't revert these commits (entirely). So far there is no indication any of the defaulted equality operators is incorrectly implemented by g++-12, which is the current compiler version on Debian 12. However, I would make a PR which removes all the inequality operators, since this is what the LRM suggests.

bomilkar, 2024-06-07:
> @August2111 heads up to you, as this affects ov builds.

Since docker/bitbake will continue with g++-10 for some time we should add a patch to the OV build which changes the defaulted equality operators to explicit implementation. And I would change all 5 or 6 instances which are currently in the code although we know that only 2 are incorrect. Explicit implementation won't hurt.
As I said affected files are:
src/Geo/Flat/FlatGeoPoint.hpp
src/Geo/GeoPoint.hpp
src/Math/Point2D.hpp
src/PageSettings.hpp

bomilkar, 2024-06-09:
I have added PR https://github.com/Openvario/meta-openvario/pull/376 to replace the defaulted "operator==" with explicit declarations. That fixes the issue for Openvario as long as the build container is still on g++-10.

lordfolken, 2024-06-10:
Looking at the explicity declartions in your ov patch: I think explicitly declaring them in xcsoar would be not a bad idea.

bomilkar, 2024-06-11:
> Looking at the explicity declartions in your ov patch: I think explicitly declaring them in xcsoar would be not a bad idea.

@lordfolken : Just to be clear: you want me to NOT patch OV, instead revert the defaulted comparisons in XCSoar code.
If that is what you want then we should **revert the commits** (as you suggested earlier) and then remove the inequality declarations in **one additional commit**.
Agree?

bomilkar, 2024-06-14:
Changing to code to a style supported by older compilers (g++-10 and g++11) is unacceptable:
https://github.com/Openvario/meta-openvario/pull/376
Therefor I drop this PR and place a new one which just removes the excessive inequality operators.

closed 2024-06-14 by bomilkar