XCSoar/XCSoar #1488: Compile  with  OPTIMIZE="-O0"

ubx, 2024-06-07:
```
 make  -j$(nproc) OPTIMIZE="-O0"
...
...
  LINK    output/UNIX/bin/xcsoar
/usr/bin/ld: output/UNIX/dbg/src/Gauge/LogoView.o:/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/src/Gauge/LogoView.o:(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/src/ProgressWindow.o:/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/src/ProgressWindow.o:(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/src/Dialogs/SimulatorPromptWindow.o:/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/src/Dialogs/SimulatorPromptWindow.o:(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/src/Dialogs/dlgCredits.o:/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/src/Dialogs/dlgCredits.o:(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/libtopo.a(Index.o):/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/libtopo.a(Index.o):(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/liblook.a(DialogLook.o):/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/liblook.a(DialogLook.o):(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/liblook.a(VarioLook.o):/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/liblook.a(VarioLook.o):(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/liblook.a(MapLook.o):/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/liblook.a(MapLook.o):(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/liblook.a(OverlayLook.o):/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/liblook.a(OverlayLook.o):(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/liblook.a(AirspaceLook.o):/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/liblook.a(AirspaceLook.o):(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/liblook.a(TaskLook.o):/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/liblook.a(TaskLook.o):(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/liblook.a(TrafficLook.o):/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/liblook.a(TrafficLook.o):(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/liblook.a(WaypointLook.o):/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/liblook.a(WaypointLook.o):(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/liblook.a(NOAALook.o):/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/liblook.a(NOAALook.o):(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/liblook.a(IconLook.o):/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/liblook.a(IconLook.o):(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
/usr/bin/ld: output/UNIX/dbg/liblook.a(ClimbPercentLook.o):/media/andreas/data2/workspace2/XCSoar/./output/UNIX/include/MakeResource.hpp:1:(.data.rel.ro+0x0): undefined reference to `resource_IDI_XCSOAR'
/usr/bin/ld: output/UNIX/dbg/liblook.a(ClimbPercentLook.o):(.data.rel.ro+0x8): undefined reference to `resource_IDI_XCSOAR_size'
collect2: error: ld returned 1 exit status
make: *** [build/main.mk:685: output/UNIX/bin/xcsoar] Error 1
(

```

lordfolken, 2024-06-08:
Interesting. Could this be related to the icon backtrace in #1416

lordfolken, 2024-06-08:
This resource should only be used on windows targets.
Is this reproducible with a clean build?

ubx, 2024-06-09:
> Is this reproducible with a clean build?

The problem appears only on TARGET=UNIX, WAYLAND, and KOBO. Others build fine.

ubx, 2024-06-09:
git bisect with `make -j18 TARGET=UNIX OPTIMIZE="-O0"`:

```
c079bbf566d071de70c59d8f061269fc38636a32 is the first bad commit
commit c079bbf566d071de70c59d8f061269fc38636a32
Author: Max Kellermann <max.kellermann@gmail.com>
Date:   Thu Feb 29 14:34:46 2024 +0100

    Resources.hpp: move MAKE_RESOURCE() lines to generated header
    
    This auto-generates the Windows resource ids and eliminates redundant
    data that is already in `Data/resources.txt`.

 Makefile                          |   4 +
 build/generate.mk                 |   2 +-
 build/resource.mk                 |  26 ++++---
 src/Resources.hpp                 | 149 +-------------------------------------
 tools/GenerateAndroidResources.pl |   4 +-
 tools/GenerateMakeResource.pl     |  24 ++++++
 tools/LinkResources.pl            |   5 +-
 7 files changed, 49 insertions(+), 165 deletions(-)
 create mode 100755 tools/GenerateMakeResource.pl
```

lordfolken, 2024-06-10:
I cannot reproduce the build failure, however that resource was only in use on GDI targets before the mentioned commit changed this to all targets. 
Does 8bbd752fd2cfcb9ef2b59a0eee56e90e26489455 fix this for you?

closed 2024-06-10 by lordfolken