XCSoar/XCSoar #1503: TMZ colour coding different in 7.42 - 7.43

fraktal-sb, 2024-06-14:
XCSoar versions having the problem
-------------------------------------------------

v7.43.rc0

XCSoar versions not having the problem
-------------------------------------------------

v7.42

System information
------------------

Kobo GLO-HD

Steps to reproduce the behavior
-------------------------------

take airspace data from

https://www.daec.de/fachbereiche/luftraum-flugsicherheit-betrieb/luftraumdaten/

Expected behavior
-----------------

color coding should equal over versions

Actual behavior
---------------

colour coding is different, see screenshots

[image]

example:

AC TMZ
AN TMZ-EDFH A0424 125.600
AH FL75
AL 4500ft MSL
DP 50:05:06 N 007:12:19 E 
.
.
.

data from 2023 and 2024 do not differ

Do you have any idea what may have caused this?
-----------------------------------------------

no

Do you have an idea how to solve the issue?
-------------------------------------------

no

fraktal-sb, 2024-06-14:
settings have not been changed between versions. There was no message at launch of 7.43. My opinion: TMZ colour coding should be the same over the versions

lordfolken, 2024-06-14:
A TMZ is an unclassified airspace, in the sense that it isn't an ICAO class airspace.
The coloring represents this now correctly as previously it would label it the same as a controlled airspace.

fraktal-sb, 2024-06-14:
but why is it presented differently now?

Scumi, 2024-06-14:
So #1349 broke backwards compatibility in a way?

> A TMZ is an unclassified airspace, in the sense that it isn't an ICAO class airspace. (separation rules, clearance, traffic information, atc contact) The coloring represents this now correctly as previously it would label it the same as a controlled airspace.

While it might be more correct now, it will confuse a lot of users who are used to a certain coloring of the airspaces. If I understand correctly, there is no (or only one) default coloring for the airspace types? Maybe the best way would be to assign a similar coloring to ED-Rs, TMZs and RMZs as before the change of #1349. Then only users have to change their configs if they changed the colors on their end (which would still be bad in my view - see backwards compatibility).

lordfolken, 2024-06-14:
> but why is it presented differently now?
because you are supposed to act different when flying into an icao class airspace, as opposed to not.

lordfolken, 2024-06-14:
> Edit: I just compiled the latest version and without any changes I personally could not use the OpenAIP airspaces. Germany is just covered in two colours.

Missing FIS_SECTOR airspace type. 

https://github.com/XCSoar/XCSoar/issues/1504

fraktal-sb, 2024-06-14:
> I don't think it is an issue that user have to go to there settings after this update. They just need to be told to do so.

I think users will not do that. 95% are happy if everything continues to work after an update. But it has to be decided

lordfolken, 2024-06-14:
AC TMZ
No AY field

[image]

AC UNCLASSIFIED
AY TMZ

[image]

lordfolken, 2024-11-25:
is this still an issue, or can we consider it fixed?

lordfolken, 2024-12-02:
Would you have a look and please suggest or PR the changes ?

fraktal-sb, 2024-12-03:
This is my proposal based on change in AirspaceRenderer where we can find the default colour assignment. I changed only the line for TMZ coloring from bluegrey to red accordingly. For my "taste" the red colour gives the user a more important impression of the TMZ airspaces then the the light-grey-blue colour. When we look for example at the official airspace maps of Germany and France there is no common standard in cartography. So this is also only one additional colour coding more.

The proposed colouring is on the right side of the screenschot (on the left we have 7.42)

[image]

fraktal-sb, 2024-12-14:
we should not leave TMZ as it is. Now it has the same colour code as Class-G airspace. It should have RGB8_BLUE. See also the comparision of all old and new colours in the attached excel table
[ColourTable.xlsx](https://github.com/user-attachments/files/18134655/ColourTable.xlsx)

fraktal-sb, 2024-12-16:
OK, I understand. We can close this issue. But before I rise an other one I will ask in the forum wether everybody is happy with the current color coding. So we can collect which airspaces should come to an other colour group.

closed 2024-12-16 by fraktal-sb