XCSoar/XCSoar #1507: Energy management at statt

GeorgeDowning20, 2024-06-15:
Which XCSoar version are you using?
7

What should XCSoar do differently, what functionality should be added?

Start line Energy management is now annex A. Energy management requires you to be below a given ground speed and altitude at the start line. This is very difficult to perform with xcsoar at the moment. 

Perhaps a nav box that calculates arrival altitude at the start line by intersecting ground track with the line. It should also show ground speed. Ideally be highlighted green when both conditions are satisfied.

lordfolken, 2024-06-15:
Can you link to this rule update?

lordfolken, 2024-06-15:
Its true. Currently you just don't get a valid start when setting the rules for speed / alt in expert settings.

lordfolken, 2024-06-15:
Want to try to design a an infobox, or similar? 
src/Monitor/TaskContraintsMonitor.cpp particulary TaskConstraintsMonitor::Check() may be of interest. 
But in the end you probably wan't to have a glide solution to the start point taking speed / polar / altitude into account.

GeorgeDowning20, 2024-06-15:
> Can you link to this rule update?

https://www.fai.org/sites/default/files/sc3a_2023.pdf

7.4.6
Energy Control at the Start
The total energy of the glider as it crosses the
Start Line or Ring will be controlled as follows:
a.
Maximum Start Altitude
The Organisers shall determine a Maximum Start
Altitude (MSA) for each class. An announcement of the Maximum Start
Altitude MSL shall be included with each radio announcement concerning the
opening of the Start for that class.
The maximum start altitude should normally be at least 100 m below cloudbase or top
of lift and chosen to allow a fair start for all competitors. The maximum start altitude
should be an integral multiple of 100 in the altitude units used.
The procedures for communicating the Maximum Start Altitude to the Teams
must be specified in the Local Procedures.
The MSA may be set at Briefing, on the grid before the launch begins, or before the
first announcement of the opening of the Start. The Local Procedures will explain
when the MSA will be announced and when it may be updated.
FAI SC3 ANNEX A
Page 31A Start higher than the Maximum Start Altitude will be penalized.
b.
Maximum Start Groundspeed
For each class, the Maximum Start
Groundspeed must be announced at Briefing and included on the Task Sheet.
The glider’s groundspeed at the start will be determined by the straight line
distance between the pair of fixes nearest to eight seconds before and after
the Start, divided by the elapsed time between those fixes. A Start at a
groundspeed greater than the Maximum Start Groundspeed will be penalized.
The Maximum Start Groundspeed should be an integral multiple of 10 in the speed
units used. The Maximum Start Groundspeed cannot be changed after Briefing without
notification of the Team Captains. The Organisers should choose a maximum
groundspeed taking into account the wind component on the first leg of the Task. The
“no wind” value of the Maximum Start Groundspeed should be at least 170 kph (or the
equivalent in the speed units used).

GeorgeDowning20, 2024-06-15:
> Want to try to design a an infobox, or similar? src/Monitor/TaskContraintsMonitor.cpp particulary TaskConstraintsMonitor::Check() may be of interest. But in the end you probably wan't to have a glide solution to the start point taking speed / polar / altitude into account.

I will have a go.