XCSoar/XCSoar #1512: Impossible d'utiliser un fichier personnel

Jean-Yves-Fayence, 2024-07-02:
Depuis moins d'un mois je n'ai plus accès à mes fichiers Wpts et Espace aérien. Xcsoar 7jet pointe sur Android/data/com.zinuzoid qui n'est pas accessible depuis les gestionnaires de fichiers des téléphones Android.

Welcome! - We kindly ask that you:

  1. Fill out the issue template below - not doing so needs a good reason.
  2. Use the forum if you have a question rather than a bug or feature request.

The forum is at: https://forum.xcsoar.org

Thanks for understanding, and for contributing to the project!

-->

XCSoar versions having the problem
-------------------------------------------------

XCSoar versions not having the problem
-------------------------------------------------

System information
------------------

Steps to reproduce the behavior
-------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

mlep, 2024-07-02:
Pour poser vos questions en français à propos du fonctionnement d'XCSoar, nous avons un forum francophone : [https://forum.xcsoar.org/viewforum.php?f=15](https://forum.xcsoar.org/viewforum.php?f=15).
En version courte : Android segmente plus qu'avant les répertoires (certains devenant inaccessibles à certaines applications). Peut-être est-ce lié ?   (la suite de la discussion sur le forum).

Jean-Yves-Fayence, 2024-07-02:
En effet vous avez raison. La nouvelle version XCSOAR 7jet s'installe dans Android/DATA alors que les précédentes versions s'installaient dans storage/emulated/0/XCSoarData. Android interdit maintenant l'accès à son répertoire système (ce qui est logique) et on ne peut plus installer ses fichiers personnels (champ à vaches, points caractéristiques de la région, Airspace planeur…). Donc à court terme au fur et à mesure des mises à jours des téléphones, plus aucun XCSOAR ne fonctionnera et nous somme en pleine saison de vol.

Il faut que je migre le plus vite possible sur une autre application avant que mon vieux téléphone tombe en panne et que je me retrouve sans logiciel de navigation pour voler. Dommage j'adorait XCSOAR!

Cordialement, Jean-Yves Clément.

_____________________UK version

Indeed you are right. The new XCSOAR 7jet version is installed in Android/DATA while previous versions were installed in storage/emulated/0/XCSoarData. Android now prohibits access to its system directory (which is logical) and you can no longer install your personal files (cow fields, characteristic points of the region, Airspace glider, etc.). So in the short term as the phones are updated, no XCSOAR will work and we are in the middle of the flight season.

I need to migrate to another app as quickly as possible before my old phone breaks and I find myself without navigation software to fly. Too bad I loved XCSOAR!

Kind regards, Jean-Yves Clément.

----- Mail original -----

mlep, 2024-07-02:
Je n'utilise pas XCSOAR Jet.
Sur la page www.xcsoar.org, il y a un lien de téléchargement vers le dépôt f-droid [https://f-droid.org/en/packages/org.xcsoar/](https://f-droid.org/en/packages/org.xcsoar/).

Dans tous les cas, je serais très surpris que l'on ne puisse pas utiliser de fichiers personnels. Pour ces cas-là, il y a le forum.
Merci de fermer le sujet.

closed 2024-07-21 by lordfolken