XCSoar/XCSoar #1514: Graphic bug: lines strarting from upper left corner

Lucaborro97, 2024-07-05:
[image]

Just appeared this graphic bug that consist of this lines strarting from upper left corner of the phone (android samsung a55) finishing on the elements of the map i dont know what to do.

lordfolken, 2024-07-05:
Thanks for opening a bug.
What is needed is the xcsoar.log, the exact Version of xcsoar you are using, the exact phone Name and Android Version, as stated in the bug template.

Does it go away when you turn off topology?

Lucaborro97, 2024-07-05:
[xcsoar[1].log](https://github.com/user-attachments/files/16108680/xcsoar.1.log)

phone name: Samsung Galaxy A55 5g
android vers: 14
xcsoar vers: 7.42

yes i turn off the topology and it seems like there's still 1 or 2 lines, and i saw that if i add more and more elements like airspaces and labels the lines multiply..

Gregy, 2024-07-23:
I have the exact same issue. It only occurs in the "map panning mode" and only if there are airports in the map view (or slighly around it). The lines start in the top left corner and go to each airport. It doesn't appear always and sometimes it randomly disappears when panning. The most consistent way to reproduce for me is to zoom out to show a large amounts of airports during panning.

[image]

Phone: Samsung Galaxy s22 (SM-S901B/DS)
Android: 14
XCSoar: 7.42

Unfortunately my XCSoar doesn't seem to generate any xcsoar.log in the xcsoarddata directory so I cannot attach it :(

dssherrill, 2024-07-23:
The log files are now in **/Android/data/org.xcsoar/files**.

Gregy, 2024-07-23:
That directory is inaccessible. "To protect your privacy choose another folder". No way for me to copy out the log file :(

Sundown3867, 2025-01-25:
Today I saw it on friend's Samsung Galaxy A55. Lines disapear if you deactivate Map Display - Waypoints - Detailed landables

lordfolken, 2026-01-11:
@coderabbitai can you use the bug report attached xcsoar-flarm.txt and the renderer for the detailed landables, to investigate why lines are drawn from the top left of the screen to the top left of the shape. Note this seems to happen to shapes that are defined as a BulkPixelPoint such as https://github.com/XCSoar/XCSoar/blob/master/src/Renderer/TrafficRenderer.cpp#L25 and later drawn with DrawPolygon. Please note that they are also rotated with PolygonRotateShift(). 
the main occurrence is traffic rendering and detailed landables. 

The issue seems to affect mainly the android build with the samsung devices.

lordfolken, 2026-01-11:
@coderabbitai is it possible that this is a driver bug? This doesn't seem to happen on other android socs.