XCSoar/XCSoar #1524: Android 14: not able to copy airspace and waypoint files into the XCsoar folder

TillBerthold, 2024-08-03:
XCSoar versions having the problem
On Android 14 it is not possible to copy something into Android/data/org.xcsoar/files because this folder is blocked from Android

 7.42

It worked on Android 13

System information
Android 14, Xiaomi 14

Expected behavior
Would like to be able to copy WP files and Airspace

lordfolken, 2024-08-04:
are you using the files app from playstore, and the documents provider in the files app?

lordfolken, 2024-08-04:
Also which airspace /waypoint files are missing from the  online repository (File manager)?

TillBerthold, 2024-08-04:
I tried with different apps from play store and with the file app from the phone.
I need the waypoints from the local competition.

lordfolken, 2024-08-05:
And you are using the documents provider "files app" in https://play.google.com/store/apps/details?id=com.marc.files
Then hamburger menu, and then xcsoar?

TillBerthold, 2024-08-05:
I tied with the files app. With the app I can show the folder but can not copy anything to the xcSoar folder.
What do you mean with "Then hamburger menu, and then you use the xcsoar menu entry to manage files?"

schonstudent, 2024-08-05:
Copy the file to the cache folder with your computer. After copy the file from the cache folder to the final folder.... Had the same issue...

TillBerthold, 2024-08-05:
After all I managed with a windows computer. This is a very bad workaround. 
Is there a fix in sight? Can someone change the folder to a public location?

schonstudent, 2024-08-05:
Looks like this is more an android14 security issue....

TillBerthold, 2024-08-06:
It is because xcsoar is not following the android security policy. It is not the right path for the file folder like this.

micheal813, 2024-08-07:
Yes, it would be great if this issue could be fixed. I shouldn't have to connect my phone to a PC to copy files to the XCSoar folder. I can't access my log igc files unless I connect it to a PC. There should be options for a custom folder that contain files like maps or task files or some other fix for this issue.

TillBerthold, 2024-08-07:
There is no option to copy files to this folder.

TillBerthold, 2024-08-07:
No this does not work on android 14.

lordfolken, 2024-08-07:
Uploading 2024-08-07_17-48-03.mp4…

jorgemarmor, 2024-08-07:
Newest Android versions are hardening apps folders. I think the only clean option would be to move the XCSoar folder from Android/data to Android/media, which is open and easy to handle without workarounds.

lordfolken, 2024-08-08:
> Newest Android versions are hardening apps folders. I think the only clean option would be to move the XCSoar folder from Android/data to Android/media, which is open and easy to handle without workarounds.

Unfortunately this is a deprecated API call:
https://developer.android.com/reference/android/content/Context#getExternalMediaDirs()

I however see that whatsapp,telegram,nextcloud also use this path.

Nevertheless I made a PR with it and it seems to work. Please test: https://github.com/XCSoar/XCSoar/pull/1528

jorgemarmor, 2024-08-09:
Sorry, not very used to this... How can I test it?

lordfolken, 2024-08-09:
The GitHub action builds an apk you can use: 
https://github.com/XCSoar/XCSoar/actions/runs/10309426528 (artifacts at the bottom)

This should work with any android version that xcsoar supports. 6+. I tested 13/14.

lordfolken, 2024-08-09:
You need to uninstall your current version as this one has a,different signing key. (Working on that)

jorgemarmor, 2024-08-09:
Excellent. Installed and working. In this mode I can access to the foder with any app, without needing Shizuku. Great job!

micheal813, 2024-08-10:
Thank you for the fix.

lordfolken, 2024-08-13:
Merged in 7.43

closed 2024-08-13 by lordfolken