XCSoar/XCSoar #1535: Device WayTec Android 6.0 - XCSoar crash on startup when initialising display

pkreinh, 2024-08-11:
XCSoar versions having the problem
-------------------------------------------------
XCSoar JET Version 7.43_JET_1.23

-------------------------------------------------

System information
------------------
7

WayTec x995, Android 6 Kernel 3.18.22, Buid MRA58K

Steps to reproduce the behavior
-------------------------------
Crashes on startup

after getting the update from app store, XCsoar does crash on startup when initialising display. I did un- and reinstall XCsoar, but it didn't solve the issue.

Best regards
Peter

pkreinh, 2024-08-24:
I installed XCSoar.apk from the releases of 7.43

[log_2024_08_24_17_30_51.log](https://github.com/user-attachments/files/16737087/log_2024_08_24_17_30_51.log)
[log_2024_08_24_17_28_43.log](https://github.com/user-attachments/files/16737088/log_2024_08_24_17_28_43.log)
[xcsoar.log](https://github.com/user-attachments/files/16737089/xcsoar.log)

From WayTeQ tablet, see only this informations
Model x995
Android 6.0
Patch 01. May 2016
Kernel 3.18.22
xly1@ubuntu640a #1
Thu Dec 31 14:16:12 CST 2020
Build MRA58K test-keys
Build alps-mp-m0.mp8-V1.69.1_rmt8127.ce_P8

HTH
Peter

lordfolken, 2024-08-29:
Can you please test the attached PR? @pkreinh

closed 2024-09-02 by lordfolken