XCSoar/XCSoar #1538: editing a waypoint removes the first waypoint of the list after restart

acollign, 2024-08-15:
XCSoar versions having the problem
-------------------------------------------------

local build of commit a7770

System information
------------------

Ubuntu Noble Numbat

Steps to reproduce the behavior
-------------------------------

The steps below assume a local build was produced under `output/UNIX`.

- go to the root folder with `cd output/UNIX`
- create a data folder `mkdir data`
- in that folder, create a file user.cup with the following content
```
name,code,country,lat,lon,elev,style,rwdir,rwlen,rwwidth,freq,desc,userdata,pics
"Waypoint 1",,,463.228N,641.856E,1588M,20,,,,"Waypoint 1"
"Waypoint 2",,,466.936N,643.267E,1740M,20,,,,"Waypoint 2"
"Waypoint 3",,,464.836N,642.918E,689M,20,,,,"Waypoint 3"
```
- From `output/UNIX` folder, run XCSoar as follows `bin/xcsoar -datapath=data`
- Click `Sim`
- Open `Nav > Waypoint List`
- Click on `Waypoint 1`
- Click on `Right Arrow`
- Click on `Edit`
- Change Short Name to `TOUCHED`
- Click `OK`
- Close XCSoar
- Run XCSoar as follows `bin/xcsoar -datapath=data` 
- Open `Nav > Waypoint List`

Expected behavior
-----------------

Waypoint list contains three waypoints.

Actual behavior
---------------

Waypoint list contains only two waypoints

Do you have any idea what may have caused this?
-----------------------------------------------

I believe the problem comes from the fact that WriteCup from `CupWriter.cpp` writes waypoints directly to the `user.cup` file while `ParseSeeYou` defined in `WaypointReaderSeeYou.cpp` reads the first line of the `user.cup` file as an header, see line 205. This makes `ParseSeeYou` ignores the first waypoint written before.

Do you have an idea how to solve the issue?
-------------------------------------------

I think the trick is to align the behavior of the reader and writer.

acollign, 2024-08-15:
I can confirm that a) the file written upon exit does not contain the header b) adding an header like below seems to fix the issue.

````
diff --git a/src/Waypoint/CupWriter.cpp b/src/Waypoint/CupWriter.cpp
index 53602d585a..495b70a8e3 100644
--- a/src/Waypoint/CupWriter.cpp
+++ b/src/Waypoint/CupWriter.cpp
@@ -169,6 +169,7 @@ void
 WriteCup(BufferedOutputStream &writer, const Waypoints &waypoints,
          WaypointOrigin origin)
 {
+  writer.Write("name,code,country,lat,lon,elev,style,rwdir,rwlen,rwwidth,freq,desc,userdata,pics\n");
   // Iterate through the waypoint list and save each waypoint with
   // the right file number to the BufferedOutputStream
   /// @todo JMW: iteration ordered by ID would be preferred

````

The patch is rather simple and I am happy to open a PR. I didn't do it so far as I could not find the guidelines for running and writing tests.

lordfolken, 2024-08-23:
We don't really have guidelines for tests. A test for this would be highly appreciated. 
The writer has a test here: https://github.com/XCSoar/XCSoar/blob/master/test/src/TestWaypointReader.cpp

lordfolken, 2024-09-02:
Fixed in https://github.com/XCSoar/XCSoar/commit/404362d3577a1f579dd267060e03970cf69f6efe