XCSoar/XCSoar #1540: Editing waypoint may truncate comment

acollign, 2024-08-15:
XCSoar versions having the problem
-------------------------------------------------

local build of commit a7770

System information
------------------

Ubuntu Noble Numbat

Steps to reproduce the behavior
-------------------------------

The steps below assume a local build was produced under `output/UNIX`.

- go to the root folder with `cd output/UNIX`
- create a data folder with `mkdir data`
- in that folder, create a file `user.cup` with the following content
```
name,code,country,lat,lon,elev,style,rwdir,rwlen,rwwidth,freq,desc,userdata,pics
"Waypoint 1",,,463.228N,641.856E,1588M,20,,,,,"Waypoint 1 has a long description which will be truncated"
```
- From `output/UNIX` folder, run XCSoar as follows `bin/xcsoar -datapath=data`
- Click `Sim`
- Open `Nav > Waypoint List`
- Click on `Waypoint 1`
- Click on `Right Arrow`
- Click on `Edit`
- Change Short Name to `TOUCHED`
- Click `OK`
- Open `Nav > Waypoint List`

Expected behavior
-----------------

Waypoint 1 has the comment "Waypoint 1 has a long description which will be truncated"

Actual behavior
---------------

Waypoint 1 has the comment "Waypoint 1 has a long descripti"

Do you have any idea what may have caused this?
-----------------------------------------------

I am guessing the problem comes from the size of the editable field of the comment which may be limited to 32 characters.

Do you have an idea how to solve the issue?
-------------------------------------------

If I am correct about the cause, then the comment field must be made long enough, ideally without any limitation in length as stated by the CUP specification for the description field [0].

[0] https://downloads.naviter.com/docs/SeeYou_CUP_file_format.pdf

ubx, 2024-09-04:
The problem here is that the edit field has a maximum of [31 characters](https://github.com/XCSoar/XCSoar/commit/7c4f6893a2fcb5aef0932014c1d11d1cd8ad6435).

A quick fix to solve the issue would be to make the **Comment** field read-only, but obviously, it would be immutable, which may be acceptable.

lordfolken, 2026-02-08:
@coderabbitai which commit fixed this?