XCSoar/XCSoar #1541: add support for long waypoint comment

acollign, 2024-08-15:
XCSoar version
--------------

- XCSoar v7.32-Kobo
- local build of commit a7770

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

You could make the waypoint comment field bigger so it allows displaying long comment/description. This applies to both the view and edit dialogs for waypoint, but should be considered on the view dialog first if deemed valid. 

It seems to me that there is generally enough screen estate to expand the comment field quite a bit without disrupting current usage. Alternatively, the comment could be displayed in a text area with scrollbars, but that would need extra interactions which may not be that easy to perform while flying.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

I am currently importing CUP file of paragliding takes off and landing zones with highly detailed description, imported as waypoint comment. At the moment, the waypoint view dialogs does not allow to read the comments entirely, only the first 4 lines. Less importantly, the edit dialog does not allow to edit the comment as it only shows the first characters in an input field.

lordfolken, 2024-08-22:
We support the waypoint_details file. Described in the manual https://download.xcsoar.org/releases/7.42/XCSoar-manual.pdf
page 164 / Section 14.5. 

This file adds additional pages to the waypoint details dialog to display that information. 
We also support pictures with the image= tag.

acollign, 2024-09-06:
> We support the waypoint_details file.

Thanks for the tip. I would need to generate the waypoint_details file from the cup file, that is not complicated but not very practical. Having the possibility to only import a cup file offers a better user experience.

lordfolken, 2024-09-06:
One of the next features needs to be this

https://github.com/XCSoar/XCSoar/issues/1025

Then finally it wold be easy to supply a file to the enduser with detailed info of the waypoint. Including pictures etc.