XCSoar/XCSoar #1542: XCSoar v7.43 on Samsung Galaxy Tab A9 crashes after starting SIM or FLY mode

sasam, 2024-08-17:
XCSoar versions having the problem
-------------------------------------------------
XCSoar v7.43 &  v7.42

XCSoar versions not having the problem
-------------------------------------------------
XCSoar v7.41

System information
------------------
Samsung Galaxy Tab A9
Android: 14
One UI version: 6.1

Steps to reproduce the behavior
-------------------------------
Samsung Galaxy Tab A9 crashes after pressing the SIM or FLY button. 

Expected behavior
-----------------
Xcsoar should start normally after pressing the FLY or SIM button.

Actual behavior
---------------
Samsung Galaxy Tab A9 crashes after pressing the SIM or FLY button. 
It briefly appears on the screen to initialize the display and then the application exits without any messages. I didn't find anything in the xcsoar.log that would indicate an error.

xcsoar.log:
```
[2024-08-17T10:35:29Z] Starting XCSoar XCSoar v7.43-Android
[2024-08-17T10:35:29Z] EGL vendor: Android
[2024-08-17T10:35:29Z] EGL version: 1.4 Android META-EGL
[2024-08-17T10:35:29Z] EGL extensions: EGL_ANDROID_front_buffer_auto_refresh EGL_ANDROID_get_native_client_buffer EGL_ANDROID_presentation_time EGL_EXT_surface_CTA861_3_metadata EGL_EXT_surface_SMPTE2086_metadata EGL_KHR_get_all_proc_addresses EGL_KHR_swap_buffers_with_damage EGL_ANDROID_get_frame_timestamps EGL_ANDROID_image_native_buffer EGL_ANDROID_native_fence_sync EGL_ANDROID_recordable EGL_EXT_create_context_robustness EGL_EXT_image_gl_colorspace EGL_EXT_pixel_format_float EGL_EXT_protected_content EGL_EXT_yuv_surface EGL_IMG_context_priority EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_fence_sync EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_mutable_render_buffer EGL_KHR_no_config_context EGL_KHR_partial_update EGL_KHR_surfaceless_context EGL_KHR_wait_sync EGL_NV_context_priority_realtime 
[2024-08-17T10:35:29Z] EGL config: RGB=8/8/8 alpha=8 depth=24 stencil=8
[2024-08-17T10:35:29Z] GL vendor: ARM
[2024-08-17T10:35:29Z] GL version: OpenGL ES 3.2 v1.r32p1-01eac0.3d0f892e790e4319a235910ce1559a8a
[2024-08-17T10:35:29Z] GL renderer: Mali-G57 MC2
[2024-08-17T10:35:29Z] GL extensions: GL_EXT_debug_marker GL_ARM_rgba8 GL_ARM_mali_shader_binary GL_OES_depth24 GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_OES_packed_depth_stencil GL_OES_rgb8_rgba8 GL_EXT_read_format_bgra GL_OES_compressed_paletted_texture GL_OES_compressed_ETC1_RGB8_texture GL_OES_standard_derivatives GL_OES_EGL_image GL_OES_EGL_image_external GL_OES_EGL_image_external_essl3 GL_OES_EGL_sync GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_required_internalformat GL_OES_vertex_array_object GL_OES_mapbuffer GL_EXT_texture_format_BGRA8888 GL_EXT_texture_rg GL_EXT_texture_type_2_10_10_10_REV GL_OES_fbo_render_mipmap GL_OES_element_index_uint GL_EXT_shadow_samplers GL_OES_texture_compression_astc GL_KHR_texture_compression_astc_ldr GL_KHR_texture_compression_astc_hdr GL_KHR_texture_compression_astc_sliced_3d GL_EXT_texture_compression_astc_decode_mode GL_EXT_texture_compression_astc_decode_mode_rgb9e5 GL_KHR_debug GL_EXT_occlusion_query_boolean GL_EXT_disjoint_timer_query GL_EXT_blend_minmax GL_EXT_discard_framebuffer GL_OES_get_program_binary GL_OES_texture_3D GL_EXT_texture_storage GL_EXT_multisampled_render_to_texture GL_EXT_multisampled_render_to_texture2 GL_OES_surfaceless_context GL_OES_texture_stencil8 GL_EXT_shader_pixel_local_storage GL_ARM_shader_framebuffer_fetch GL_ARM_shader_framebuffer_fetch_depth_stencil GL_ARM_mali_program_binary GL_EXT_sRGB GL_EXT_sRGB_write_control GL_EXT_texture_sRGB_decode GL_EXT_texture_sRGB_R8 GL_EXT_texture_sRGB_RG8 GL_KHR_blend_equation_advanced GL_KHR_blend_equation_advanced_coherent GL_OES_texture_storage_multisample_2d_array GL_OES_shader_image_atomic GL_EXT_robustness GL_EXT_draw_buffers_indexed GL_OES_draw_buffers_indexed GL_EXT_texture_border_clamp GL_OES_texture_border_clamp GL_EXT_texture_cube_map_array GL_OES_texture_cube_map_array GL_OES_sample_variables GL_OES_sample_shading GL_OES_shader_multisample_interpolation GL_EXT_shader_io_blocks GL_OES_shader_io_blocks GL_EXT_tessellation_shader GL_OES_tessellation_shader GL_EXT_primitive_bounding_box GL_OES_primitive_bounding_box GL_EXT_geometry_shader GL_OES_geometry_shader GL_ANDROID_extension_pack_es31a GL_EXT_gpu_shader5 GL_OES_gpu_shader5 GL_EXT_texture_buffer GL_OES_texture_buffer GL_EXT_copy_image GL_OES_copy_image GL_EXT_shader_non_constant_global_initializers GL_EXT_color_buffer_half_float GL_EXT_unpack_subimage GL_EXT_color_buffer_float GL_EXT_float_blend GL_EXT_YUV_target GL_OVR_multiview GL_OVR_multiview2 GL_OVR_multiview_multisampled_render_to_texture GL_KHR_robustness GL_KHR_robust_buffer_access_behavior GL_EXT_draw_elements_base_vertex GL_OES_draw_elements_base_vertex GL_EXT_protected_textures GL_EXT_buffer_storage GL_EXT_external_buffer GL_EXT_EGL_image_array GL_EXT_texture_filter_anisotropic GL_OES_texture_float_linear GL_ARM_texture_unnormalized_coordinates GL_EXT_shader_framebuffer_fetch 
[2024-08-17T10:35:29Z] Display dpi=179,179
[2024-08-17T10:35:29Z] OpenGL: mda=false npot=true stencil=0x88f0
[2024-08-17T10:35:30Z] Startup dialog
[2024-08-17T10:35:30Z] Loading profiles
[2024-08-17T10:35:30Z] Failed to load profile: Failed to open '/storage/emulated/0/Android/media/org.xcsoar/default.prf': No such file or directory
[2024-08-17T10:35:30Z] Loading language file
[2024-08-17T10:35:30Z] Loading input events file
[2024-08-17T10:35:30Z] Loading Topography File...
[2024-08-17T10:35:30Z] ReadWaypoints
[2024-08-17T10:35:30Z] Failed to read waypoint file: /storage/emulated/0/Android/media/org.xcsoar/user.cup
[2024-08-17T10:35:30Z] Failed to open '/storage/emulated/0/Android/media/org.xcsoar/user.cup': No such file or directory
[2024-08-17T10:35:30Z] RASP load
[2024-08-17T10:35:30Z] No rasp data file: Failed to open ZIP archive /storage/emulated/0/Android/media/org.xcsoar/xcsoar-rasp.dat
[2024-08-17T10:35:30Z] ReadAirspace
[2024-08-17T10:35:30Z] Register serial devices
[2024-08-17T10:35:30Z] ProgramStarted
[2024-08-17T10:35:30Z] cannot open /storage/emulated/0/Android/media/org.xcsoar/lua/init.lua: No such file or directory
```

Do you have any idea what may have caused this?
-----------------------------------------------
Maybe something with the initialization of support for Lua scripts is not OK, or maybe there is some problem with using FS, considering that the policy for writing and reading files was changed on newer versions of Android.
A friend of mine after upgrading Android to Samsung S21, Xcsoar files were no longer available. He had to add them again in such a way that it can be connected to the mobile phone via USB. He could not do that using the file manager on his mobile phone.
Due to these new restrictions on Android, a local File manager may need to be implemented on XCSoar, or an existing one will need to be called from XCSoar and thus enable manipulation of files related to the XCSoar application.

Do you have an idea how to solve the issue?
-------------------------------------------
No

lordfolken, 2024-08-17:
Could you please provide a logcat?

lordfolken, 2024-08-22:
Since it crashes after the fly/sim screen, i think what fails is the fullscreen initialization. 
If you watch closely there are several resolution changes when starting either fly or sim. 
The devices do have different resolutions, but they are all generally low dpi (179~) screens. Maybe its somewhere there.

lordfolken, 2024-08-22:
I tested 17 samsung devices. None seem to have a problem:
[image]

sasam, 2024-08-22:
> Could you please provide a logcat?

The logcat is included in the attach file.
I got the file by:
adb logcat -c
adb logcat > logcat_SIM.txt
=> start xcsoar & press SIM button
=> ctrl C (stop logging)

[logcat_SIM.txt](https://github.com/user-attachments/files/16717119/logcat_SIM.txt)

sasam, 2024-08-22:
> Since it crashes after the fly/sim screen, i think what fails is the fullscreen initialization. If you watch closely there are several resolution changes when starting either fly or sim. The devices [...]

In the attach file is the output of the command:
adb shell dumpsys window > adb_shell_dumpsys_window.txt

I think there is enough information about the resolution of the tablet.

[adb_shell_dumpsys_window.txt](https://github.com/user-attachments/files/16718797/adb_shell_dumpsys_window.txt)

sasam, 2024-08-24:
logcat grep by PIDs related to XCSoar

```
	Line   80: 08-22 21:57:49.197   791   791 D Zygote  : Forked child process 15251
	Line   81: 08-22 21:57:49.198  1282  1320 I ActivityManager: Start proc 15251:org.xcsoar/u0a268 for activelaunch {org.xcsoar/org.xcsoar.XCSoar}
	Line   86: 08-22 21:57:49.209 15251 15251 I org.xcsoar: Using CollectorTypeCMC GC.
	Line   87: 08-22 21:57:49.212 15251 15251 E org.xcsoar: Not starting debugger since process cannot load the jdwp agent.
	Line   88: 08-22 21:57:49.217 15251 15251 D ActivityThread: setConscryptValidator
	Line   89: 08-22 21:57:49.217 15251 15251 D ActivityThread: setConscryptValidator - put
	Line  129: 08-22 21:57:49.235 15251 15251 D CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10268; state: ENABLED
	Line  130: 08-22 21:57:49.235 15251 15251 D CompatibilityChangeReporter: Compat change id reported: 242716250; UID 10268; state: DISABLED
	Line  144: 08-22 21:57:49.241 15251 15251 D nativeloader: Configuring clns-6 for other apk /data/app/~~atI1QQ-qQE9rY0dWHuHmlg==/org.xcsoar-n88uCLCuS0jZH9UEeNKNjA==/base.apk. target_sdk_version=33, uses_libraries=, library_path=/data/app/~~atI1QQ-qQE9rY0dWHuHmlg==/org.xcsoar-n88uCLCuS0jZH9UEeNKNjA==/lib/arm64:/data/app/~~atI1QQ-qQE9rY0dWHuHmlg==/org.xcsoar-n88uCLCuS0jZH9UEeNKNjA==/base.apk!/lib/arm64-v8a, permitted_path=/data:/mnt/expand:/data/user/0/org.xcsoar
	Line  180: 08-22 21:57:49.249 15251 15251 V GraphicsEnvironment: Currently set values for:
	Line  182: 08-22 21:57:49.249 15251 15251 V GraphicsEnvironment:   angle_gl_driver_selection_pkgs=[]
	Line  183: 08-22 21:57:49.249 15251 15251 V GraphicsEnvironment:   angle_gl_driver_selection_values=[]
	Line  184: 08-22 21:57:49.249 15251 15251 V GraphicsEnvironment: ANGLE GameManagerService for org.xcsoar: false
	Line  186: 08-22 21:57:49.249 15251 15251 V GraphicsEnvironment: org.xcsoar is not listed in per-application setting
	Line  187: 08-22 21:57:49.249 15251 15251 V GraphicsEnvironment: Neither updatable production driver nor prerelease driver is supported.
	Line  210: 08-22 21:57:49.255  1282  3708 I ActivityManager: Flag disabled. Ignoring finishAttachApplication from uid: 10268. pid: 15251
	Line  279: 08-22 21:57:49.268 15251 15251 D CompatibilityChangeReporter: Compat change id reported: 160794467; UID 10268; state: ENABLED
	Line  282: 08-22 21:57:49.269 15251 15251 I NotificationManager: org.xcsoar: notify(1, null, Notification(channel=xcsoar shortcut=null contentView=null vibrate=null sound=null defaults=0x0 flags=0x2 color=0x00000000 vis=PRIVATE semFlags=0x0 semPriority=0 semMissedCount=0)) as user
	Line  322: 08-22 21:57:49.275  1282  2088 W ActivityManager: Starting FGS with type location code=4 callerApp=ProcessRecord{8c3a32b 15251:org.xcsoar/u0a268} targetSDK=33 requiredPermissions=all of the permissions allOf=true [android.permission.FOREGROUND_SERVICE_LOCATION] any of the permissions allOf=false [android.permission.ACCESS_COARSE_LOCATION, android.permission.ACCESS_FINE_LOCATION]  and the app must be in the eligible state/exemptions to access the foreground only permission
	Line  404: 08-22 21:57:49.289  1282  1309 D PkgPredictorService-SecIpmManagerServiceImpl: onForegroundActivitiesChanged pid:15251 uid:10268
	Line  436: 08-22 21:57:49.298 15251 15251 D XCSoar  : ABI=arm64-v8a
	Line  437: 08-22 21:57:49.298 15251 15251 D XCSoar  : PRODUCT=gta9wifieea
	Line  438: 08-22 21:57:49.298 15251 15251 D XCSoar  : MANUFACTURER=samsung
	Line  439: 08-22 21:57:49.298 15251 15251 D XCSoar  : MODEL=SM-X110
	Line  440: 08-22 21:57:49.298 15251 15251 D XCSoar  : DEVICE=gta9wifi
	Line  441: 08-22 21:57:49.298 15251 15251 D XCSoar  : BOARD=gta9wifi
	Line  442: 08-22 21:57:49.298 15251 15251 D XCSoar  : FINGERPRINT=samsung/gta9wifieea/gta9wifi:14/UP1A.231005.007/X110XXS3BXG1:user/release-keys
	Line  472: 08-22 21:57:49.303 15251 15251 D XCSoar  : Successfully added bootstrap class: ioio.lib.impl.SocketIOIOConnectionBootstrap
	Line  473: 08-22 21:57:49.303 15251 15251 D XCSoar  : Successfully added bootstrap class: ioio.lib.android.accessory.AccessoryConnectionBootstrap
	Line  474: 08-22 21:57:49.303 15251 15251 D CompatibilityChangeReporter: Compat change id reported: 263076149; UID 10268; state: DISABLED
	Line  484: 08-22 21:57:49.306 15251 15251 I BluetoothAdapter: BluetoothAdapter() : org.xcsoar
	Line  485: 08-22 21:57:49.306 15251 15251 D XCSoar  : Successfully added bootstrap class: ioio.lib.android.bluetooth.BluetoothIOIOConnectionBootstrap
	Line  487: 08-22 21:57:49.306 15251 15251 D XCSoar  : Successfully added bootstrap class: ioio.lib.android.device.DeviceConnectionBootstrap
	Line  516: 08-22 21:57:49.309 15251 15251 V DevConnBootstrap: onCreate()
	Line  564: 08-22 21:57:49.314 15251 15251 D CompatibilityChangeReporter: Compat change id reported: 210923482; UID 10268; state: ENABLED
	Line  589: 08-22 21:57:49.318 15251 15251 I DecorView: setWindowBackground: isPopOver=false color=fff0f0f3 d=android.graphics.drawable.ColorDrawable@f0c5a31
	Line  610: 08-22 21:57:49.322   855  1145 D SurfaceFlinger: [SF client] NEW(0xb4000078a4a14880) for (15251:org.xcsoar)
	Line  619: 08-22 21:57:49.325   855   913 D SurfaceFlinger: [SF client] NEW(0xb400007884830f00) for (15251:org.xcsoar)
	Line  622: 08-22 21:57:49.326 15251 15251 D CompatibilityChangeReporter: Compat change id reported: 237531167; UID 10268; state: DISABLED
[... cut ...]
	Line 2730: 08-22 21:57:51.532   855   855 I Layer   : id=717 removeFromCurrentState org.xcsoar/org.xcsoar.XCSoar$_15251#717 (116)
	Line 2741: 08-22 21:57:51.533   855   855 I SurfaceFlinger: id=717 Removed org.xcsoar/org.xcsoar.XCSoar$_15251#717 (116)
	Line 2762: 08-22 21:57:51.548   855   855 I BufferQueueDebug: [org.xcsoar/org.xcsoar.XCSoar$_15251#717](this:0xb400007873ced790,id:-1,api:0,p:-1,c:855) onDestructor()
	Line 2763: 08-22 21:57:51.548   855   855 I Layer   : id=717 Destroyed org.xcsoar/org.xcsoar.XCSoar$_15251#717
```

lordfolken, 2024-08-25:
Can you please take an XCSoar-Debug.apk from github actions, and trigger the crash again?
Also add the crash dump again. The address would then point to which function triggers the crash. 
And you need to upload or tell me which binary you took.

sasam, 2024-08-25:
I can't find XCSoar-Debug.apk how to find it?

sasam, 2024-08-28:
I installed Ubuntu-22.04 on wsl2, I configured according to instructions. I managed to compile the android dbg version. The Xcsoar dbg version crashes right after starting, the initial screen with the FLY and SIM buttons does not appear. When I compile with the DEBUG=n option, the initial screen appears, but XCSoar crashes after pressing the SIM or FLY button.

XCSoar crashes in Icon.cpp. Problem with low dpi icon preparation.
```
08-28 22:37:05.179 15158 15624 I XCSoar : Display dpi=179,179
08-28 22:37:05.180 15158 15624 I XCSoar : OpenGL: mda=false npot=true stencil=0x88f0
--------- beginning of crash
08-28 22:37:05.186 15158 15624 F libc : src/ui/canvas/Icon.cpp:70: void MaskedIcon::LoadResource(ResourceId, ResourceId, ResourceId, bool): assertion "IsDefined()" failed
08-28 22:37:05.186 15158 15624 F libc : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 15624 (NativeMain), pid 15158 (org.xcsoar)
```
Here is the complete log filtered by PIDs related to XCSoar

```
08-28 22:37:04.890   413   413 I lmkd    : check proc pid launching app : 15158
08-28 22:37:04.973  1279  1301 D PkgPredictorService-SecIpmManagerServiceImpl: onForegroundActivitiesChanged pid:15158 uid:10269
08-28 22:37:04.983 15158 15158 D XCSoar  : ABI=armeabi-v7a
08-28 22:37:04.984 15158 15158 D XCSoar  : PRODUCT=gta9wifieea
08-28 22:37:04.984 15158 15158 D XCSoar  : MANUFACTURER=samsung
08-28 22:37:04.984 15158 15158 D XCSoar  : MODEL=SM-X110
08-28 22:37:04.984 15158 15158 D XCSoar  : DEVICE=gta9wifi
08-28 22:37:04.984 15158 15158 D XCSoar  : BOARD=gta9wifi
08-28 22:37:04.984 15158 15158 D XCSoar  : FINGERPRINT=samsung/gta9wifieea/gta9wifi:14/UP1A.231005.007/X110XXS3BXG1:user/release-keys
08-28 22:37:04.990 15158 15158 D XCSoar  : Successfully added bootstrap class: ioio.lib.impl.SocketIOIOConnectionBootstrap
08-28 22:37:04.990 15158 15158 D XCSoar  : Successfully added bootstrap class: ioio.lib.android.accessory.AccessoryConnectionBootstrap
08-28 22:37:04.990 15158 15158 D CompatibilityChangeReporter: Compat change id reported: 263076149; UID 10269; state: DISABLED
08-28 22:37:04.995 15158 15158 I BluetoothAdapter: BluetoothAdapter() : org.xcsoar
08-28 22:37:04.995 15158 15158 D XCSoar  : Successfully added bootstrap class: ioio.lib.android.bluetooth.BluetoothIOIOConnectionBootstrap
08-28 22:37:04.995 15158 15158 D XCSoar  : Successfully added bootstrap class: ioio.lib.android.device.DeviceConnectionBootstrap
08-28 22:37:04.997 15158 15158 V DevConnBootstrap: onCreate()
08-28 22:37:05.006 15158 15158 D CompatibilityChangeReporter: Compat change id reported: 210923482; UID 10269; state: ENABLED
08-28 22:37:05.011 15158 15158 I DecorView: setWindowBackground: isPopOver=false color=fff0f0f3 d=android.graphics.drawable.ColorDrawable@2b3bfb
08-28 22:37:05.016   857   917 D SurfaceFlinger: [SF client] NEW(0xb4000071ac0c8500) for (15158:org.xcsoar)
08-28 22:37:05.019   857   917 D SurfaceFlinger: [SF client] NEW(0xb4000071ac0c8680) for (15158:org.xcsoar)
08-28 22:37:05.020 15158 15158 D CompatibilityChangeReporter: Compat change id reported: 237531167; UID 10269; state: DISABLED
08-28 22:37:05.025 15158 15158 I InsetsController: onStateChanged: host=org.xcsoar/org.xcsoar.XCSoar, from=android.view.ViewRootImpl.setView:1753, state=InsetsState: {mDisplayFrame=Rect(0, 0 - 800, 1340), mDisplayCutout=DisplayCutout{insets=Rect(0, 0 - 0, 0) waterfall=Insets{left=0, top=0, right=0, bottom=0} boundingRect={Bounds=[Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0)]} cutoutPathParserInfo={CutoutPathParserInfo{displayWidth=0 displayHeight=0 physicalDisplayWidth=0 physicalDisplayHeight=0 density={0.0} cutoutSpec={} rotation={0} scale={0.0} physicalPixelDisplaySizeRatio={0.0}}}}, mRoundedCorners=RoundedCorners{[RoundedCorner{position=TopLeft, radius=17, center=Point(17, 17)}, RoundedCorner{position=TopRight, radius=17, center=Point(783, 17)}, RoundedCorner{position=BottomRight, radius=17, center=Point(783, 1323)}, RoundedCorner{position=BottomLeft, radius=17, center=Point(17, 1323)}]}  mRoundedCornerFrame=Rect(0, 0 - 800, 1340), mPrivacyIndicatorBounds=PrivacyIndicatorBounds {static bounds=Rect(742, 0 - 800, 32) rotation=0}, mDisplayShape=DisplayShape{ spec=-311912193 displayWidth=800 displayHeight=1340 physicalPixelDisplaySizeRatio=1.0 rotation=0 offsetX=0 offsetY=0 scale=1.0}, mSources= { InsetsSource: {3 mType=ime mFrame=[0,0][0,0] mVisible=false mFlags=[]}, InsetsSource: {3dc00001 mType=navigationBars mFrame=[0,1276][800,1340] mVisible=true mFlags=[]}, InsetsSource: {3dc00004 mType=systemGestures mFrame=[0,0][0,0] mVisible=true mFlags=[]}, InsetsSource: {3dc00005 mType=mandatorySystemGestures mFrame=[0,1276][800,1340] mVisible=true mFlags=[]}, InsetsSource: {3dc00006 mType=tappableElement mFrame=[0,1276][800,1340] mVisible=true mFlags=[]}, InsetsSource: {3dc00024 mType=systemGestures mFrame=[0,0][0,0] mVisible=true mFlags=[]}, InsetsSource: {46420000 mType=statusBars mFrame=[0,0][800,32] mVisible=true mFlags=[]}, InsetsSource: {46420005 mType=mandatorySystemGestures mFrame=[0,0][800,32] mVisible=true mFlags=[]}, InsetsSource: {46420006 mType=tappableEleme ...
08-28 22:37:05.026 15158 15158 I ViewRootImpl@7eeefe2[XCSoar]: synced displayState. AttachInfo displayState=2
08-28 22:37:05.027 15158 15158 I ViewRootImpl@7eeefe2[XCSoar]: setView = com.android.internal.policy.DecorView@c77baa9 TM=true
08-28 22:37:05.027 15158 15158 I IDS_TAG : Starting IDS observe window
[... cut ...]
08-28 22:37:05.788   857   857 I Layer   : Layer [org.xcsoar/org.xcsoar.XCSoar$_15158#724] hidden!! flag(1)
08-28 22:37:05.788   857   857 I Layer   : id=724 removeFromCurrentState org.xcsoar/org.xcsoar.XCSoar$_15158#724 (144)
08-28 22:37:05.789   857   857 I SurfaceFlinger: id=724 Removed org.xcsoar/org.xcsoar.XCSoar$_15158#724 (144)
08-28 22:37:05.804   857   857 I BufferQueueDebug: [org.xcsoar/org.xcsoar.XCSoar$_15158#724](this:0xb40000717b078790,id:-1,api:0,p:-1,c:857) onDestructor()
08-28 22:37:05.804   857   857 I Layer   : id=724 Destroyed org.xcsoar/org.xcsoar.XCSoar$_15158#724
```

lordfolken, 2024-08-29:
So i think i found it... 

This is a 120dpi virtual device: it selects the map_teammate_160 resource

[2024-08-29T21:55:22Z] Starting XCSoar XCSoar v7.44-Android~git#2020f3e181
[2024-08-29T21:55:22Z] Failed to initialise Bluetooth: java.lang.Exception: No Bluetooth adapter found
[2024-08-29T21:55:22Z] EGL vendor: Android
[2024-08-29T21:55:22Z] EGL version: 1.4 Android META-EGL
[2024-08-29T21:55:22Z] EGL extensions: EGL_KHR_get_all_proc_addresses EGL_ANDROID_presentation_time EGL_KHR_swap_buffers_with_damage EGL_KHR_image_base EGL_KHR_gl_texture_2D_image EGL_KHR_fence_sync EGL_KHR_create_context EGL_ANDROID_image_native_buffer EGL_ANDROID_recordable 
[2024-08-29T21:55:22Z] EGL config: RGB=8/8/8 alpha=8 depth=24 stencil=8
[2024-08-29T21:55:22Z] GL vendor: Google (AMD)
[2024-08-29T21:55:22Z] GL version: OpenGL ES 3.1 (4.6 (Core Profile) Mesa 22.3.6)
[2024-08-29T21:55:22Z] GL renderer: Android Emulator OpenGL ES Translator (AMD Radeon Graphics (renoir, LLVM 15.0.6, DRM 3.49, 6.1.0-18-amd64))
[2024-08-29T21:55:22Z] GL extensions: GL_EXT_debug_marker GL_OES_EGL_sync GL_OES_EGL_image GL_OES_EGL_image_external GL_OES_depth24 GL_OES_depth32 GL_OES_element_index_uint GL_OES_texture_float GL_OES_texture_float_linear GL_OES_compressed_paletted_texture GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth_texture GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_packed_depth_stencil GL_OES_vertex_half_float GL_OES_texture_npot GL_OES_rgb8_rgba8 GL_EXT_color_buffer_float GL_EXT_color_buffer_half_float GL_EXT_texture_format_BGRA8888 GL_APPLE_texture_format_BGRA8888 ANDROID_EMU_CHECKSUM_HELPER_v1 ANDROID_EMU_native_sync_v2 ANDROID_EMU_native_sync_v3 ANDROID_EMU_native_sync_v4 ANDROID_EMU_dma_v1 ANDROID_EMU_sync_buffer_data GL_OES_EGL_image_external_essl3 GL_OES_vertex_array_object GL_KHR_texture_compression_astc_ldr ANDROID_EMU_host_side_tracing ANDROID_EMU_gles_max_version_3_1 
[2024-08-29T21:55:22Z] Display dpi=120,120
[2024-08-29T21:55:22Z] OpenGL: mda=false npot=true stencil=0x88f0
Loading resource: map_teammate_160
[2024-08-29T21:55:24Z] Startup dialog
[2024-08-29T21:55:24Z] Loading profiles
[2024-08-29T21:55:24Z] Failed to load profile: Failed to open '/storage/emulated/0/Android/media/org.xcsoar/default.prf': No such file or directory
Loading resource: map_small_160
Loading resource: map_turnpoint_160
Loading resource: map_taskturnpoint_160
Loading resource: map_mountain_top_160
Loading resource: map_pass_160
Loading resource: map_bridge_160
Loading resource: map_tunnel_160
Loading resource: map_tower_160
Loading resource: map_power_plant_160
Loading resource: map_obstacle_160
Loading resource: map_thermal_hotspot_160
Loading resource: map_flag_160
Loading resource: map_vor_160
Loading resource: map_ndb_160
Loading resource: map_dam_160
Loading resource: map_castle_160
Loading resource: map_intersection_160
Loading resource: map_reporting_point_160
Loading resource: map_pgtakeoff_160
Loading resource: map_pglanding_160
Loading resource: winpilot_reachable_160
Loading resource: winpilot_marginal_160
Loading resource: winpilot_landable_160
Loading resource: winpilot_reachable_160
Loading resource: winpilot_marginal_160
Loading resource: winpilot_landable_160
Loading resource: map_target_160
Loading resource: map_weather_station_160
Loading resource: map_terrainw_160
Loading resource: flarm_traffic_160
Loading resource: flarm_warning_160
Loading resource: flarm_alarm_160
Loading resource: map_thermal_source_160
Loading resource: flarm_traffic_160
Loading resource: flarm_warning_160
Loading resource: flarm_alarm_160
Loading resource: mode_cruise_160
Loading resource: mode_climb_160
Loading resource: mode_finalglide_160
Loading resource: mode_abort_160
Loading resource: gps_acquiring_160
Loading resource: gps_disconnected_160
Loading resource: airspace_intercept_160
Loading resource: scalearrow_left_160
Loading resource: scalearrow_right_160
Loading resource: task_160
Loading resource: wrench_160
Loading resource: settings_160
Loading resource: calculator_160
Loading resource: globe_160
[... 9 lines cut ...]
[2024-08-29T21:55:24Z] RASP load
[2024-08-29T21:55:24Z] No rasp data file: Failed to open ZIP archive /storage/emulated/0/Android/media/org.xcsoar/xcsoar-rasp.dat
[2024-08-29T21:55:24Z] ReadAirspace
[2024-08-29T21:55:24Z] Register serial devices
[2024-08-29T21:55:24Z] ProgramStarted
[2024-08-29T21:55:24Z] cannot open /storage/emulated/0/Android/media/org.xcsoar/lua/init.lua: No such file or directory

However when the device is really low dpi:

[2024-08-29T21:59:43Z] Starting XCSoar XCSoar v7.44-Android~git#2020f3e181
[2024-08-29T21:59:43Z] Failed to initialise Bluetooth: java.lang.Exception: No Bluetooth adapter found
[2024-08-29T21:59:43Z] EGL vendor: Android
[2024-08-29T21:59:43Z] EGL version: 1.4 Android META-EGL
[2024-08-29T21:59:43Z] EGL extensions: EGL_KHR_get_all_proc_addresses EGL_ANDROID_presentation_time EGL_KHR_swap_buffers_with_damage EGL_KHR_image_base EGL_KHR_gl_texture_2D_image EGL_KHR_fence_sync EGL_KHR_create_context EGL_ANDROID_image_native_buffer EGL_ANDROID_recordable 
[2024-08-29T21:59:43Z] EGL config: RGB=8/8/8 alpha=8 depth=24 stencil=8
[2024-08-29T21:59:43Z] GL vendor: Google (AMD)
[2024-08-29T21:59:43Z] GL version: OpenGL ES 3.1 (4.6 (Core Profile) Mesa 22.3.6)
[2024-08-29T21:59:43Z] GL renderer: Android Emulator OpenGL ES Translator (AMD Radeon Graphics (renoir, LLVM 15.0.6, DRM 3.49, 6.1.0-18-amd64))
[... cut ...] ES_EGL_image_external_essl3 GL_OES_vertex_array_object GL_KHR_texture_compression_astc_ldr ANDROID_EMU_host_side_tracing ANDROID_EMU_gles_max_version_3_1 
[2024-08-29T21:59:43Z] Display dpi=120,120
[2024-08-29T21:59:43Z] OpenGL: mda=false npot=true stencil=0x88f0
Loading resource: map_teammate
src/ui/canvas/Icon.cpp:118: void MaskedIcon::LoadResource(ResourceId, ResourceId, ResourceId, bool): assertion "IsDefined()" failed

Its trying to load map_teammate, it should however load map_teammate_96.

lordfolken, 2024-08-29:
@sasam i believe my attached pr fixes the issue. Can you please test it on your device?

sasam, 2024-08-29:
Yes it work now. THX!

closed 2024-09-02 by lordfolken

OllieRz, 2024-09-08:
Hi, sorry can you please explain the solution step by step as I am not that strong in programming? Thank you!

lordfolken, 2024-09-09:
You install the package from https://download.xcsoar.org/testing/ANDROID/XCSoar.apk

This should start without issue.
Will be in 7.44

OllieRz, 2024-09-09:
Thanks a lot, Phillipp!
Best regards