XCSoar/XCSoar #1546: Startup Wizard - Display Gesture Help

lordfolken, 2024-08-23:
When starting XCSoar for the first time, display a Popup that introduces you to the guestures that are possible.

WolfgangDrescher, 2025-08-28:
I'm working on a small "onboarding" dialog:

```cpp
void
dlgOnboardingShowModal()
{
  const DialogLook &look = UIGlobals::GetDialogLook();

  WidgetDialog dialog(WidgetDialog::Full{}, UIGlobals::GetMainWindow(),
                      look, _("Onboarding"));

  auto text_widget = std::make_unique<LargeTextWidget>(
    look,
    _("Welcome to XCSoar!\nPlease follow the instructions.")
  );

  dialog.FinishPreliminary(std::move(text_widget));
  dialog.AddButton(_("Close"), mrCancel);
  dialog.ShowModal();
}
```

And in `Startup.cpp`:

```cpp
  // Show onboarding dialog
  if (Profile::GetPath(ProfileKeys::MapFile) == nullptr) {
    dlgOnboardingShowModal();
  }
```

However, I have a few questions, as I don't have an overview of the available widgets.

1. To determine whether XCSoar has already been set up, I use `Profile::GetPath(ProfileKeys::MapFile)`. Is this appropriate?
2. What is the best way to create a list with e.g. multiple `LargeTextWidget` items that provide the user with tips on what to do (setup “Site Files,” airspaces, InfoBox pages, etc.)? Is `ListWidget` useful for this?
3. How can I use a button `dialog.AddButton(_(“Settings”), [](){});` to refer to `Config` -> `System`?
4. Should I create a new `GridWidget` for the icons of the gestures with the corresponding descriptions, or is there already something useful for this?
5. Can multiple widgets be combined? (`LargeTextWidget` + `GridWidget`)

lordfolken, 2025-09-02:
1. Please use your own ProfileKey:: e.g. GestureHelpShown. Write that one into the Profile if displayed once, unless there a check box  "show again on startup" is selected. It also should only be shown if HasTouchScreen() is true.
2. LargeTextWidget was filling the screen typically. There is also the issue of scrolling and overflow espeically when using the keyboard to operate. Maybe have a look at dlgCredits.cpp there multiple LTWs are in use. 
3. That one is a little ... tricky. The menu calls SystemConfiguration();
4. Both the waypoint and airspace list  dialog use entries with pictures. Potentially that could be used?
5. I'm worried for keyboard operability there.

WolfgangDrescher, 2025-09-02:
Should I use the icons from `doc/manual/en/figures/*.png` or would you prefer `Data/gestures/*.png`? I think the icons from the docs look better.

How can I properly add the PNGs as resources?

Currently I have in `resource.mk`:

```
GESTURES = dr.png rd.png dl.png ud.png
GESTURES_DST = $(addprefix gestures/,$(GESTURES))

gestures/dirstamp:
	@$(NQ)echo "  MKDIR   gestures/"
	$(Q)mkdir -p gestures
	@touch $@

$(GESTURES_DST): gestures/%.png: doc/manual/en/figures/%.png | gestures/dirstamp
	$(Q)cp $< $@

RESOURCE_FILES += $(GESTURES_DST)
```

`resources.txt`:

```
bitmap_graphic IDB_GESTURE_DR "gesture_dr"
bitmap_graphic IDB_GESTURE_RD "gesture_rd"
bitmap_graphic IDB_GESTURE_DL "gesture_dl"
bitmap_graphic IDB_GESTURE_UD "gesture_ud"
```

`resources.c` can't compile and it looks like the images need to be copied into `graphics2`:

```
#include <stddef.h>
#include <stdint.h>
asm(
".p2align 3\n"
".data\n"
"resource_IDB_GESTURE_DR:\n"
"_resource_IDB_GESTURE_DR:\n"
".globl resource_IDB_GESTURE_DR\n"
".globl _resource_IDB_GESTURE_DR\n"
".incbin \"output/data/graphics2/gesture_dr.png\"\n"
);
extern const uint8_t resource_IDB_GESTURE_DR[];
const uint8_t * const resource_IDB_GESTURE_DR_end
    = resource_IDB_GESTURE_DR + ;
const size_t resource_IDB_GESTURE_DR_size
    = ;
```

How can I properly copy the gesture icons as resources?

WolfgangDrescher, 2025-09-03:
I managed to get `resource.mk` working. I added the code snipped at the end of the file, but `RESOURCE_FILES` had to be populated in before calling `MAKE_RESOURCE`.

lordfolken, 2025-09-04:
> The question is what we want to achieve. Is a simple “Display Gesture Help” dialog sufficient, or do we want to create a more comprehensive onboarding process that briefly explains the most important configurations?

> Some younger members of our club download XCSoar and don't know where to start (and are too lazy to read the documentation...). I think it would be useful to at least include a note about loading the maps.

I hope they read the manual of the glider before getting in. *scnr* 

While On a more serious note: yes "discoverability" in XCSoar is mostly in existent and that does need to be changed. 

> I have also experimented with `ArrowPagerWidget`. We could have a page for explanations off the most important settings (and gestures). With `ProfileKey::*`, we can hide individual pages or the entire dialog once the settings have been made.

What would be cool would be a button on the menu -> info -> info -> info -> Gesture Help or similar, where the dialog could be called at will.

lordfolken, 2026-04-01:
Linking related PR: https://github.com/XCSoar/XCSoar/pull/2043

closed 2026-04-01 by lordfolken